Koenkk/zigbee2mqtt #3159: Cannot bind or unbind Aqara MCCGQ11LM door sensor

sjorge, 2020-03-20:
# Bug Report
## What happened
```
mosquitto_pub -h localhost -p 1883 -t zigbee2mqtt/bridge/unbind/0x00158d00044faf3e -m 'coordinator'
```
Doesn't seem to do anything, when I try to bind a group (or device)
```
mosquitto_pub -h localhost -p 1883 -t zigbee2mqtt/bridge/bind/0x00158d00044faf3e -m 'bedroom/desk_lamp'
```

It has the same effect, nothing is logged at all with log level set to debug?

## What did you expect to happen
The device seems to send message to the genOnOff cluster
```
zigbee2mqtt:debug 2020-03-20 12:12:08: Received Zigbee message from '0x00158d00044faf3e', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-20 12:12:09: MQTT publish: topic 'zigbee2mqtt/0x00158d00044faf3e', payload '{"battery":100,"voltage":3055,"contact":false,"linkquality":47}'
zigbee2mqtt:debug 2020-03-20 12:12:10: Received Zigbee message from '0x00158d00044faf3e', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
```

So I was expecting to be able to unbind it from the coordiantor, and then bind it to a group of devices... e.g. bulbs, so they would go off or on depending on the state.

## How to reproduce it (minimal and precise)
Pair a Aqara MCCGQ11LM, try to (un)bind a device or group from it... It's odd nothing is logged at all, at minimal I would expect a 'not support' or something message to be logged.

## Debug Info
zigbee2mqtt version: `1.12.0-dev (commit #ac8a84f)`
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}`

sjorge, 2020-03-20:
More specifically, the idea was to test it with a bulb first... then to link it to a TRV

Koenkk, 2020-03-20:
AFAIK these devices don't support binding.

sjorge, 2020-03-20:
I guess we should at least log a message then? Doesn't look like z2m even attempted to bind though.

Also do you know of any that do support binding?

Koenkk, 2020-03-20:
- Added an error log to the latest dev branch
- I've checked contact sensor using `genOnOff` cluster, these are Xiaomi (don't support binding) and the https://www.zigbee2mqtt.io/devices/DIYRuZ_magnet.html (not sure if it supports binding)

Can you confirm if it works as expected?

sjorge, 2020-03-21:
Error log is working, but they are not yet 100% correct:
```
zigbee2mqtt:debug 2020-03-21 20:29:02: Received MQTT message on 'zigbee2mqtt/bridge/unbind/0x00158d00044faf3e' with data 'coordinator'
zigbee2mqtt:error 2020-03-21 20:29:02: Nothing to bind from '0x00158d00044faf3e' to 'Coordinator'
```

This should say 'Nothing to unbind' instead of 'Nothing to bind'

sjorge, 2020-03-21:
It's a super simple fix: PR #3172

Koenkk, 2020-03-21:
thanks, merged! :smile:

closed 2020-03-21 by sjorge