Koenkk/zigbee2mqtt #3162: Problems after update to 1.12.0

Hubschra01, 2020-03-20:
Hello,
alter update on 1.12.0 my Coordinator Stick will not work. 
Every 10 sec i get a new log file with this content 

info  2020-03-20 15:34:24: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-20.15-34-24' filename: log.txt
debug 2020-03-20 15:34:24: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-03-20.15-29-50'
debug 2020-03-20 15:34:25: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2020-03-20 15:34:25: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
info  2020-03-20 15:34:25: Starting zigbee-herdsman...
debug 2020-03-20 15:34:25: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":null}}'
error 2020-03-20 15:34:32: Error while starting zigbee-herdsman

I Use the Stick on a headless openhab 2.5.2 on the Raspberry pi 3+ Hardware
The Stick (CC2531) software is from jun 2019.

Thanks 
André

Koenkk, 2020-03-24:
Could you provide the zigbee herdsman debug log when the configure of the tradfri signal repeater happens?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Hubschra01, 2020-03-25:
Here is the log from the tradfri signal repeater when I send the update check via 
`zigbee2mqtt/bridge/ota_update/check` 

[https://hastebin.com/hesikaxeni.coffeescript](url)

Hubschra01, 2020-03-25:
I disconnect the Stick from the pi and start a new interview by pressing the reset button of the Tradfri signal repeater. Afer the interview the Stick disconnected by Error...
[https://hastebin.com/dubupepeqo.coffeescript](url)

Koenkk, 2020-03-26:
```
2020-03-25T13:34:45.789Z zigbee-herdsman:adapter:zStack:znp:log Port closed
2020-03-25T13:34:45.790Z zigbee-herdsman:controller:log Adapter disconnected'
```

it seems that the adapter disconnects, could it be a power issue? How is the CC2531 connected?

Hubschra01, 2020-03-27:
The Adapter is directly plugged in the USB Port at the Raspberry Pi. Without any cable. 
The LED of the Stick is always on.
It´s possible the serial driver disconnect the Stick by Error ? I think the Error comes with the update...

Koenkk, 2020-03-27:
If you think it comes with this update, can you check on which version it still works?

Hubschra01, 2020-03-27:
I remember there were some bugs in the update process. I looked over the npm list and found some missing sentences requested by Zigbee-Herdman sentences at the bottom of the list.
is that normal.
[npm list](https://hastebin.com/sajefiqulo.pl)

Koenkk, 2020-03-27:
No this is not related to this.

Hubschra01, 2020-03-28:
I tested with 1.11.0 i think the Interview were stable but i don´t know why the Apapter disconnects...

`2020-03-28T19:18:34.504Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-03-28T19:19:01.689Z zigbee-herdsman:adapter:zStack:znp:log Port closed
2020-03-28T19:19:01.690Z zigbee-herdsman:controller:log Adapter disconnected'
`

[Lo1.11.0](https://hastebin.com/awupahuvey.coffeescript)

After the disconnect the Stick isn´t aviable I have to remove the Stick from the USB Port an reconnect it.
Is the Stick buggy?

Koenkk, 2020-03-28:
So on 1.11.0 the same issue? This means it's not related to the update?

Maybe the stick is buggy indeed.

Hubschra01, 2020-03-28:
Tomorrow i will give another Stick a try with 1.11.0

Hubschra01, 2020-03-29:
It seems to be the Stick was defect. I test a new on from a friend and this Stick works fine on 1.11.0
Later i will try with 1.12.0

pixeldoc2000, 2020-04-04:
@Hubschra01 

Maybe reflash the stick with the latest firmware?

Otherwise i guess is a power issus. maybe connect the stick via a powered usb hub.

Hubschra01, 2020-04-04:
Hi,
got a new Stick and this this works fine since 3 days. So I think the Stick has an issuse. I will try to fash the the stick later when i get a cheap cc debugger.
So I will close this.
Thanks for help

closed 2020-04-04 by Hubschra01