Koenkk/zigbee2mqtt #3163: Some lights turn off when brightness is set to 1

mariusmotea, 2020-03-20:
# Bug Report
## What happened
Light is turned off when the brightness is set to 1 (minimum). This happens for my only with Philips LCA001 bulb and GLEDOPTO RGB-CCT controller.

## What did you expect to happen
Have the light at minimum possible brightness

## How to reproduce it (minimal and precise)
Set the brightness to 1 .

## Debug Info

zigbee2mqtt version: latest
CC253X firmware version: latest

Koenkk, 2020-03-20:
This is not something that can be directly solved from Zigbee2mqtt as this logic is inside the firmware of the bulb.

mariusmotea, 2020-03-20:
Any plan to fix it at zigbee2mqtt level, if not i will implement an workaround.

Also, on brightness change the light is turned on. Is this also in the light firmware?

Koenkk, 2020-03-21:
We can implement some logic like if brightness === 1 we change it to 2 internally (but its rather hacky I feel)

closed 2020-05-30 by stale[bot]