Koenkk/zigbee2mqtt #3164: Problem Heiman Smoke Detector :  no alert info

kwetnico, 2020-03-20:
# Bug Report
## What happened 
hello, i just added a new device to my zigbee network. I have the latest version Zigbeemqtt (1.12.0) with Jeedom on a rapsberry pi 4. and my cc2531 firmware key from June 2019.

I just added a Heiman zigbee smoke detector. It is well detected under zigbeemqtt but it already indicates a model which is not mine. On my label it says that I have an HS1SA (zigbee) and in the logs it says that I have one: HS3A.
"
mars 20 17:35:41 raspberrypi npm[1926]: zigbee2mqtt:info  2020-03-20 17:35:41: Starting interview of '0x00158d0004c07fa8'
mars 20 17:35:41 raspberrypi npm[1926]: zigbee2mqtt:info  2020-03-20 17:35:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0004c07fa8"}}'
mars 20 17:35:41 raspberrypi npm[1926]: zigbee2mqtt:info  2020-03-20 17:35:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d0004c07fa8"}}'
mars 20 17:36:38 raspberrypi npm[1926]: zigbee2mqtt:info  2020-03-20 17:36:38: Successfully interviewed '0x00158d0004c07fa8', device has successfully been paired
mars 20 17:36:38 raspberrypi npm[1926]: zigbee2mqtt:info  2020-03-20 17:36:38: Device '0x00158d0004c07fa8' is supported, identified as: HEIMAN Smoke detector (HS3SA)
mars 20 17:36:38 raspberrypi npm[1926]: zigbee2mqtt:info  2020-03-20 17:36:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d0004c07fa8","model":"HS3SA","vendor":"HEIMAN","description":"Smoke detector","supported":true}}'
mars 20 17:36:38 raspberrypi npm[1926]: zigbee2mqtt:info  2020-03-20 17:36:38: Configuring '0x00158d0004c07fa8'
mars 20 17:36:41 raspberrypi npm[1926]: zigbee2mqtt:info  2020-03-20 17:36:41: MQTT publish: topic 'zigbee2mqtt/0x00158d0004c07fa8', payload '{"battery":80,"linkquality":78}'
mars 20 17:36:41 raspberrypi npm[1926]: zigbee2mqtt:info  2020-03-20 17:36:41: Successfully configured '0x00158d0004c07fa8'
" 

The informations I have are just Battery and Linkquality. I launched a smoke test, I do not have SMOKE going back but a message that here:

"
mars 20 17:40:15 raspberrypi npm[15236]: zigbee2mqtt:info  2020-03-20 17:40:15: MQTT publish: topic 'zigbee2mqtt/DetecteurFumee', payload '{"battery":80,"linkquality":78}'
mars 20 17:40:15 raspberrypi npm[15236]: zigbee2mqtt:info  2020-03-20 17:40:15: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.0","commit":"840b9d9","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
mars 20 17:43:25 raspberrypi npm[15236]: zigbee2mqtt:info  2020-03-20 17:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"DetecteurFumee"}}'
mars 20 17:43:51 raspberrypi npm[15236]: zigbee2mqtt:info  2020-03-20 17:43:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"DetecteurFumee"}}'
mars 20 17:43:52 raspberrypi npm[15236]: zigbee2mqtt:info  2020-03-20 17:43:52: MQTT publish: topic 'zigbee2mqtt/DetecteurFumee', payload '{"battery":50,"linkquality":78}'
"
Please help, i don't have this device class : 
binary_sensor:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/<FRIENDLY_NAME>"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_on: true
    payload_off: false
    value_template: "{{ value_json.smoke }}"
    device_class: "smoke"

Thanks

Best regards

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.12.0
CC253X firmware version: 062019

kwetnico, 2020-03-20:
Hello, I just exchanged with the seller.  It is a new model of the Heiman HS1SA brand in ZigBee 3.0.  It should therefore be integrated into the zigbeemqtt solution, because he believes that it is the heiman HS3SA model and suddenly I do not have any SMOKE alert feedback if it sounds.  thank you.

kwetnico, 2020-03-21:
Hello, I found that the device.js file should be updated in the next version for this new Heiman model, there is an error:
Create a new Device: HS1SA and delete the device: 'SmokeSensor-N-3.0' from HS3SA

 {
        zigbeeModel: ['SmokeSensor-N-3.0', 'HS1SA'],
        model: 'HS1SA',
        vendor: 'HEIMAN',
        description: 'Smoke detector',
        supports: 'smoke',
        fromZigbee: [
            fz.heiman_smoke,
            fz.battery,
            fz.heiman_smoke_enrolled,
        ],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
            await configureReporting.batteryAlarmState(endpoint);
        },
    },	
    {
        zigbeeModel: ['SmokeSensor-N'],
        model: 'HS3SA',
        vendor: 'HEIMAN',
        description: 'Smoke detector',
        supports: 'smoke',
        fromZigbee: [fz.heiman_smoke, fz.battery],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },

And it'sok : 

mars 21 12:28:17 raspberrypi npm[3304]: zigbee2mqtt:debug 2020-03-21 12:28:17: Received Zigbee message from '0x00158d0004c07fa8', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":1}' from endpoint 1 with groupID 0
mars 21 12:28:17 raspberrypi npm[3304]: zigbee2mqtt:info  2020-03-21 12:28:17: MQTT publish: topic 'zigbee2mqtt/0x00158d0004c07fa8', payload '{"battery":80,"linkquality":94,"enrolled":true}'
mars 21 12:28:25 raspberrypi npm[3304]: zigbee2mqtt:debug 2020-03-21 12:28:25: Received Zigbee message from '0x00158d0004c07fa8', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b00194ad855","zoneState":1}' from endpoint 1 with groupID 0
mars 21 12:28:25 raspberrypi npm[3304]: zigbee2mqtt:info  2020-03-21 12:28:25: MQTT publish: topic 'zigbee2mqtt/0x00158d0004c07fa8', payload '{"battery":80,"linkquality":94,"enrolled":true}'
mars 21 12:28:25 raspberrypi npm[3304]: zigbee2mqtt:info  2020-03-21 12:28:25: Successfully interviewed '0x00158d0004c07fa8', device has successfully been paired
mars 21 12:28:25 raspberrypi npm[3304]: zigbee2mqtt:info  2020-03-21 12:28:25: Device '0x00158d0004c07fa8' is supported, identified as: HEIMAN Smoke detector (HS1SA)
mars 21 12:28:25 raspberrypi npm[3304]: zigbee2mqtt:info  2020-03-21 12:28:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d0004c07fa8","model":"HS1SA","vendor":"HEIMAN","description":"Smoke detector","supported":true}}'
mars 21 12:28:25 raspberrypi npm[3304]: zigbee2mqtt:info  2020-03-21 12:28:25: Configuring '0x00158d0004c07fa8'

And alert : 

mars 21 12:29:03 raspberrypi npm[3304]: zigbee2mqtt:info  2020-03-21 12:29:03: MQTT publish: topic 'zigbee2mqtt/0x00158d0004c07fa8', payload '{"battery":96.5,"linkquality":92,"enrolled":true,"smoke":true,"battery_low":false}'

and the alerts go up well (smoke) .. Did not work until now ..

Database.db : {"id":26,"type":"EndDevice","ieeeAddr":"0x00158d0004c07fa8","nwkAddr":25651,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"SmokeSensor-N-3.0","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,1282,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SmokeSensor-N-3.0","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":2,"appVersion":20,"stackVersion":2,"hwVersion":16,"dateCode":"2018.9.27","swBuildId":"2000-0001"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00194ad855","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":193}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad855","endpointID":1}]}},"appVersion":20,"stackVersion":2,"hwVersion":16,"dateCode":"2018.9.27","swBuildId":"2000-0001","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1584790113861}

dfsx1, 2020-03-21:
I can confirm my HS1SA is also listed as HS3SA in Zigbee2MQTT.
I bought this 2 weeks ago ($19,99 AliExpress), and it has zigbee 3.0 on the label.
I did not test with real smoke yet, but I guess I should now :)

PS:
Not sure where the HS3SA model comes from.
HS3SA is nowhere on their product list on heimantech.com / heiman.com.cn

[image]

kwetnico, 2020-03-21:
yes this is zigbee 3.0 new version .. This is why it is necessary to update in devices.js for the next version, the I just made the modification it works but I have a problem, it sends every 3 seconds of notifications !!! and tells me smoke: true alos or I launched a test to see the alert raised but do I have to wait until it goes back to standby? or is it normal? Thank you

March 21 13:17:19 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:19: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 89, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:22 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:22: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 89, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:25 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:25: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 89, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:28 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:28: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 89, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:31 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:31: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 89, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:34 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:34: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 86, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:38 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:38: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 86, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:41 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:41: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 89, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:44 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:44: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 92, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:47 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:47: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 92, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:50 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:50: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 89, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:53 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:53: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 86, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:17:57 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:17:57: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 92, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:18:00 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:18:00: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 89, "enrolled": true, "smoke": true, "battery_low": false} '
March 21 13:18:03 raspberrypi npm [32243]: zigbee2mqtt: info 2020-03-21 13:18:03: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 76.5, "linkquality": 89, "enrolled": true, "smoke": true, "battery_low": false} '

dfsx1, 2020-03-21:
If I press the "test" button, it beeps 3 times and reports the smoke only for 1 second.
This behaviour looks good to me.

Note that I did not do your suggested change in device.js.
I think it's only a name issue, nothing functional (not tested with real smoke however).
```
zigbee2mqtt:info  2020-03-21 13:30:23: MQTT publish: topic 'zigbee2mqtt/0x00158d0004c0a747', payload '{"battery":100,"linkquality":52,"smoke":true,"battery_low":false}'
zigbee2mqtt:info  2020-03-21 13:30:24: MQTT publish: topic 'zigbee2mqtt/0x00158d0004c0a747', payload '{"battery":100,"linkquality":23,"smoke":false,"battery_low":false}'
```

kwetnico, 2020-03-21:
> Si j'appuie sur le bouton "test", il émet 3 bips et ne signale la fumée que pendant 1 seconde.
> Ce comportement me semble bon.
> 
> Notez que je n'ai pas effectué la modification suggérée dans device.js.
> Je pense que HS3SA n'est qu'un problème de nom, rien de fonctionnel (non testé avec de la vraie fumée cependant).
> 
> ```
> zigbee2mqtt:info  2020-03-21 13:30:23: MQTT publish: topic 'zigbee2mqtt/0x00158d0004c0a747', payload '{"battery":100,"linkquality":52,"smoke":true,"battery_low":false}'
> zigbee2mqtt:info  2020-03-21 13:30:24: MQTT publish: topic 'zigbee2mqtt/0x00158d0004c0a747', payload '{"battery":100,"linkquality":23,"smoke":false,"battery_low":false}'
> ```

Me if I press the test button once, it rings continuously without stopping ... I have to remove the battery and I don't have the same feedback as you on MQTT .. :(

dfsx1, 2020-03-21:
I think it stays in alarm mode, until you press the button to "hush".
http://www.heimantech.com/uploads/2018/01/030813172350.pdf

[image]

kwetnico, 2020-03-21:
I press once to start the test.
it rings without stopping .... even if I press the same button ..

dfsx1, 2020-03-21:
> I press once to start the test.
> it rings without stopping .... even if I press the same button ..

Press the button (while alarm sounds) for 3 seconds maybe?

kwetnico, 2020-03-21:
already done... :(
I can't even match it in my zigbeemqtt system anymore I gave the original device.js file deleted from database.db + state.json + configuration.yaml and impossible .. I will have to reflash my cc2531 key I believe....
It's a great but fragile system.

kwetnico, 2020-03-21:
> > J'appuie une fois pour lancer le test. 
> > ça sonne sans s'arrêter .... même si j'appuie sur le même bouton ..
> 
> Appuyez sur le bouton (pendant que l'alarme sonne) pendant 3 secondes peut-être?

I tried by pressing once, 3 seconds ...
there must be a defect :(

kwetnico, 2020-03-21:
With the original file devices.js : 

mars 21 14:21:08 raspberrypi npm[1513]: zigbee2mqtt:info  2020-03-21 14:21:08: Device '0x00158d0004c07fa8' joined
mars 21 14:21:08 raspberrypi npm[1513]: zigbee2mqtt:info  2020-03-21 14:21:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0004c07fa8"}}'
mars 21 14:21:08 raspberrypi npm[1513]: zigbee2mqtt:info  2020-03-21 14:21:08: Starting interview of '0x00158d0004c07fa8'
mars 21 14:21:08 raspberrypi npm[1513]: zigbee2mqtt:info  2020-03-21 14:21:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0004c07fa8"}}'
mars 21 14:21:08 raspberrypi npm[1513]: zigbee2mqtt:debug 2020-03-21 14:21:08: Device '0x00158d0004c07fa8' announced itself
mars 21 14:21:08 raspberrypi npm[1513]: zigbee2mqtt:info  2020-03-21 14:21:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d0004c07fa8"}}'
mars 21 14:21:32 raspberrypi npm[1513]: zigbee2mqtt:debug 2020-03-21 14:21:32: Received Zigbee message from '0x00158d0004c07fa8', type 'readResponse', cluster 'genBasic', data '{"modelId":"SmokeSensor-N-3.0","manufacturerName":"HEIMAN"}' from endpoint 1 with groupID 0
mars 21 14:21:32 raspberrypi npm[1513]: zigbee2mqtt:debug 2020-03-21 14:21:32: No converter available for 'HS3SA' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SmokeSensor-N-3.0","manufacturerName":"HEIMAN"}'
mars 21 14:21:47 raspberrypi npm[1513]: zigbee2mqtt:error 2020-03-21 14:21:47: Failed to interview '0x00158d0004c07fa8', device has not successfully been paired
mars 21 14:21:47 raspberrypi npm[1513]: zigbee2mqtt:info  2020-03-21 14:21:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00158d0004c07fa8"}}'
mars 21 14:21:47 raspberrypi npm[1513]: zigbee2mqtt:info  2020-03-21 14:21:47: Configuring '0x00158d0004c07fa8'
mars 21 14:21:52 raspberrypi npm[1513]: zigbee2mqtt:debug 2020-03-21 14:21:52: Received Zigbee message from '0x00158d0004c07fa8', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":93}' from endpoint 1 with groupID 0
mars 21 14:21:52 raspberrypi npm[1513]: zigbee2mqtt:info  2020-03-21 14:21:52: MQTT publish: topic 'zigbee2mqtt/0x00158d0004c07fa8', payload '{"battery":46.5,"linkquality":78}'
mars 21 14:21:52 raspberrypi npm[1513]: zigbee2mqtt:info  2020-03-21 14:21:52: Successfully configured '0x00158d0004c07fa8'

dfsx1, 2020-03-21:
So if you only put the battery back in, doesn't beep right? 
Which mode is the LED indicating? (standby or fault maybe?)

To confirm hardware issue, maybe try to temporary disable/poweroff the coordinator to make sure the device is also broken in "standalone" mode without network.

dfsx1, 2020-03-21:
PS: I hope your not deaf by now :(
On the box it only says 85db>3m, but on the website also 120db<1m.
Just 3 test-beeps was already quite painful for me, as you cannot press the buttom from distance.
Please be carefull 120db is very dangerous for your ears!

dfsx1, 2020-03-21:
So, HS1SA-M and HS3SA both work with fz.heiman_smoke and fz.battery in devices.js
Only difference I see is that HS3SA is not having:
- fromZigbee[ fz.heiman_smoke_enrolled ]
- configureReporting.batteryAlarmState(endpoint);
I'm now trying to find out more info about this...

kwetnico, 2020-03-21:
> PS: j'espère que vous
> n'êtes pas sourd maintenant :( Sur la boîte, il est seulement indiqué 85db> 3m, mais sur le site Web également 120db <1m.
> Juste 3 bips de test étaient déjà assez douloureux pour moi, car vous ne pouvez pas appuyer sur le bouton à distance .
> S'il vous plaît soyez prudent 120dB est très dangereux pour vos oreilles!

yes it's true since it rings continuously I have ears that smoke :)

kwetnico, 2020-03-21:
Cela fonctionne bien pour vous sans modifier le fichier device.js ?

kwetnico, 2020-03-21:
I tested by stopping my raspberry therefore my coordinator and by pressing once on the button the Heiman still rings continuously .. I must remove the battery so that it stops ...

> Donc, si vous remettez uniquement la batterie en place, ne sonne pas correctement?
> Quel mode la LED indique-t-elle? (veille ou faute peut-être?)
> 
> Pour confirmer le problème matériel, essayez peut-être de désactiver / éteindre temporairement le coordinateur pour vous assurer que l'appareil est également cassé en mode "autonome" sans réseau.
s ...

> Donc, si vous remettez uniquement la batterie en place, ne sonne pas correctement?
> Quel mode la LED indique-t-elle? (veille ou faute peut-être?)
> 
> Pour confirmer le problème matériel, essayez peut-être de désactiver / éteindre temporairement le coordinateur pour vous assurer que l'appareil est également cassé en mode "autonome" sans réseau.

I tested by stopping my raspberry therefore my coordinator and by pressing once on the button the Heiman still rings continuously .. I must remove the battery so that it stops ...

kwetnico, 2020-03-21:
> Donc, si vous remettez uniquement la batterie en place, ne sonne pas correctement?
> Quel mode la LED indique-t-elle? (veille ou faute peut-être?)
>
> Pour confirmer le problème matériel, essayez [...]

I tested by stopping my raspberry therefore my coordinator and by pressing once on the button the Heiman still rings continuously .. I must remove the battery so that it stops ...

dfsx1, 2020-03-21:
> I tested by stopping my raspberry therefore my coordinator and by pressing once on the button the Heiman still rings continuously .. I must remove the battery so that it stops ...

And before/not pressing test-button, what is the LED status?
It should start in standby and only beep 3x after pressing the button.
If it's a faulty device I would ask seller for a refund.

> Does this work well for you without modifying the device.js file?

Yes for me the "smoke" state works fine with original device.js without modification.
Only the name HS3Sa is wrong and I'm woried about the configureReporting.batteryAlarmState() difference.

dfsx1, 2020-03-21:
https://github.com/Koenkk/zigbee2mqtt/issues/2394

kwetnico, 2020-03-21:
> > Does this work well for you without modifying the device.js file?
> 
> Yes for me the "smoke" state works fine with original device.js without modification.
> Only the name HS3SA is wrong and I'm woried about the configureReporting.batteryAlarmState() difference.
> 
> > I tested by stopping my raspberry therefore my coordinator and by pressing once on the button the Heiman still rings continuously .. I must remove the battery so that it stops ...
> 
> And before/not pressing test-button, what is the LED status?
> It should start in standby (red LED flash every 53 seconds) and only beep 3x after pressing the button.
> If it's a faulty device (red LED flash "TWICE" every 53 seconds) I would ask seller for a refund.

And before/not pressing test-button, what is the LED status? 
   On my device : Red LED flash every 53 seconds

I reset everything, I find the same problems just info battery and signal.
and error message if I remove the battery and put it back .. otherwise no communication:

mars 21 17:49:38 raspberrypi npm[1495]: zigbee2mqtt:info  2020-03-21 17:49:38: Starting interview of 'DetecteurFumee'
mars 21 17:49:38 raspberrypi npm[1495]: zigbee2mqtt:info  2020-03-21 17:49:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"DetecteurFumee"}}'
mars 21 17:49:38 raspberrypi npm[1495]: zigbee2mqtt:info  2020-03-21 17:49:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"DetecteurFumee"}}'
mars 21 17:49:39 raspberrypi npm[1495]: zigbee2mqtt:info  2020-03-21 17:49:39: MQTT publish: topic 'zigbee2mqtt/DetecteurFumee', payload '{"battery":86.5,"linkquality":81}'
mars 21 17:50:22 raspberrypi npm[1495]: zigbee2mqtt:error 2020-03-21 17:50:22: Failed to interview 'DetecteurFumee', device has not successfully been paired
mars 21 17:50:22 raspberrypi npm[1495]: zigbee2mqtt:info  2020-03-21 17:50:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"DetecteurFumee"}}'

kwetnico, 2020-03-21:
> > Est-ce que cela fonctionne bien pour vous sans modifier le fichier device.js?
> 
> Oui pour moi l'état "fumée" fonctionne très bien avec device.js d'origine sans modification.
> Seul le nom HS3SA est erroné et je m'inquiète de la différence de configureReporting.batteryAlarmState ().
> 
> > J'ai testé en arrêtant ma framboise donc mon coordinateur et en appuyant une fois sur le bouton le Heiman sonne toujours en continu .. Je dois retirer la batterie pour qu'elle s'arrête ...
> 
> Et avant / sans appuyer sur le bouton de test, quel est l'état de la LED?
> Il doit démarrer en veille (LED rouge clignote toutes les 53 secondes) et émettre un bip 3x seulement après avoir appuyé sur le bouton.
> S'il s'agit d'un appareil défectueux (flash LED rouge "DEUX FOIS" toutes les 53 secondes), je demanderais au vendeur un remboursement.

I redo the tests if I press even 1 time: 1 second or 3 seconds while holding down it does not change anything the device rings without stopping ... impossible to cut it (I let ring 5 minutes) indicates smoke: true.

I tested by lighting a candle, it works well with smoke: true (to stop it I had to remove the battery)

But it always stays with the value: true

dfsx1, 2020-03-21:
In the FAQ there is only "Power on and alarm" and "Press test button but no alarm" phenomenon.
Nothing about continuous alarm after pushing test button.

You can still try to clean dust with vacuum cleaner or compressed air.
Or check if the button hangs inside when you press it.

But I really think your device is broken and you should dispute / ask refund with the seller.

[image]

kwetnico, 2020-03-21:
> Dans la FAQ, il n'y a que le phénomène "Mise sous tension et alarme" et "Appuyez sur le bouton de test mais pas d'alarme".
> Rien sur l'alarme continue après avoir appuyé sur le bouton de test.
> 
> Vous pouvez toujours essayer de nettoyer la poussière avec un aspirateur ou de l'air comprimé.
> Ou vérifiez si le bouton se bloque à l'intérieur lorsque vous appuyez dessus.
> 
> Mais je pense vraiment que votre appareil est cassé et vous devriez contester / demander un remboursement au vendeur.
> 
> [image]

it always stays on smoke: true even after several minutes :(
mars 21 19:14:00 raspberrypi npm[9900]: zigbee2mqtt:info  2020-03-21 19:14:00: MQTT publish: topic 'zigbee2mqtt/DetecteurFumee', payload '{"battery":76.5,"linkquality":92,"smoke":true,"battery_low":false}'

I dismantled the device by removing the 4 screws and cleaned but the same ...

Koenkk, 2020-03-21:
Could you provide the zigbee herdsman debug log when pairing this device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

kwetnico, 2020-03-21:
pi@raspberrypi:~$ cd /opt
pi@raspberrypi:/opt$ dir
eff.org       pigpio  Wolfram      zigbee2mqttSave
minecraft-pi  vc      zigbee2mqtt
pi@raspberrypi:/opt$ zigbee2mqtt
-bash: zigbee2mqtt : commande introuvable
pi@raspberrypi:/opt$ cd zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt$ dir
azure-pipelines.yaml  lib                  README.md
data                  LICENSE              scripts
docker                node_modules         test
images                npm-shrinkwrap.json  update.sh
index.js              package.json
pi@raspberrypi:/opt/zigbee2mqtt$ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.12.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-21 22:06:02: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-21.22-06-02' filename: log.txt
zigbee2mqtt:info  2020-03-21 22:06:02: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-03-21 22:06:02: Starting zigbee-herdsman...
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
zigbee2mqtt:error 2020-03-21 22:06:03: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-21 22:06:03: Failed to start zigbee
zigbee2mqtt:error 2020-03-21 22:06:03: Exiting...
zigbee2mqtt:error 2020-03-21 22:06:03: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.serialPort.open (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at binding.open.then.err (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-03-21T21_06_03_455Z-debug.log
pi@raspberrypi:/opt/zigbee2mqtt$

kwetnico, 2020-03-21:
> Could you provide the zigbee herdsman debug log when pairing this device?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Before to launch with this command :
DEBUG=zigbee-herdsman* npm start

 you want to delete to database. Db and repairing my device ?
Thanks

Koenkk, 2020-03-21:
Don't delete the database
- Remove device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove
- Rejoin when running with `DEBUG=zigbee-herdsman* npm start`.

kwetnico, 2020-03-21:
> Ne supprimez pas la base de données
> 
> * Supprimer l'appareil via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove
> * Rejoin lors de l'exécution avec `DEBUG=zigbee-herdsman* npm start`.

I'm doing this tomorrow morning, thanks for the help because I'm going crazy with this smoke detect

kwetnico, 2020-03-22:
> Ne supprimez pas la base de données
> 
> * Supprimer l'appareil via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove
> * Rejoin lors de l'exécution avec `DEBUG=zigbee-herdsman* npm start`.

Hello,

I deleted this device with force_remove 👍 
mars 22 10:10:32 raspberrypi npm[23296]: zigbee2mqtt:info  2020-03-22 10:10:32: Force removing 'DetecteurFumee'
mars 22 10:10:32 raspberrypi npm[23296]: zigbee2mqtt:info  2020-03-22 10:10:32: Successfully force_removed DetecteurFumee
mars 22 10:10:32 raspberrypi npm[23296]: zigbee2mqtt:info  2020-03-22 10:10:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_force_removed","message":"DetecteurFumee"}'

I reboot my system, i enable : the debug mode in configuration.yaml and launch with DEBUG=zigbee-herdsman* npm start

ID my Device Heiman is : 0x00158d0004c07fa8 :
pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.12.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-22 10:23:15: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-22.10-23-15' filename: log.txt
zigbee2mqtt:debug 2020-03-22 10:23:15: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-03-21.19-43-56'
zigbee2mqtt:debug 2020-03-22 10:23:16: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-03-22 10:23:16: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-03-22 10:23:16: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-03-22 10:23:16: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +12ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +2ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,1,26] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,1,26] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,1] - 26 +3ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":377} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +14ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +39ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +33ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,144,21,52,1,2,0,0,0,0] - 218 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608} +22ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +19ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +18ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +16ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,21,211] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,21,211] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,21] - 211 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":21} +36ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms

dfsx1, 2020-03-22:
> No converter available for 'HS3SA' with cluster 'ssIasZone' and type 'readResponse'

Difference I can see in devices.js between HS1SA-M and HS3SA:
- fz.heiman_smoke_enrolled
- await configureReporting.batteryAlarmState(endpoint);

is not added for the HS3SA.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L5116-L5149

I think you need that fz.heiman_smoke_enrolled to fix your error.
```
heiman_smoke_enrolled: {
        cluster: 'ssIasZone',
        type: ['attributeReport', 'readResponse'],
```
---

Actually I'm wondering why the HS3SA was added seperately in first place (related|stale: #961)?

In https://github.com/Koenkk/zigbee-herdsman-converters/commit/2020dc4048e55d1e5e83056834df63f6ae866939 @Koenkk added 'SmokeSensor-N-3.0' zigbeeModel for HS3SA instead of HS1SA(-N).
The ticket https://github.com/Koenkk/zigbee2mqtt/issues/2394 which that commit refers to was however originally about HS1SA(-N).

In home-assistant.js they both have:
- 'HS1SA-M': [cfg.binary_sensor_smoke, cfg.binary_sensor_battery_low],
- 'HS3SA': [cfg.binary_sensor_smoke, cfg.binary_sensor_battery_low],

Isn't configureReporting.batteryAlarmState() also required for both (related to battery_low)?

kwetnico, 2020-03-22:
I even tried to reflash my CC2531 key and suddenly re-pair all the deviations, I have the same problems. He is now paired with the live coordinator and same.

Koenkk, 2020-03-23:
The issue is that the device is not enrolled correctly.

I've pushed some improvements to the latest zigbee2mqtt dev branch, please try to repair it there.

kwetnico, 2020-03-23:
> The issue is that the device is not enrolled correctly.
> 
> I've pushed some improvements to the latest zigbee2mqtt dev branch, please try to repair it there.

Hello,

Ok so I have to re-delete the device, install the Dev version (can we switch back to the stable version after?) And reappear the device?

Koenkk, 2020-03-23:
Yes, and you can switch to stable afterwards

kwetnico, 2020-03-23:
> Oui, et vous pouvez ensuite passer à stable

Isn't write Dev Version or Master version ? 

 Started zigbee2mqtt.
mars 23 12:37:26 raspberrypi npm[3063]: > zigbee2mqtt@1.12.0 start /opt/zigbee2mqtt
mars 23 12:37:26 raspberrypi npm[3063]: > node index.js
mars 23 12:37:29 raspberrypi npm[3063]: zigbee2mqtt:info  2020-03-23 12:37:29: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-23.12-37-29' filename: log.txt
mars 23 12:37:30 raspberrypi npm[3063]: zigbee2mqtt:info  2020-03-23 12:37:30: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
mars 23 12:37:30 raspberrypi npm[3063]: zigbee2mqtt:info  2020-03-23 12:37:30: Starting zigbee-herdsman...
mars 23 12:37:32 raspberrypi npm[3063]: zigbee2mqtt:info  2020-03-23 12:37:32: zigbee-herdsman started
mars 23 12:37:32 raspberrypi npm[3063]: zigbee2mqtt:info  2020-03-23 12:37:32: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
mars 23 12:37:32 raspberrypi npm[3063]: zigbee2mqtt:info  2020-03-23 12:37:32: Currently 23 devices are joined:

kwetnico, 2020-03-23:
And to go back to the master version should I use this command?
git checkout master 

?

kwetnico, 2020-03-23:
So 

It's ok for you now :
 zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":24195,"extaddr":"0x00158d0004c07fa8","parentaddr":41417} +12s
  zigbee-herdsman:controller:log Device '0x00158d0004c07fa8' joined +12s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:controller:log Device '0x00158d0004c07fa8' accepted by handler +2ms
  zigbee-herdsman:controller:log New device '0x00158d0004c07fa8' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x00158d0004c07fa8' +0ms
zigbee2mqtt:info  2020-03-23 12:46:31: Device '0x00158d0004c07fa8' joined
zigbee2mqtt:info  2020-03-23 12:46:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0004c07fa8"}}'
  zigbee-herdsman:controller:log Interview '0x00158d0004c07fa8' start +72ms
zigbee2mqtt:info  2020-03-23 12:46:31: Starting interview of '0x00158d0004c07fa8'
zigbee2mqtt:info  2020-03-23 12:46:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0004c07fa8"}}'
  zigbee-herdsman:controller:device Interview - start device '0x00158d0004c07fa8' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":24195,"nwkaddrofinterest":24195} +12s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,131,94,131,94,35] +12s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,131,94,168,127,192,4,0,141,21,0,201,161,189] +85ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,131,94,168,127,192,4,0,141,21,0,201,161,189] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [131,94,168,127,192,4,0,141,21,0,201,161] - 189 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":24195,"extaddr":"0x00158d0004c07fa8","parentaddr":41417} +90ms
  zigbee-herdsman:controller:log Device '0x00158d0004c07fa8' joined +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x00158d0004c07fa8' accepted by handler +1ms
  zigbee-herdsman:controller:log Not interviewing '0x00158d0004c07fa8', completed 'false', in progress 'true' +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +12s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,131,94,131,94,168,127,192,4,0,141,21,0,128,130] +141ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,131,94,131,94,168,127,192,4,0,141,21,0,128,130] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [131,94,131,94,168,127,192,4,0,141,21,0,128] - 130 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":24195,"nwkaddr":24195,"ieeeaddr":"0x00158d0004c07fa8","capabilities":128} +154ms
  zigbee-herdsman:controller:log Device announce '0x00158d0004c07fa8' +153ms
zigbee2mqtt:debug 2020-03-23 12:46:31: Device '0x00158d0004c07fa8' announced itself
zigbee2mqtt:info  2020-03-23 12:46:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d0004c07fa8"}}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,131,94,1,201,161,48] +187ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,131,94,1,201,161,48] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [131,94,1,201,161] - 48 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":24195,"relaycount":1,"relaylist":[41417]} +191ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x00158d0004c07fa8', attempt 1 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":24195,"nwkaddrofinterest":24195} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,131,94,131,94,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,131,94,1,201,161,48] +999ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,131,94,1,201,161,48] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [131,94,1,201,161] - 48 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":24195,"relaycount":1,"relaylist":[41417]} +11s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,18,69,130,131,94,0,131,94,2,64,128,11,18,127,100,0,0,44,100,0,0,93] +18ms
[... cut ...]
  zigbee-herdsman:controller:endpoint Read 0x00158d0004c07fa8/1 genBasic(["powerSource","zclVersion"], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms

kwetnico, 2020-03-23:
> Oui, et vous pouvez ensuite passer à stable

I can't dev version :
 
pi@raspberrypi:/opt/zigbee2mqtt $ git checkout dev
error: Vos modifications locales aux fichiers suivants seraient écrasées par l'extraction :
        package.json
Veuillez valider ou remiser vos modifications avant de basculer de branche.
Abandon

kwetnico, 2020-03-23:
Version Dev :

zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,5,98,168,127,192,4,0,141,21,0,201,161,7] +28ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,5,98,168,127,192,4,0,141,21,0,201,161,7] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [5,98,168,127,192,4,0,141,21,0,201,161] - 7 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":25093,"extaddr":"0x00158d0004c07fa8","parentaddr":41417} +28ms
  zigbee-herdsman:controller:log Device '0x00158d0004c07fa8' joined +23s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:controller:log Device '0x00158d0004c07fa8' accepted by handler +1ms
  zigbee-herdsman:controller:log New device '0x00158d0004c07fa8' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x00158d0004c07fa8' +0ms
zigbee2mqtt:info  2020-03-23 13:41:37: Device '0x00158d0004c07fa8' joined
zigbee2mqtt:info  2020-03-23 13:41:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0004c07fa8"}}'
  zigbee-herdsman:controller:log Interview '0x00158d0004c07fa8' start +47ms
zigbee2mqtt:info  2020-03-23 13:41:37: Starting interview of '0x00158d0004c07fa8'
zigbee2mqtt:info  2020-03-23 13:41:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0004c07fa8"}}'
  zigbee-herdsman:controller:device:log Interview - start device '0x00158d0004c07fa8' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":25093,"nwkaddrofinterest":25093} +23s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,5,98,5,98,35] +23s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,5,98,168,127,192,4,0,141,21,0,201,161,7] +62ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,5,98,168,127,192,4,0,141,21,0,201,161,7] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [5,98,168,127,192,4,0,141,21,0,201,161] - 7 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":25093,"extaddr":"0x00158d0004c07fa8","parentaddr":41417} +67ms
  zigbee-herdsman:controller:log Device '0x00158d0004c07fa8' joined +17ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x00158d0004c07fa8' accepted by handler +2ms
  zigbee-herdsman:controller:log Not interviewing '0x00158d0004c07fa8', completed 'false', in progress 'true' +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +23s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,5,98,5,98,168,127,192,4,0,141,21,0,128,130] +260ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,5,98,5,98,168,127,192,4,0,141,21,0,128,130] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [5,98,5,98,168,127,192,4,0,141,21,0,128] - 130 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":25093,"nwkaddr":25093,"ieeeaddr":"0x00158d0004c07fa8","capabilities":128} +274ms
  zigbee-herdsman:controller:log Device announce '0x00158d0004c07fa8' +273ms
zigbee2mqtt:debug 2020-03-23 13:41:37: Device '0x00158d0004c07fa8' announced itself
zigbee2mqtt:info  2020-03-23 13:41:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d0004c07fa8"}}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,5,98,1,201,161,138] +141ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,5,98,1,201,161,138] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [5,98,1,201,161] - 138 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":25093,"relaycount":1,"relaylist":[41417]} +146ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,5,98,1,201,161,138] +599ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,5,98,1,201,161,138] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [5,98,1,201,161] - 138 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":25093,"relaycount":1,"relaylist":[41417]} +602ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,18,69,130,5,98,0,5,98,2,64,128,11,18,127,100,0,0,44,100,0,0,93] +24ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,18,69,130,5,98,0,5,98,2,64,128,11,18,127,100,0,0,44,100,0,0,93] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 18 - 2 - 5 - 130 - [5,98,0,5,98,2,64,128,11,18,127,100,0,0,44,100,0,0] - 93 +0ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,8,206] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,8,206] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,8] - 206 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":8} +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,5,98,

kwetnico, 2020-03-25:
I only have a message in the zigbeemqtt log if I remove and put the battery back into the smoke detector:
March 25 11:56:31 raspberrypi npm [1503]: zigbee2mqtt: info 2020-03-25 11:56:31: MQTT publish: topic 'zigbee2mqtt / DetecteurFumee', payload '{"battery": 90, "linkquality": 84} '

By pressing the no ascent button and pressing the small hole with a needle for the same test.

nukusinji, 2020-05-22:
I have same issue.. Just reporting battery and linkquality..

steve41200, 2020-07-04:
I have the same issue.
Zigbeetomqtt 1.14
I see the 3 entity (smoke, battery, linkquality) the integration seems good, but hen i test via the button or with a real smoke, the detector ring, but the smoke line stay at "RAS"
it is write Smoke detector (HS3SA)
par HEIMAN in Home assistant
The model on the battery is HS1SA-E

I don't have try the tips to rename the.js . I don't know where find it

kwetnico, 2020-07-05:
For me this model don't works with Zigbee2mqtt..

I have the xiaomi 's device no problem

nukusinji, 2020-07-06:
> For me this model don't works with Zigbee2mqtt..
> 
> I have the xiaomi 's device no problem

https://github.com/Koenkk/zigbee2mqtt/issues/3773#issuecomment-648019661
Here is same issue.

The problem is solved.

You need install latest version and than repairing the device.

kwetnico, 2020-07-06:
> > For me this model don't works with Zigbee2mqtt..
> > I have the xiaomi 's device no problem
> 
> [#3773 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/3773#issuecomment-648019661)
> Here is same issue.
> 
> The problem is solved.
> 
> You need install latest version and than repairing the device.

Oh Thanks i waiting the new Master version.

closed 2020-08-16 by github-actions[bot]