Koenkk/zigbee2mqtt #3165: White channel of Gledopto GL-S-003Z

gjscarlett, 2020-03-20:
# Bug Report
## What happened
The Gl-S-003Z has separate RGB and white channels, however once paired to zigbee2mqtt only the RBG is controllable. Hoping this is an easy fix, I've had a go myself editing the devices.js but haven't made any progress. 
Logs say its failed to set white_value, and no converter is available.

The bulbs pair fine, and identify as GL-S-003Z, I'm wondering if there are different versions of the bulb with the same model number?

## Debug Info
zigbee2mqtt version: 1.8.0-dev (commit #2451d536f2c9e446d396df55c9493f256ab9352e)
CC253X firmware version: 20190608
Log:
`
zigbee2mqtt:debug 2020-03-20 18:22:17: Publishing 'set' 'white_value' to 'bathroom door'
zigbee2mqtt:error 2020-03-20 18:22:17: Publish 'set' 'white_value' to 'bathroom door' failed: 'TypeError: Cannot read property 'state' of undefined'
zigbee2mqtt:debug 2020-03-20 18:22:17: TypeError: Cannot read property 'state' of undefined
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1043:39)
`

`zigbee2mqtt:error 2020-03-20 18:35:29: No converter available for 'white_value' (225)`

Koenkk, 2020-03-21:
Before posting issues, first make sure it still appears in the latest zigbee2mqtt release (1.12.0) you are on 1.8.0 currently.

gjscarlett, 2020-03-22:
My bad, didn't realise the edge version had to be updated by uninstalling and reinstalling!

On the latest version now, and the white LEDs work by setting color_temperature. Home Assistant is showing the option to set white value but not color temp at the moment.

dfsx1, 2020-03-22:
According to this chart they have 2 versions of the GU10 RGBW.
Difference is 1 or 2 SSIDs (however I don't know if both have same modelnumber GL-S-003Z).
You should be able to tell based on 1 or 2 tints of the yellow/orange LEDs.

GL-S-003Z is currently configured for only 1 white color in Zigbee2MQTT, that's why color_temperature is not available.

[image]

RGBW (1 SSID):
[image]

RGBWW/CT (2 SSIDs):
[image]

RGB+CCT (GL-S-007Z):
[image]

Koenkk, 2020-03-22:
@gjscarlett can you share your `database.db` file?

sprockers, 2020-04-30:
I have the same RGBW (warm white, 1 SSID) version of the GL-S-003Z
I too have the same issue - home assistant only allows you to play with the RGB LEDs, and the White slider has no effect.

I experimented a little with the use of color_temp via mqtt (payload of {"color_temp":number}, and found that the white LEDs turn on very dimly at 160, and appear to get to max brightness at 500. 

The database.db entry for one of these bulbs looks like this:
{"id":35,"type":"Router","ieeeAddr":"0x00124b001f248f0d","nwkAddr":27986,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-S-003Z","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-S-003Z","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.1"}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.1","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1588277806941}

Will that help to debug things @Koenkk ?

closed 2020-07-10 by stale[bot]