Koenkk/zigbee2mqtt #3166: Zigbee2MQTT keeps restarting if stick is not found

bartplessers, 2020-03-20:
# Bug Report
## What happened
Zigbee2MQTT keeps restarting if stick is not found

```
zigbee2mqtt:error 2020-03-20 21:48:17: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-20 21:48:17: Failed to start zigbee
zigbee2mqtt:error 2020-03-20 21:48:17: Exiting...
zigbee2mqtt:error 2020-03-20 21:48:17: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
```

## What did you expect to happen
no attempt to start again after x (like 3) failures

Koenkk, 2020-03-20:
Is this with Docker and the `--restart always` flag?

bartplessers, 2020-03-20:
Its 
- rpi, buster
- docker
- hass.io in docker
- zigbee2mqtt addon in hass.io

I didn't configure anything in docker, all by GUI in hass.io

Koenkk, 2020-03-21:
Zigbee2mqtt has no knowledge on how many times it has been started, I feel it's the responsibility of the supervisor to stop restarting a service once it fails to start after X times.

bartplessers, 2020-03-23:
ok, fair enough

closed 2020-03-23 by bartplessers