Koenkk/zigbee2mqtt #3168: Is LQI data really necessary? What is your network range coverage?

xefil, 2020-03-21:
# Bug Report
## What happened

Network area coverage difficult.

## What did you expect to happen

Area at least covered on a room

## How to reproduce it (minimal and precise)

Read below

## Debug Info
zigbee2mqtt version: zigbee2mqtt version 1.12.0-dev (commit #9b80b67)
CC253X firmware version:
1x CC2531 (+ external antenna): Coordinator Source Routing (CC2531_SOURCE_ROUTING_20190619)
1x CC2531 (+ external antenna): (CC2531_router_2019_02)
1x CC2530+CC2591: ptvinfo cc2530_1.2.2a.44539_firmware)

Hi all,

As largely discussed on this thread/issue https://github.com/Koenkk/zigbee2mqtt/issues/2802 I'm having problems to get all works properly.
My setup actually is the follow:
- Zigbee2mqtt-dev on docker
- Coordinator in another room via raspberry cabled (no wifi) with ser2net
- 1x router cc2531
- 1x router cc2530+cc2591
Channel 11
2x Wifi networks at home (1x 2,4Ghx and 1x5Ghz with auto-assignment, actually on Channel 1)

Distances (in this test setup)
Coordinator and cc2531: 20cm distance and LQI OK
Coordinator and cc2530+cc2591: 4mt distance in free space, same room, no walls. `Failed to execute LQI `

So, is LQI a needed information to let all work properly?

Is it normal that the CC2530 is not talking to CC2531 with a distance of only 3-4meters?

Is there best practices to place routers/antennas in terms of orientation or placement (high, low on the floor)?

Does someone have suggestions about setup?

Thanks a lot guys, this is driven me really crazy!
I've tested a LOT of setups and having 20 Xiaomi sensors that should be installed in the next weeks, I'm in difficult.
I could even think to increase routers if the above is a range coverage expected.

Thanks!

Simon

sv87411, 2020-03-24:
I've said this a lot on here and on the Z2M Discourse forum, but I truly think the CC2531 is too under powered to be a coordinator. I almost gave up with Z2M because using a CC2531 was unreliable with devices constantly dropping off.

I've moved to a CC25x2R1 board (or a CC1352P, I've tried both) and since moving I now know if I have signal problems it's probably a battery powered endpoint device or a WiFi interference problem. My Zigbee network is now rock solid and my bulb/socket router devices just work and work.

I guess it could also have just been moving my WiFi channel away from my Zigbee channel that fixed the problems (yeah I know, I changed too many things at once), but it's now nice to know for sure that my coordinator isn't the problem. 

Stuart

I also changed my WiFi setup to ensure that my 2.4Ghz WiFi was at the opposite end of the channel range from my Zigbee.

Shred99, 2020-03-25:
I'm running a CC2531 coordinator + a CC2530 router.  The router is mounted in the roof cavity of the house for better range.  I have 18 of the Xiaomi battery powered sensors.  Three door switch sensors are door switches being used to monitor gates (in waterproof boxes) and one door switch is om the door of an outside shed.  The four remote door switch sensors are 20-30 metres from the router and coordinator.

Like many people, I suffered intermittent stability problems for a long time.  Usually a Xiaomi sensor would work perfectly for days or weeks, then inexplicably drop off the network and stop sending heartbeat signals.  Re-pairing, or just bringing the sensor close to the coordinator and activating it would fix the problem... until next time.  The new "herdsman" firmware helped a lot in fixing this problem.  It seems to be more persistent in talking to a sensor that has marginal signal strenght. 
 Replacing the junk OEM antenna on the coordinator with a quality 9dB gain aerial gave the system a big lift.  Two weeks ago, I finally upgraded the router aerial as well.  This has made the system rock solid since that time.

I don't find the link quality indicator useful at all and just ignore it now.  The Xiami sensors transmit a heartbeat message roughly once every 50 minutes.  I log the output from the MQTT server to MySQL database.  I now have a SQL query that shows a summary, grouping the output from each device by hour, then showing the sum of all hourly messages logged.  At any time, each ZigBee device should show 24 "heartbeats received" when I run my query.  If the number of received heartbeat messages is fewer than 24, I know that the sensor in question is struggling to be heard.  After two weeks with both the router and coordinator on the 9dB aerials, I've lost one single heartbeat message in all that time.  I'm very happy with my setup now.

This tends to suggest that the cheap CC253x series devices have enough memory and processor power for the job, but maybe they're not so great on the radio side of things.  I should point out that I live in a semi-rural area, so I have practically no interference from neighbours WiFi or ZigBee devices.  I think this improves range significantly.

xefil, 2020-03-25:
Hello both @sv87411 and @Shred99 ,
Thanks for sharing the infos. Even if LQI is not so important, I've ended that a CC2530 router cannot loose the LQI informations asked by the CC2531+external antenna coordinator, which is only 20cm away. Then if it's 10cm it works... So I fear that I'm having a hardware issue. I'm not proof on hardware, I was thinking that such deviceit works or it doesn't work at all. BTW, having a second CC2531 which works well, I fear I'm having the CC2530+CC2591 that works very bad. I've no idea how to test a sort of PING between devices... I'm asking for replacement. :-(
Simon

Shred99, 2020-03-26:
You might want to try using one module or the other as a coordinator without the added complication of having a router - just temporarily as a process of elimination.  You might find one works well and the other does not.  Your network should work reliably over a short distance without having a router.

Some of the eBay / Aliexpress modules are pretty dodgy too.  If your module has been 'hacked' to have an external antenna connector, beware.  Some of these mods actually make the range much worse than with a module using a well designed and well implemented PCB antenna.

xefil, 2020-03-26:
Ok @Shred99 thanks for the hints.
Meanwhile I've got the OK to a board replacement, so as soon I get it (coronavirus delays apart), I can test both.
I wish you a nice day!
Simon

closed 2020-06-02 by stale[bot]