Koenkk/zigbee2mqtt #3169: Xiaomi QBKG04LM

seblo0505, 2020-03-21:
Hello,  I Have succesfully binded some xiaomi parts, also an Xiaomi QBKG04LM. The Switch is working but i cant publish thje ON command to set the switch ON. i tried it with zigbee2mqtt/friendly_name/system/set as topic and the payload 1, On. 

How can I publish the switch to ON?

when I click there is no error:

Mar 21 17:34:18 loxberry npm[29612]: zigbee2mqtt:info  2020-03-21 17:34:18: MQTT publish: topic 'zigbee2mqtt/Schalter_1', payload '{"state":"ON","linkquality":86,"click":"single","action":"release"}'
Mar 21 17:34:18 loxberry npm[29612]: zigbee2mqtt:info  2020-03-21 17:34:18: MQTT publish: topic 'zigbee2mqtt/Schalter_1', payload '{"state":"OFF","linkquality":89}'
Mar 21 17:34:18 loxberry npm[29612]: zigbee2mqtt:info  2020-03-21 17:34:18: MQTT publish: topic 'zigbee2mqtt/Schalter_1', payload '{"state":"OFF","linkquality":86,"click":"single","action":"release"}'
Mar 21 17:35:15 loxberry npm[29612]: zigbee2mqtt:error 2020-03-21 17:35:15: Publish 'set' 'state' to 'Schalter_1' failed: 'Error: Cluster 'genOnOff' has no command '{
Mar 21 17:35:15 loxberry npm[29612]:   "operation_mode": {
Mar 21 17:35:15 loxberry npm[29612]:     "button": "single"|"left"|"right",
Mar 21 17:35:15 loxberry npm[29612]:     "state": "on"
Mar 21 17:35:15 loxberry npm[29612]:   }
Mar 21 17:35:15 loxberry npm[29612]: }''

Best regards

Koenkk, 2020-03-21:
Try with `zigbee2mqtt/friendly_name/set` `{"state": "OFF"}`

seblo0505, 2020-03-22:
It worked, thank you

closed 2020-03-22 by seblo0505