Koenkk/zigbee2mqtt #3173: Issue with one Sengled E11-G13 Bulb

Anon0ne, 2020-03-22:
# Bug Report
## What happened
After moving to docker, zigbee2mqtt keeps losing connections, or "mac no ack" & no network route. Same setup from raspbian not in a docker container, worked fine. I have reset the system twice now, and this one type of bulb is still giving trouble.

## What did you expect to happen

Normal on/off of the light

## How to reproduce it (minimal and precise)
Run zigbee2mqtt in docker, perform a full reset of the Home Assistant, mosquitto, and zigbee2mqtt docker containers and you will lose connection to the E11-G13 Sengled Classic LEDs. Sometimes forever, sometimes for several hours.
## Debug Info
zigbee2mqtt version: latest docker
CC253X firmware version: latest, reflashed yesterday on 1x coordinator and 2x router

```
zigbee2mqtt:debug 2020-03-22 00:27:26: Received MQTT message on 'zigbee2mqtt/0xb0ce181403106a5e/set' with data '{"state": "OFF"}'

zigbee2mqtt:debug 2020-03-22 00:27:26: Publishing 'set' 'state' to '0xb0ce181403106a5e'

zigbee2mqtt:debug 2020-03-22 00:27:29: Received MQTT message on 'zigbee2mqtt/0xb0ce181403106a5e/set' with data '{"state": "OFF"}'

zigbee2mqtt:debug 2020-03-22 00:27:29: Publishing 'set' 'state' to '0xb0ce181403106a5e'

zigbee2mqtt:error 2020-03-22 00:27:30: Publish 'set' 'state' to '0xb0ce181403106a5e' failed: 'Error: Command 0xb0ce181403106a5e/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'

zigbee2mqtt:debug 2020-03-22 00:27:30: Error: Command 0xb0ce181403106a5e/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))

    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)

    at Generator.throw (<anonymous>)

    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)

zigbee2mqtt:info  2020-03-22 00:27:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0xb0ce181403106a5e' failed: 'Error: Command 0xb0ce181403106a5e/1 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0xb0ce181403106a5e"}}'

zigbee2mqtt:error 2020-03-22 00:27:34: Publish 'set' 'state' to '0xb0ce181403106a5e' failed: 'Error: Command 0xb0ce181403106a5e/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'

zigbee2mqtt:debug 2020-03-22 00:27:34: Error: Command 0xb0ce181403106a5e/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))

    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)

    at Generator.throw (<anonymous>)

    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)

zigbee2mqtt:info  2020-03-22 00:27:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0xb0ce181403106a5e' failed: 'Error: Command 0xb0ce181403106a5e/1 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0xb0ce181403106a5e"}}'
```

Koenkk, 2020-03-22:
Can you try with the source routing firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Anon0ne, 2020-03-22:
> Can you try with the source routing firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

So you want me to reflash with router source firmware the router closest to the bulb? Is there a big difference between the two? I didn't see anything about it on the git page or documents.

I will try it

Anon0ne, 2020-03-23:
Can that be flashed with serialboot?

Anon0ne, 2020-03-23:
That router didn't join until I put the router firmware back on. This time I put diag-usb version and got this information when it finally joined the network:

```
zigbee2mqtt:info  2020-03-22 22:38:38: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f1b1d1', payload '{"battery":91,"voltage":2985,"contact":true,"linkquality":39}'
,zigbee2mqtt:info  2020-03-22 22:38:38: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f1b1d1', payload '{"battery":91,"voltage":2985,"contact":false,"linkquality":39}'
,zigbee2mqtt:info  2020-03-22 22:43:49: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f1b1d1', payload '{"battery":91,"voltage":2985,"contact":true,"linkquality":39}'
,zigbee2mqtt:info  2020-03-22 22:46:45: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
,zigbee2mqtt:info  2020-03-22 22:46:45: Disconnecting from MQTT server
,zigbee2mqtt:info  2020-03-22 22:46:46: zigbee-herdsman stopped
,[..................] \ : info lifecycle zigbee2mqtt@1.12.0~start: zigbee2mqtt@1

Using '/app/data' as data directory
,
,> zigbee2mqtt@1.12.0 start /app
,> node index.js
,
,zigbee2mqtt:info  2020-03-22 22:49:07: Logging to console and directory: '/app/data/log/2020-03-22.22-49-06' filename: log.txt
,zigbee2mqtt:info  2020-03-22 22:49:07: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
,zigbee2mqtt:info  2020-03-22 22:49:07: Starting zigbee-herdsman...
,zigbee2mqtt:info  2020-03-22 22:49:09: zigbee-herdsman started
,zigbee2mqtt:info  2020-03-22 22:49:09: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
,zigbee2mqtt:info  2020-03-22 22:49:09: Currently 5 devices are joined:
,zigbee2mqtt:info  2020-03-22 22:49:09: 0x00158d00040aa591 (0x00158d00040aa591): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
,zigbee2mqtt:info  2020-03-22 22:49:09: 0x00158d000423203c (0x00158d000423203c): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
,zigbee2mqtt:info  2020-03-22 22:49:09: 0xb0ce181403106a5e (0xb0ce181403106a5e): E11-G13 - Sengled Element Classic (A19) (EndDevice)
,zigbee2mqtt:info  2020-03-22 22:49:09: 0xb0ce1814034c2925 (0xb0ce1814034c2925): E11-G13 - Sengled Element Classic (A19) (EndDevice)
,zigbee2mqtt:info  2020-03-22 22:49:09: 0x00158d0003f1b1d1 (0x00158d0003f1b1d1): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
,zigbee2mqtt:warn  2020-03-22 22:49:09: `permit_join` set to  `true` in configuration.yaml.
,zigbee2mqtt:warn  2020-03-22 22:49:09: Allowing new devices to join.
,zigbee2mqtt:warn  2020-03-22 22:49:09: Set `permit_join` to `false` once you joined all devices.
,zigbee2mqtt:info  2020-03-22 22:49:09: Zigbee: allowing new devices to join.
,zigbee2mqtt:info  2020-03-22 22:49:09: Connecting to MQTT server at mqtt://192.168.1.128
,zigbee2mqtt:info  2020-03-22 22:49:10: Connected to MQTT server
,zigbee2mqtt:info  2020-03-22 22:49:10: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
,zigbee2mqtt:info  2020-03-22 22:49:10: MQTT publish: topic 'zigbee2mqtt/0x00158d00040aa591', payload '{"battery":100,"voltage":3005,"contact":true,"linkquality":47}'
,zigbee2mqtt:info  2020-03-22 22:49:10: MQTT publish: topic 'zigbee2mqtt/0x00158d000423203c', payload '{"battery":100,"voltage":3015,"contact":true,"linkquality":81}'
,zigbee2mqtt:info  2020-03-22 22:49:10: MQTT publish: topic 'zigbee2mqtt/0xb0ce181403106a5e', payload '{"state":"OFF"}'
,zigbee2mqtt:info  2020-03-22 22:49:10: MQTT publish: topic 'zigbee2mqtt/0xb0ce1814034c2925', payload '{"state":"ON"}'
,zigbee2mqtt:info  2020-03-22 22:49:10: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f1b1d1', payload '{"battery":91,"voltage":2985,"contact":true,"linkquality":39}'
,zigbee2mqtt:info  2020-03-22 22:49:10: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.0","commit":"840b9d9","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
,zigbee2mqtt:info  2020-03-22 22:49:10: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d00040aa591/contact/config', payload '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","state_topic":"zigbee2mqtt/0x00158d00040aa591","json_attributes_topic":"zigbee2mqtt/0x00158d00040aa591","name":"0x00158d00040aa591_contact","unique_id":"0x00158d00040aa591_contact_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00040aa591"],"name":"0x00158d00040aa591","sw_version":"Zigbee2mqtt 1.12.0","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
,zigbee2mqtt:info  2020-03-22 22:49:10: MQTT publish: topic 'homeassistant/sensor/0x00158d00040aa591/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d00040aa591","json_attributes_topic":"zigbee2mqtt/0x00158d00040aa591","name":"0x00158d00040aa591_battery","unique_id":"0x00158d00040aa591_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00040aa591"],"name":"0x00158d00040aa591","sw_version":"Zigbee2mqtt 1.12.0","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
[... cut ...]
,zigbee2mqtt:debug 2020-03-22 23:08:35: Received Zigbee message from '0x00124b0018e864e7', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"34468/0xB0CE181403106A5E","inactiveText":"CHILD_RFD_RXI","presentValue":96,"relinquishDefault":2,"minimumOffTime":1}' from endpoint 8 with groupID 0
,zigbee2mqtt:info  2020-03-22 23:08:35: MQTT publish: topic 'zigbee2mqtt/0x00124b0018e864e7', payload '{"state":true,"led_state":false,"linkquality":115,"description":"34468/0xB0CE181403106A5E","type":"CHILD_RFD_RXI","rssi":96}'
,
```

Koenkk, 2020-03-23:
Sorry, I meant flashing it on the coordinator, not the router.
If you flash it on a router the router won't indeed work anymore as this is a coordinator firmware.

closed 2020-05-30 by stale[bot]