Koenkk/zigbee2mqtt #3175: friendly_name including # char causes disconnects from mqtt server

JasperAlgra, 2020-03-22:
# Bug Report
## What happened
I found out that using hashtags in the `friendly_name` of device in configuration.yaml causes strange behaviour of zigbee2mqtt including dis/reconnect to the MQTT server.

Example from configuration.yaml
```
  '0x000d6ffffe19412f':
#    friendly_name: 'gu10 dimable white spectrum #1'
    friendly_name: '0x000d6ffffe19412f'
```

In my case I've got a bulb and a remote. When I push a button on the remote zigbee2mqtt breaks and resets the connection to the MQTT server (maybe more, I cannot see in the debug log).

## What did you expect to happen

## How to reproduce it (minimal and precise)
add a `friendly_name` with a hashtag in it. Press a button on the remote or trigger an action through Home Assistant.

## Debug Info
zigbee2mqtt version: 1.12.0 (commit #840b9d9
CC253X firmware version: default

# Logs:
With errors:
https://hastebin.com/sutavuxiba.pl
Without errors:
https://hastebin.com/owihifojot.apache

Koenkk, 2020-03-23:
MQTT doesn't allow wildcards (`+`/`#`) to be used in topic names.

> Publishers are not allowed to use the wildcard characters in their topic names.

Source: https://www.eclipse.org/paho/files/mqttdoc/MQTTClient/html/wildcard.html

Latest zigbee2mqtt dev refuses to start when this is set. Thanks!

closed 2020-03-23 by Koenkk

Girbie, 2020-08-09:
@JasperAlgra : I'm facing the same issue now. How did you resolve it? zigbee2mqtt doesn't start anymore and I can't access/change the device name.

aro-git, 2020-10-18:
I'm facing the same issue right now. I named my device with fancy name with UTF-8 characters and slash (czujnik otwarcia / zamknięcia 3b - drzwi wiatrołap) and now I can't rollback it. I can't rename device and I can't access/change the device name (503: Service Unaviable).

Koenkk, 2020-10-18:
@aro-git I've tried and it works fine in my case:

[image]

You can always stop zigbee2qmtt, change the name in the configuration.yaml.

aro-git, 2020-10-18:
Many thanks for instant reply.
I tried and changed this particular friendly name in devices.yaml file.
I very appreciate your fine work.
Greetings.