Koenkk/zigbee2mqtt #3177: Joining new devices fails after a few days of uptime

sjorge, 2020-03-22:
# Bug Report
## What happened
I tried to join a new device, it failed.

The network/coordinator usb has been up for a about 1-1.5days now, joining new devices fail.
Physically unplugging and replugging the usb (with z2m stopped ofcourse) re allows joining of new devices.

I did a capture of one of those failed join requests:
https://pkg.blackdot.be/extras/zigbee/failed_join.pcapng

- I see z2m send the allow join broadcast
- I see the nearby bulb (trådfri in this case) send the 'update device' message from the new innr bulb

Not sure where it fails afterwards, it looks like z2m doesn't respond to the update device at all?

## What did you expect to happen
The device to be able to join.

## How to reproduce it (minimal and precise)
I can always reproduce this after a about 1-3 days after plugging in/out the USB device (rebooting the server does not help)

## Debug Info
zigbee2mqtt version: `1.12.0-dev (commit #d52d520)`
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}`

sjorge, 2020-03-22:
Last time I did not have issues was around 1.6 I think.

sjorge, 2020-03-22:
Soft resetting seems to work as well! However no more message after that until I restart z2m.

klorydryk, 2020-03-23:
@sjorge I can't join new devices either. How can I soft reset?

sjorge, 2020-03-23:
@klorydryk https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigreset

Or physically removing the USB and plugging it back in. (while z2m is stopped)

After te rest via MQTT, I had to restart z2m anyway. But that might be because I am running on illumos and not linux. At least it saved a phsyical trip down to the pantry.

klorydryk, 2020-03-23:
It works after a soft reset (physically ;) )
thanks.

Koenkk, 2020-03-23:
I've investigated and it seems to go wrong at the Transport key part.

Log of successful join:
[image]

Log of failed join (from OP), no transport key is done:
[image]

One thing I noticed is that the device joins via a router of the network, not the coordinator itself (Extended source is an ember stack device):
[image]

When it will join via the coordinator you will see that the extended source is a texas instrument device, like this:
[image]

- What kind of device is: `00:0d:6f:ff:fe:8e:8d:4f`? (you should be able to find that ieeeAddr in your database)
- When powering off this device, does the device join succesfully?

sjorge, 2020-03-23:
```
{"id":2,"type":"Router","ieeeAddr":"0x000d6ffffe8e8d4f","nwkAddr":47816,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI signal repeater","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,3,9,2821,4096,64636],"outClusterList":[25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI signal repeater","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190318","swBuildId":"2.2.005"}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190318","swBuildId":"2.2.005","zclVersion":3,"interviewCompleted":true,"meta":{"reporting":1,"configured":2},"lastSeen":1584920505606}
```

It's an ikea signal repeater, also the one that occasionally stops responding to pings.

I tried again with it unplugged, same result but it used: ```{"id":3,"type":"Router","ieeeAddr":"0x000d6ffffe197fe9","nwkAddr":36186,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 CWS opal 600lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":512,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 CWS opal 600lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009"}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"currentX":29969,"currentY":26869,"colorMode":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1584920594485}```

sjorge, 2020-03-23:
When I unscrew that one, the coordinator is out of range.

Koenkk, 2020-03-23:
What if you pair the device in range of the coordinator?

sjorge, 2020-03-23:
I can try that tomorrow, but I wont be able to sniff the traffic at that location.

sjorge, 2020-03-24:
@Koenkk might be off interest, I was looking omething up in database.db and noticed this:
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001938a7e5","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}
{"id":9,"type":"Coordinator","ieeeAddr":"0x00124b001938a7e5","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":null}
```

I have 2 entries for the coordinator?! one has interviewCompleted: false... which is odd as it doesn't get interviewed right?

Koenkk, 2020-03-24:
Thats indeed interesting, is your `databse.db` old? (could it come from an old herdsman bug?)

sjorge, 2020-03-24:
Around 1.10 I had to rebuild because I couldn't join any devices, so I wipe database.db and changed my netwerk key and pan. So It's from that time. Also I only have one C2xxx device... so not sure why it would even regonize it as a diffrent one.

sjorge, 2020-03-25:
https://pkg.blackdot.be/extras/zigbee/

## 01_repeater_online_room
Should be same as the previous one, I captures this with everything online... just to make sure the issue was back.

## 02_repeater_online_coordinator
This time I moved the bulb next to the coordinator 2 rooms away

## 03_repeater_offline_room
This time I unplugged the IKEA Trådfri Repeater, bulb was back in the room (where I also run wireshark)

**Note: wireshark might not see all traffic, as it is not in the room with the coordinator, it's in the room where the bulb should go**

## 04_repeater_offline_coordinator
Same as 03, but next to the coordinator

**Note: wireshark might not see all traffic, as it is not in the room with the coordinator, it's in the room where the bulb should go**

## 05_all_trådfri_offline_room
This time I took offline all IKEA Trådfri routers (repeater + bulbs), the bulb I tried to pair was back in the room it should end up in.

**Note: wireshark might not see all traffic, as it is not in the room with the coordinator, it's in the room where the bulb should go**

## 06_hue_trådfri_offline_room
Same as 05, but I also took offline the ~~hue~~ innr bulb

**Note: wireshark might not see all traffic, as it is not in the room with the coordinator, it's in the room where the bulb should go**

## 06_hue_trådfri_offline_coordinator
Yeah I messed up the file naming, but whatever.... same as above but the bulb was near the coordinator.

**Note: wireshark might not see all traffic, as it is not in the room with the coordinator, it's in the room where the bulb should go**

@Koenkk a whole set of captures, pairning never succeeded. The last attempt was near the coordinator ~1cm away with all routers offline. Wireshark didn't capture much for that one as it was running on my desktop which is in the room the bulb should end up in.

sjorge, 2020-03-25:
```
zigbee2mqtt:debug 2020-03-25 14:34:22: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
zigbee2mqtt:info  2020-03-25 14:34:22: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-03-25 14:34:22: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.0-dev","commit":"d52d520","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
zigbee2mqtt:info  2020-03-25 14:34:29: Device '0x00158d00038801e9' joined
zigbee2mqtt:info  2020-03-25 14:34:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d00038801e9"}}'
zigbee2mqtt:info  2020-03-25 14:34:29: Starting interview of '0x00158d00038801e9'
zigbee2mqtt:info  2020-03-25 14:34:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d00038801e9"}}'
zigbee2mqtt:debug 2020-03-25 14:34:29: Device '0x00158d00038801e9' announced itself
zigbee2mqtt:info  2020-03-25 14:34:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d00038801e9"}}'
zigbee2mqtt:debug 2020-03-25 14:34:40: Received Zigbee message from '0x00158d00038801e9', type 'readResponse', cluster 'genBasic', data '{"modelId":"ZBT-ExtendedColor","manufacturerName":"MLI"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-25 14:34:40: Received Zigbee message from '0x00158d00038801e9', type 'readResponse', cluster 'genBasic', data '{"powerSource":1,"zclVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-25 14:34:40: Received Zigbee message from '0x00158d00038801e9', type 'readResponse', cluster 'genBasic', data '{"appVersion":1,"stackVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-25 14:34:40: Received Zigbee message from '0x00158d00038801e9', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20180404-42"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-25 14:34:40: Received Zigbee message from '0x00158d00038801e9', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2.0"}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-25 14:34:40: Successfully interviewed '0x00158d00038801e9', device has successfully been paired
zigbee2mqtt:info  2020-03-25 14:34:40: Device '0x00158d00038801e9' is supported, identified as: M_ller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, color, opal white (404000/404005/404012)
zigbee2mqtt:info  2020-03-25 14:34:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d00038801e9","model":"404000/404005/404012","vendor":"M_ller Licht","description":"Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, color, opal white","supported":true}}'
```

After ZNP Soft reset using`zigbee2mqtt/bridge/config/reset`, the pairing works fine from in the room. See capture `08_after_reset_pair_ok` if that is at all needed or not.

Koenkk, 2020-03-26:
Thanks for the sniff, I don't know why the transport key is not occurring, I've asked TI for support (https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/892149)

sjorge, 2020-03-27:
```
digraph G {
node[shape=record];
  "0x00124b001938a7e5" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001938a7e5 (0)|2020-03-27T14:51:51+00:00}"];
  "0x00124b001938a7e5" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="110"]
  "0x00124b001938a7e5" -> "0x000d6ffffe197fe9" [penwidth=0.5, weight=0, color="#994444", label="14"]
  "0x00124b001938a7e5" -> "0x14b457fffe2bd760" [penwidth=0.5, weight=0, color="#994444", label="32"]
  "0x000d6ffffe8e8d4f" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{masterbedroom/repeater|0x000d6ffffe8e8d4f (47816)|IKEA TRADFRI signal repeater (E1746)|2020-03-27T14:51:43+00:00}"];
  "0x000d6ffffe8e8d4f" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="58"]
  "0x000d6ffffe8e8d4f" -> "0x000d6ffffe197fe9" [penwidth=0.5, weight=0, color="#994444", label="58"]
  "0x000d6ffffe8e8d4f" -> "0x14b457fffe2bd760" [penwidth=0.5, weight=0, color="#994444", label="138"]
  "0x000d6ffffe197fe9" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/night_light/bulb|0x000d6ffffe197fe9 (1554)|IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)|2020-03-27T14:51:11+00:00}"];
  "0x000d6ffffe197fe9" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x000d6ffffe197fe9" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="65"]
  "0x000d6ffffe197fe9" -> "0x14b457fffe2bd760" [penwidth=0.5, weight=0, color="#994444", label="129"]
  "0x14b457fffecc1315" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/night_light/remote|0x14b457fffecc1315 (44663)|IKEA TRADFRI ON/OFF switch (E1743)|2020-03-26T23:53:01+00:00}"];
  "0x14b457fffecc1315" -> "0x14b457fffe2bd760" [penwidth=1, weight=0, color="#994444", label="173"]
  "0x14b457fffeca351b" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/desk_lamp/remote|0x14b457fffeca351b (24931)|IKEA TRADFRI ON/OFF switch (E1743)|2020-03-27T06:24:54+00:00}"];
  "0x14b457fffeca351b" -> "0x14b457fffe2bd760" [penwidth=1, weight=0, color="#994444", label="171"]
  "0x00158d0004148f89" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/motion|0x00158d0004148f89 (4263)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2020-03-27T14:49:47+00:00}"];
  "0x00158d0004148f89" -> "0x00124b001938a7e5" [penwidth=1, weight=0, color="#994444", label="1"]
  "0x00158d00033ddfaa" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/sensor|0x00158d00033ddfaa (28407)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-03-27T14:23:58+00:00}"];
  "0x00158d00033ddfaa" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="211"]
  "0x00158d0003f115c5" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{serverroom/sensor|0x00158d0003f115c5 (23324)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-03-27T14:21:24+00:00}"];
  "0x00158d0003f115c5" -> "0x00124b001938a7e5" [penwidth=1, weight=0, color="#994444", label="157"]
  "0x00158d0001ffaffc" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/radiator|0x00158d0001ffaffc (49490)|Eurotronic Spirit Zigbee wireless heater thermostat (SPZB0001)|2020-03-27T14:46:36+00:00}"];
  "0x00158d0001ffaffc" -> "0x14b457fffe2bd760" [penwidth=1, weight=0, color="#994444", label="188"]
  "0x04cf8cdf3c771820" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/light_sensor|0x04cf8cdf3c771820 (24216)|Xiaomi MiJia light intensity sensor (GZCGQ01LM)|2020-03-27T14:51:38+00:00}"];
  "0x04cf8cdf3c771820" -> "0x14b457fffe2bd760" [penwidth=1, weight=0, color="#994444", label="162"]
  "0x14b457fffe2bd760" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/desk_lamp/bulb|0x14b457fffe2bd760 (18409)|Innr E14 bulb RGBW (RB 250 C)|2020-03-27T14:51:17+00:00}"];
  "0x14b457fffe2bd760" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x14b457fffe2bd760" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="114"]
  "0x14b457fffe2bd760" -> "0x000d6ffffe197fe9" [penwidth=0.5, weight=0, color="#994444", label="98"]
}
```

I assume you're going to ask for this next :)

Koenkk, 2020-03-27:
jup, thanks! replied in the thread.

sjorge, 2020-03-27:
@Koenkk I was reading the things the TI person posted... didn't we change something about polling frequency a while ago for the eurotronic trvs? Or am I miss remembering something.

Koenkk, 2020-03-27:
No we didnt change anything related to that (some retry mechanism was added in herdsman but this for sure doesnt cause this issue as herdsman is not even aware of this).

What device are you trying to join?

sjorge, 2020-03-27:
Any device will do, I have had this problem with:
- hue E14 bulb
- innr E14 bulb
- muller light tint E27 bulb
- mijia light sensor
- trådfri E27 1000lm bulb

Koenkk, 2020-03-27:
- Could you provide the same sniff when pairing a zigbee router?
- Can you also test with this suggestion? https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/892149/3299688#3299688

sjorge, 2020-03-27:
The pcaps from earlier were with a muller light tint E27... which I believe is a zigbee router?

Also 05_all_trådfri_offline_room would only have a hue bulb online as router at the time and a TRV as it is very annoying to remove.

Koenkk, 2020-03-27:
Which device did you try to pair in: https://pkg.blackdot.be/extras/zigbee/failed_join.pcapng ?

sjorge, 2020-03-27:
That was the Innr E14 I wanted to use to replace the Hue E14 because genOnOff reporting was no longer supported on the Hue after the update.

All the numbered ones were with the Hue replaced with the Innr already, where I tried to pair a new Tiny E27 (https://www.zigbee2mqtt.io/devices/404000_404005_404012.html) to look into what it supported reporting wise as mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/3177#issuecomment-602861258

Koenkk, 2020-03-27:
And just to be sure, this Innr is marked as a `Router` in database.db right?

sjorge, 2020-03-27:
```
{
  "id": 16,
  "type": "Router",
  "ieeeAddr": "0x14b457fffe2bd760",
  "nwkAddr": 18409,
  "manufId": 4454,
  "manufName": "innr",
  "powerSource": "Mains (single phase)",
  "modelId": "RB 250 C",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 269,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        8,
        768,
        2821,
        4096
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "RB 250 C",
            "manufacturerName": "innr",
            "powerSource": 1,
            "zclVersion": 3,
            "appVersion": 16,
            "stackVersion": 98,
            "hwVersion": 1,
            "dateCode": "20190326-87",
            "swBuildId": "2.1"
          }
        },
        "lightingColorCtrl": {
          "attributes": {
            "currentHue": 0,
            "currentSaturation": 254,
            "colorMode": 1,
            "currentX": 45914,
            "currentY": 19615,
            "colorTemperature": 153
          }
        },
        "genLevelCtrl": {
          "attributes": {
            "currentLevel": 254
          }
        },
        "genOnOff": {
          "attributes": {
            "onOff": 1
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001938a7e5",
          "endpointID": 1
        },
        {
          "cluster": 8,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001938a7e5",
          "endpointID": 1
        },
        {
          "cluster": 768,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001938a7e5",
          "endpointID": 1
        }
      ]
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [
        
      ],
      "outClusterList": [
        33
      ],
      "clusters": {
        
      },
      "binds": [
        
      ]
    }
  },
  "appVersion": 16,
  "stackVersion": 98,
  "hwVersion": 1,
  "dateCode": "20190326-87",
  "swBuildId": "2.1",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "reporting": 1
  },
  "lastSeen": 1585152024465
}
```
Yep

Koenkk, 2020-03-28:
Thanks, could you also provide a sniff with all the traffic when pairing close to the coordinator fails? (when the association response comes from the coordinator)

sjorge, 2020-03-28:
I'll have to get creative, but I can try. But since I reset the stick today to do the group thing for the other issue, currently pairing is working fine. So we have to wait a few days before it 'breaks' again.

FaBjE, 2020-03-28:
I experience the same issue/symptoms. I'm following this issue with great interest.

What I noticed: 
- When you keep joining devices It doesn't occur. Only after "idle" time. (I haven't tried joining devices for hours off course, but I was always able to join a large number of devices.
- For me it happens after a few hours
- The Zigbee network itself functions fine (Can communicate with all existing nodes)
- To fix it, the USB stick has to be physically re-plugged
- Forcing a reset of the stick using the mqtt topic either freezes everything or doesn't help
- It happened when running "bare" on my RPI and now running in a docker-container on a desktop.

Version info:
{"version":"1.12.0","commit":"840b9d9",
"coordinator":{"type":"zStack12","meta":"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}

sjorge, 2020-03-28:
>     * Forcing a reset of the stick using the mqtt topic either freezes everything or doesn't help

Yeah it freeze z2m for me too, but after the reset... I can just restart z2m and it starts working again including joining... so it does save me a physical trip to the server room

groenmarsmannetje, 2020-03-30:
It also feels like devices are dropped from network more often then before. Maybe it is a problem for these devices to reconnect after a while. This happened to both wired as battery devices, but in my case most of them are battery devices. For instance all my smoke detectors are dropped. They normally only wake up and connect once per day. Never had issues with these devices, they were always able to connect even through other routers/routes without any issue.

sjorge, 2020-04-03:
> It also feels like devices are dropped from network more often then before. Maybe it is a problem for these devices to reconnect after a while. This happened to both wired as battery devices, but in [...]

Oh yeah definately... my Aqara Motion sensor drops nearly every 3 days.

sjorge, 2020-04-03:
> Thanks, could you also provide a sniff with all the traffic when pairing close to the coordinator fails? (when the association response comes from the coordinator)

OK I finally managed to get my laptop close enough to sniff... it took me a few tries...
But I accidently bumped my usb so it was working fine because it got unplugged :s

sjorge, 2020-04-08:
I'll try to do a capture tomorrow, can't today but joining broke again!
I also noticed that currently all trådfri bulbs/repeaters are no longer reporting either!

I tried plug/unplug of a bulb but it fails to setup reporting!
```
zigbee2mqtt:error 2020-04-08 19:23:36: Failed to setup reporting for '0x90fd9ffffee77fcf' - Error: Bind 0x90fd9ffffee77fcf/1 genOnOff from '0x00124b001938a7e5/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
```

I wonder if it is related.

sjorge, 2020-04-09:
@Koenkk  Good news! I had the issue again:
- https://pkg.blackdot.be/extras/zigbee/01_bedroom_fail.pcapng (in the bedroom, to test if it was failing, it was)
- https://pkg.blackdot.be/extras/zigbee/03_coordinator_fail_attempt2.pcapng (first attempt I did not capture the Permit Join requests, the TK should follow those so I redid it)

extra captures probably not needed:
- https://pkg.blackdot.be/extras/zigbee/04_coordinator_touchlink_fail.pcapng (same but tried resetting with touchlink)
- https://pkg.blackdot.be/extras/zigbee/05_coordinator_after_reset_OK.pcapng (after unplugging the coordinator and plugging it back in, works fine and bulb joined and got configured)

sjorge, 2020-04-09:
This was on the dev branch from yesterday eveningish

Koenkk, 2020-04-10:
Can you try with this firmware: 
[CC2531ZNP-Prod_Source_Routing_20200410.zip](https://github.com/Koenkk/zigbee2mqtt/files/4462296/CC2531ZNP-Prod_Source_Routing_20200410.zip)

sjorge, 2020-04-10:
Does the fixes only apply to the source routing firmware? I'm using the default one currently.

I don't mind switching though, but the last time I tried I had issues when repairing my devices (after flashing I had to repair for some reason, although I thought the docs said that was not needed)

Koenkk, 2020-04-11:
Reflashing shouldn't be needed indeed. But here is the default firmware: 
[CC2531ZNP-Prod_20200410_default.zip](https://github.com/Koenkk/zigbee2mqtt/files/4464277/CC2531ZNP-Prod_20200410_default.zip)

sjorge, 2020-04-11:
I'll flash the default one later today, should give the best results to see if those changes help or not.

sjorge, 2020-04-11:
Done, I flashed the default one... will keep you posted.
```json
{
  "type": "zStack12",
  "meta": {
    "transportrev": 2,
    "product": 0,
    "majorrel": 2,
    "minorrel": 6,
    "maintrel": 3,
    "revision": 20200410
  }
}
```

I did figure out why I had to reflash last time! The instructions are incorrect!
https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html as I was using windows as that was easiest.

If do as the instructions say and copy the settings form the screenshot, you uncheck: `Retain IEEE address when reprogramming the chip`, which seems to then require repairing of all devices. This time I left it checked (default) and it worked, all devices are still here and working aside from the Trådfri repeater... I had to unplug it and it was fine.

Also noticed a new `Coordinator` entry in my database.db after flashing the newer firmware!
```json
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001938a7e5","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":null}
```

```
{"id":9,"type":"Coordinator","ieeeAddr":"0x00124b001938a7e5","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,47,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[]},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":null}
```

Two new endpoints showed up, 13 and 47?

Koenkk, 2020-04-11:
New endpoints are expected, but duplicate entry not. Could you reproduce the issue and provide the herdsman debug logging of this?

sjorge, 2020-04-11:
Not sure how to reproduce, aside form flashing another firmware that has different endpoints maybe?

sjorge, 2020-04-11:
I could not reproduce:
- stop z2m
- flash the stick with old firmware
- start z2m
- check database (only one entry)
- stop z2m
- flash back to testing firmware
- start z2m
- check database, still only one entry

I had manually cleaned up the dup entry already before.

FaBjE, 2020-04-11:
You are right about the Retain IEEE, on a new stick it should be unchecked as it may not be initialized.
On updating an existing stick it must be checked.

On a long stretch, maybe the duplicate entries changing of the coordinator address had something to do with it?
I imagine something like, adding a new entry because of the new address, but later updating all the coordinator entries with the new address. 
But that is just a wild guess.

sjorge, 2020-04-11:
Hmm yeah it might be from the last time I tried the source firmware and had retain IEEE unchecked. That was before the changes where database.db would update the coordinator's IEEE... so it could be that yeah.

sjorge, 2020-04-12:
@Koenkk is it me... or does the new firmware respond slower?

Koenkk, 2020-04-13:
Not sure, I don't expect this from the diff with your previous firmware. Please check if reverting to that the old firmware solves the issue.

sjorge, 2020-04-13:
I'll keep on this one until next weekend and try to pair a device again before reflashing.

I was sniffing yesterday evening a bit and it seems to be slow when a lot of route discovery are done. Usually when a attReport overlaps with a msg send to a group. So the source routing firmware might also help.

sjorge, 2020-04-14:
So good news, the slowness doesn't seem to have anything todo with the new firmware.
That seems to be because the trådfri repeater again. sfjhaskhf

sjorge, 2020-04-15:
@Koenkk bad news :(

I tried to join a device this morning and nothing again. After a usb reset, it worked fine.

After about ~30min (I wasn't watching the log) the ikea bulb behind the repeater came back online... As soon as I had reset the USB the repeater started responding to pings again.
And shortly thereafter the bulb did a deviceannounce and reconfigure of reporting!

I did not have any issues with these last week, but I did reset the usb stick every few days testing stuff... so it was never a 24 period where joining was broken. I suspect the issue with the trådfri stuff is a result of the coordinator issues. (I had already tried without any trådfri devices a few weeks ago, and the coordinator would still refuse device joins too.)

sjorge, 2020-04-15:
I will try flashing the source routing firmware you posted in tonight or tomorrow depending on when i have time to flash the stick. (Not that I expect it will make a difference, but it's worth a shot)

sjorge, 2020-04-15:
z2m does not start with the source routing firmware: https://gist.github.com/sjorge/df385a8cc8741b912dc4be9d53c127b7

sjorge, 2020-04-15:
Looks like the C2531 is now completely dead.
I tried:
- patched default
- patched source
- normal default
- normal source

All result in the same error.

sjorge, 2020-04-15:
Fixed by adding `rtscts: false` to my configuration.yaml *visible confused*

Now I lost track which firmware I had on it. It's one of the patched ones that I can see from
`'{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"main
trel":3,"revision":20200410}}`

How do I check if it is the source routing or normal one?

sjorge, 2020-04-16:
Sofar the patched source routiing firmware is working well... the network is much more responsive than before,... it certainly does not have the 'slowness' I experienced at times on the default one.

sjorge, 2020-04-18:
Some further observations:
- only 2 ping failures so far, they were when I was actively working on some homekit integration I currently have in node-red so putting a lot of load on the network
- networkmap returns near instant?!?!
- networkmap looks pretty stable after ~2 days it stopped changing.
- briefly I had a routing loop where bed_lamp en desk_lamp would have the green arrow in a loop and desk_lamp having one go to the repeater
- current netmap seems to form the best route (based on physical location) back to the coordinator
- lots less traffic on the network, before it was mostly route requests from the coordinator
- was still able to pair yesterday, will check every few days for another week or so to be sure.

```
digraph G {
node[shape=record];
  "0x00124b001938a7e5" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{0x00124b001938a7e5|0x00124b001938a7e5 (0)|2020-04-18T11:18:56+00:00}"];
  "0x00124b001938a7e5" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="91"]
  "0x00124b001938a7e5" -> "0x14b457fffe2bd760" [penwidth=0.5, weight=0, color="#994444", label="32"]
  "0x000d6ffffe8e8d4f" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{masterbedroom/repeater|0x000d6ffffe8e8d4f (46726)|IKEA TRADFRI signal repeater (E1746)|2020-04-18T11:17:23+00:00}"];
  "0x000d6ffffe8e8d4f" -> "0x00124b001938a7e5" [penwidth=2, weight=1, color="#009900", label="46 (routes: 31787,49490)"]
  "0x000d6ffffe8e8d4f" -> "0x90fd9ffffee77fcf" [penwidth=0.5, weight=0, color="#994444", label="48"]
  "0x000d6ffffe8e8d4f" -> "0x14b457fffe2bd760" [penwidth=0.5, weight=0, color="#994444", label="119"]
  "0x90fd9ffffee77fcf" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/bed_lamp/bulb|0x90fd9ffffee77fcf (31787)|IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)|2020-04-18T11:17:05+00:00}"];
  "0x90fd9ffffee77fcf" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x90fd9ffffee77fcf" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="59"]
  "0x90fd9ffffee77fcf" -> "0x14b457fffe2bd760" [penwidth=2, weight=1, color="#009900", label="123 (routes: 31787)"]
  "0x14b457fffe2bd760" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/desk_lamp/bulb|0x14b457fffe2bd760 (39924)|Innr E14 bulb RGBW (RB 250 C)|2020-04-18T11:17:29+00:00}"];
  "0x14b457fffe2bd760" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x14b457fffe2bd760" -> "0x000d6ffffe8e8d4f" [penwidth=2, weight=1, color="#009900", label="98 (routes: 39924,31787)"]
  "0x14b457fffe2bd760" -> "0x90fd9ffffee77fcf" [penwidth=0.5, weight=0, color="#994444", label="91"]
  "0x14b457fffecc1315" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/bed_lamp/remote|0x14b457fffecc1315 (44663)|IKEA TRADFRI ON/OFF switch (E1743)|2020-04-18T04:33:23+00:00}"];
  "0x14b457fffecc1315" -> "0x90fd9ffffee77fcf" [penwidth=1, weight=0, color="#994444", label="255"]
  "0x14b457fffeca351b" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/desk_lamp/remote|0x14b457fffeca351b (24931)|IKEA TRADFRI ON/OFF switch (E1743)|2020-04-18T11:15:29+00:00}"];
  "0x14b457fffeca351b" -> "0x90fd9ffffee77fcf" [penwidth=1, weight=0, color="#994444", label="255"]
  "0x00158d0004148f89" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/motion|0x00158d0004148f89 (12102)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2020-04-18T11:14:36+00:00}"];
  "0x00158d0004148f89" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="134"]
  "0x00158d00033ddfaa" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/sensor|0x00158d00033ddfaa (2897)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-04-18T11:16:34+00:00}"];
  "0x00158d00033ddfaa" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="184"]
  "0x00158d0003f115c5" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{serverroom/sensor|0x00158d0003f115c5 (23324)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-04-18T11:16:21+00:00}"];
  "0x00158d0003f115c5" -> "0x00124b001938a7e5" [penwidth=1, weight=0, color="#994444", label="140"]
  "0x00158d0001ffaffc" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/radiator|0x00158d0001ffaffc (49490)|Eurotronic Spirit Zigbee wireless heater thermostat (SPZB0001)|2020-04-18T11:18:53+00:00}"];
  "0x00158d0001ffaffc" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="120"]
  "0x04cf8cdf3c771820" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/light_sensor|0x04cf8cdf3c771820 (24216)|Xiaomi MiJia light intensity sensor (GZCGQ01LM)|2020-04-18T11:18:37+00:00}"];
  "0x04cf8cdf3c771820" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="169"]
}
```

sjorge, 2020-04-20:
@Koenkk sadly joining stopped working too now :(

So it looks like the chances have no effect with either the default or source routing firmware.
And the extra network stability was just from using source routing in general.

Any other ideas? Do you want me to do another capture with this firmware?
- near coordinator
- near a router

sjorge, 2020-04-20:
Well well this is new and interesting!
- capture in the bedroom, so it has to go through a router to get the coordinator --> [fails](https://pkg.blackdot.be/cores/zigbee/srcrouting_via_router.pcapng)
- captured at the coordinator --> [success](https://pkg.blackdot.be/cores/zigbee/srcrouting_coordinator.pcapng)

[108 more replies]

closed 2020-11-28 by Koenkk