Koenkk/zigbee2mqtt #3178: Wrong state after device remove _ rename

sjorge, 2020-03-22:
# Bug Report
## What happened
I was trying to swap out E14 Hue bulb to my E15 Innr bulb (since on/off reporting is broken on the former)

- pair innr bulb
- remove hue bulb (blinks that it is reset)
- unscrew hue bulb
- swap innr from mobile test socket to new socket
- rename innr so it has the same friendly name the hue had (so far so good)
- add innr to the a group using friendly name (bad stuff)

bad stuff ==> it somehow used the ieeeAddr from the hue bulb and not the innr bulb and stuff failed.

I think this has to due with how the settings are stored,

after removing the hue bulb, the device/friendly_name config remains in configuration.yaml, after renaming the innr bulb the file gets update and the hue bulb config is no longer to be found (good).

I think the in memory version might still have the incorrect information at that point.

## What did you expect to happen
The ieeeAddr from the innr bulb to be used.

## How to reproduce it (minimal and precise)
See what happened.

## Debug Info
zigbee2mqtt version: `1.12.0-dev (commit #d52d520)`
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}`

sjorge, 2020-03-22:
On a slight fun note, not relevant but wanted to share somewhere
```
zigbee2mqtt:debug 2020-03-22 23:09:52: Received Zigbee message from 'bedroom/desk_lamp/bulb', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorMode":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-22 23:09:52: Received Zigbee message from 'bedroom/desk_lamp/bulb', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentHue":136,"currentSaturation":36,"currentX":20427,"currentY":21476,"colorTemperature":153}' from endpoint 1 with groupID 0
```

The innr seems to accept both XY, ColorTemperatue and Hue/sat (not enhanced) and properly report the color_mode changes! <3 This is the only brand I found so far that does this!

Koenkk, 2020-03-23:
Could you provide the debug + herdsman debug log of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

sjorge, 2020-03-28:
I tried today but I cannot get my Hue bulb to repair (also not with touchlink)... sniffing sees no traffic form the hue bulb at all!

So my plan to do the reverse of what I did the first time won't work.

sjorge, 2020-03-28:
Edit: I got it to work after using a hue remote to reset it 🤷‍♂ I thought touchlink would do the same.

- [X] pair hue bulb
- [X] remove innr bulb (blinks that it is reset)
- [X] unscrew innr bulb
- [X] swap hue from mobile test socket to new socket
- [X] rename hue so it has the same friendly name the hue had (so far so good)
- [X] add hue to the a group using friendly name (bad stuff)
- [X] toggled the group with the remote a few times to trigger the error repeatedly

https://pkg.blackdot.be/extras/zigbee/debug.log the log

The `data/configuration.yaml` after the above steps before stopping z2m.
```
devices:
  '0x000d6ffffe8e8d4f':
    friendly_name: masterbedroom/repeater
    debounce: 0
    retain: false
  '0x000d6ffffe197fe9':
    friendly_name: bedroom/night_light/bulb
  '0x14b457fffecc1315':
    friendly_name: bedroom/night_light/remote
    debounce: 0
    retain: false
  '0x14b457fffeca351b':
    friendly_name: bedroom/desk_lamp/remote
    debounce: 0
    retain: false
  '0x00158d0004148f89':
    friendly_name: bedroom/motion
  '0x00158d00033ddfaa':
    friendly_name: bedroom/sensor
  '0x00158d0003f115c5':
    friendly_name: serverroom/sensor
  '0x00158d0001ffaffc':
    friendly_name: bedroom/radiator
    filtered_attributes:
      - occupied_heating_setpoint
      - unoccupied_heating_setpoint
      - eurotronic_system_mode
    debounce: 5
  '0x04cf8cdf3c771820':
    friendly_name: bedroom/light_sensor
  '0x0017880104259333':
    friendly_name: bedroom/desk_lamp/bulb
groups:
  '20861':
    friendly_name: bedroom/night_light
    devices:
      - 0x000d6ffffe197fe9/1
  '20862':
    friendly_name: bedroom/desk_lamp
    devices:
      - 0x14b457fffe2bd760/1
```

sjorge, 2020-03-28:
Some extra data, I stopped the debug log and swapped back to the innr bulb as that one has reporting...

After removing the hue bulb form the network:
```
devices:
  '0x000d6ffffe8e8d4f':
    friendly_name: masterbedroom/repeater
    debounce: 0
    retain: false
  '0x000d6ffffe197fe9':
    friendly_name: bedroom/night_light/bulb
  '0x14b457fffecc1315':
    friendly_name: bedroom/night_light/remote
    debounce: 0
    retain: false
  '0x14b457fffeca351b':
    friendly_name: bedroom/desk_lamp/remote
    debounce: 0
    retain: false
  '0x00158d0004148f89':
    friendly_name: bedroom/motion
  '0x00158d00033ddfaa':
    friendly_name: bedroom/sensor
  '0x00158d0003f115c5':
    friendly_name: serverroom/sensor
  '0x00158d0001ffaffc':
    friendly_name: bedroom/radiator
    filtered_attributes:
      - occupied_heating_setpoint
      - unoccupied_heating_setpoint
      - eurotronic_system_mode
    debounce: 5
  '0x04cf8cdf3c771820':
    friendly_name: bedroom/light_sensor
groups:
  '20861':
    friendly_name: bedroom/night_light
    devices:
      - 0x000d6ffffe197fe9/1
  '20862':
    friendly_name: bedroom/desk_lamp
    devices:
      - 0x14b457fffe2bd760/1
```

Looks like the device, looks like the hue was never in the group according to the configuration.yaml... it did get nicely remove form the devices section. (pretty sure the innr bulb will automatically end up in the group after pairing and restarting z2m)

Koenkk, 2020-03-29:
Should be fixed in the latest dev branch now.

closed 2020-03-29 by sjorge