Koenkk/zigbee2mqtt #3180: service will not start any longer

prankousky, 2020-03-23:
Hi,

some of my zigbee devices stopped working yesterday, so I updates via the bash script in `/opt/zigbee2mqtt/`. Since then, I always get this when trying to start the service

```
-- Logs begin at Mon 2020-03-23 09:56:46 CET. --
Mär 23 09:57:10 beelink npm[3060]:     at Module.load (internal/modules/cjs/loader.js:653:32)
Mär 23 09:57:10 beelink npm[3060]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
Mär 23 09:57:10 beelink npm[3060]: (node:3189) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
Mär 23 09:57:10 beelink npm[3060]: (node:3189) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Mär 23 09:57:10 beelink systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Mär 23 09:57:10 beelink systemd[1]: Stopped zigbee2mqtt.
Mär 23 09:57:10 beelink systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Mär 23 09:57:10 beelink systemd[1]: Failed to start zigbee2mqtt.
Mär 23 09:57:10 beelink systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Mär 23 09:57:10 beelink systemd[1]: zigbee2mqtt.service: Failed with result 'start-limit-hit'.
```

This started last night, so I ran `sudo systemctl stop zigbee2mqtt.service` and kept it off until this morning. Now I unplugged the coordinator, restarted the entire server, and plugged it back in. The output above is from this morning when I **just** tried running it again, so I don't understand how a limit could be reached...

Please help! Currently nothing related to zigbee2mqtt (so all my zigbee devices) works.
Thank you in advance :)

prankousky, 2020-03-23:
Here is a more detailed log output

```
Mär 23 10:21:27 beelink npm[8036]:     at Array.forEach (<anonymous>)
Mär 23 10:21:27 beelink npm[8036]:     at Object.validate (/opt/zigbee2mqtt/lib/util/settings.js:316:50)
Mär 23 10:21:27 beelink npm[8036]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.js:79:41)
Mär 23 10:21:27 beelink npm[8036]:     at Object.<anonymous> (/opt/zigbee2mqtt/index.js:11:12)
Mär 23 10:21:27 beelink npm[8036]:     at Module._compile (internal/modules/cjs/loader.js:778:30)
Mär 23 10:21:27 beelink npm[8036]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
Mär 23 10:21:27 beelink npm[8036]:     at Module.load (internal/modules/cjs/loader.js:653:32)
Mär 23 10:21:27 beelink npm[8036]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
Mär 23 10:21:27 beelink npm[8036]: (node:8052) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
Mär 23 10:21:27 beelink npm[8036]: (node:8052) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Mär 23 10:21:27 beelink systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Mär 23 10:21:27 beelink systemd[1]: Stopped zigbee2mqtt.
Mär 23 10:21:27 beelink systemd[1]: Started zigbee2mqtt.
Mär 23 10:21:28 beelink npm[8066]: > zigbee2mqtt@1.12.0 start /opt/zigbee2mqtt
Mär 23 10:21:28 beelink npm[8066]: > node index.js
Mär 23 10:21:28 beelink npm[8066]: zigbee2mqtt:info  2020-03-23 10:21:28: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-23.10-21-28' filename: log.txt
Mär 23 10:21:28 beelink npm[8066]: (node:8078) UnhandledPromiseRejectionWarning: Error: Friendly name cannot end with a "/DIGIT" ('Button/02')
Mär 23 10:21:28 beelink npm[8066]:     at check (/opt/zigbee2mqtt/lib/util/settings.js:313:43)
Mär 23 10:21:28 beelink npm[8066]:     at Object.values.forEach (/opt/zigbee2mqtt/lib/util/settings.js:316:65)
Mär 23 10:21:28 beelink npm[8066]:     at Array.forEach (<anonymous>)
Mär 23 10:21:28 beelink npm[8066]:     at Object.validate (/opt/zigbee2mqtt/lib/util/settings.js:316:50)
Mär 23 10:21:28 beelink npm[8066]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.js:79:41)
Mär 23 10:21:28 beelink npm[8066]:     at Object.<anonymous> (/opt/zigbee2mqtt/index.js:11:12)
Mär 23 10:21:28 beelink npm[8066]:     at Module._compile (internal/modules/cjs/loader.js:778:30)
Mär 23 10:21:28 beelink npm[8066]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
Mär 23 10:21:28 beelink npm[8066]:     at Module.load (internal/modules/cjs/loader.js:653:32)
Mär 23 10:21:28 beelink npm[8066]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
Mär 23 10:21:28 beelink npm[8066]: (node:8078) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
Mär 23 10:21:28 beelink npm[8066]: (node:8078) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Mär 23 10:21:28 beelink systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Mär 23 10:21:28 beelink systemd[1]: Stopped zigbee2mqtt.
Mär 23 10:21:28 beelink systemd[1]: Started zigbee2mqtt.
Mär 23 10:21:29 beelink npm[8090]: > zigbee2mqtt@1.12.0 start /opt/zigbee2mqtt
Mär 23 10:21:29 beelink npm[8090]: > node index.js
Mär 23 10:21:29 beelink npm[8090]: zigbee2mqtt:info  2020-03-23 10:21:29: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-23.10-21-29' filename: log.txt
Mär 23 10:21:30 beelink npm[8090]: (node:8102) UnhandledPromiseRejectionWarning: Error: Friendly name cannot end with a "/DIGIT" ('Button/02')
Mär 23 10:21:30 beelink npm[8090]:     at check (/opt/zigbee2mqtt/lib/util/settings.js:313:43)
Mär 23 10:21:30 beelink npm[8090]:     at Object.values.forEach (/opt/zigbee2mqtt/lib/util/settings.js:316:65)
Mär 23 10:21:30 beelink npm[8090]:     at Array.forEach (<anonymous>)
Mär 23 10:21:30 beelink npm[8090]:     at Object.validate (/opt/zigbee2mqtt/lib/util/settings.js:316:50)
Mär 23 10:21:30 beelink npm[8090]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.js:79:41)
Mär 23 10:21:30 beelink npm[8090]:     at Object.<anonymous> (/opt/zigbee2mqtt/index.js:11:12)
Mär 23 10:21:30 beelink npm[8090]:     at Module._compile (internal/modules/cjs/loader.js:778:30)
Mär 23 10:21:30 beelink npm[8090]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
Mär 23 10:21:30 beelink npm[8090]:     at Module.load (internal/modules/cjs/loader.js:653:32)
Mär 23 10:21:30 beelink npm[8090]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
Mär 23 10:21:30 beelink npm[8090]: (node:8102) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
[... cut ...]
Mär 23 10:21:32 beelink systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Mär 23 10:21:32 beelink systemd[1]: Stopped zigbee2mqtt.
Mär 23 10:21:32 beelink systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Mär 23 10:21:32 beelink systemd[1]: Failed to start zigbee2mqtt.
Mär 23 10:21:32 beelink systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Mär 23 10:21:32 beelink systemd[1]: zigbee2mqtt.service: Failed with result 'start-limit-hit'.

```

Koenkk, 2020-03-23:
One of your friendly_names in configuration.yaml is ending with a /DIGIT (e.g. /1), this isn't allowed (but previously wasn't checked)

prankousky, 2020-03-23:
Thank you @Koenkk 
You're the best!

closed 2020-03-23 by prankousky