Koenkk/zigbee2mqtt #3181: adding new in wall switch but cannot get on/off state

kelvin5290, 2020-03-23:
## What happened
i add a new switch on device.js, i can control on/off. However, the state of the switch is not consist. For example, i turn on the light manually, the status will not change and i cannot see any log which the switch changed status. 
this is the device 
`{
        zigbeeModel: ['cc891c853291418daa4e26cb315a3a0c'],
        model: 'cc891c853291418daa4e26cb315a3a0c',
        vendor: 'ORVIBO',
        description: 'switch 2 gang',
        supports: 'on/off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        meta: {configureKey: 2, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint1 = device.getEndpoint(1);
            await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint1);
            const endpoint2 = device.getEndpoint(2);
            await bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint2);
            },
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        }
       }`

the only thing i got is "device_announced".
`2020-03-23T12:46:20.502Z zigbee-herdsman:controller:log Device announce '0x00124b001eb183e3'
zigbee2mqtt:debug 2020-03-23 20:46:20: Device '0x00124b001eb183e3' announced itself
zigbee2mqtt:info  2020-03-23 20:46:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001eb183e3"}}'`

database.db
`{"id":3,"type":"Router","ieeeAddr":"0x00124b001eb183e3","nwkAddr":53279,"manufId":0,"manufName":"0000000","powerSource":"Mains (single phase)","modelId":"cc891c853291418daa4e26cb315a3a0c","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,5,4,6],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"cc891c853291418daa4e26cb315a3a0c","manufacturerName":"0000000","powerSource":1,"zclVersion":1,"appVersion":1,"hwVersion":16,"dateCode":"2019-01-22","swBuildId":"v3.3.7"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff042606","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[0,5,4,6],"outClusterList":[0],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff042606","endpointID":1}]}},"appVersion":1,"hwVersion":16,"dateCode":"2019-01-22","swBuildId":"v3.3.7","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1584966755455}`

## Debug Info
zigbee2mqtt version: 
i am using deconz conbee ii

kelvin5290, 2020-03-23:
i can get the state using zigbee2mqtt/<friendly_name>/get
`zigbee2mqtt:debug 2020-03-23 21:33:25: Received Zigbee message from '0x00124b001eb183e3', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
zigbee2mqtt:info  2020-03-23 21:33:25: MQTT publish: topic 'zigbee2mqtt/0x00124b001eb183e3', payload '{"state":"OFF","state_l1":"OFF","state_l2":"OFF","linkquality":157}'`

is it all switch need to get state manually?

kelvin5290, 2020-03-23:
and also, i only can get stase l1, the l2 didnt change even it is on.

Koenkk, 2020-03-23:
Your `devices.js` definition looks OK and I don't see anything else strange. Could you sniff the zigbee traffic when manually controlling the switch? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

kelvin5290, 2020-03-31:
[image]

kelvin5290, 2020-03-31:
BTW, i am using conbee II and i found when Binding the switch, there are malformed packet.

[image]

Koenkk, 2020-03-31:
That's probably the cause of the issue, please report it here so that they can fix it; https://github.com/Koenkk/zigbee-herdsman/issues/72

kelvin5290, 2020-05-16:
hi @Koenkk ,

i tried to add the device again, but the log show a new ERROR as below. 

`zigbee2mqtt:error 2020-05-16 22:14:02: Failed to configure '0x00124b0019912a27', attempt 1 (Error: ConfigureReporting 0x00124b0019912a27/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (TypeError: Cannot read property 'frame' of null)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))`

Koenkk, 2020-05-17:
This also looks like a deconz specific issue. Can you make sure that this issue is not fixed yet by trying with the latest dev branch?

closed 2020-07-25 by stale[bot]

kelvin5290, 2020-09-13:
hi @Koenkk,
Sorry to find you again. i Changed to use cc2531, but the problem is still occurred. It look like not about the deconz. the error log is shown as belown.

```
Zigbee2MQTT:error 2020-09-13 21:54:48: Failed to configure '0x00124b0019912a27', attempt 1 (Error: ConfigureReporting 0x00124b0019912a27/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 35528 - 1 - 16 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)) 
```
please let me know what information should i provide to you. Thank you

Koenkk, 2020-09-14:
@kelvin5290 could you provide the sniff of this again?