Koenkk/zigbee2mqtt #3182: Support new HUE device, unable to access hassio host

knutasund, 2020-03-23:
I am trying to add support for Hue Econic outdoor light (1743830P7), but I am not able to get steps in the guide to work - it stops with "Enable access to your Hass.io host". Do I need to to ALL steps 
https://developers.home-assistant.io/docs/hassio_debugging/ to get access to the host? 

Any other easier way of getting the access to the files I need?

## Debug Info
zigbee2mqtt version: 1.12.0

markus1540, 2020-03-23:
So it seems you are running Home Assistant OS, this repo would be better suited for questions then:
https://github.com/danielwelch/hassio-zigbee2mqtt
Also if you just want to customize the devices.js file there's this config option for the addon documented:

> Adding Support for New Devices
If you are interested in adding support for new devices to zigbee2mqtt, set the optional, top-level zigbee_shepherd_devices option to true in your configuration. When set, the add-on will scan your data_path for a devices.js file, and will run zigbee2mqtt using this custom file.

knutasund, 2020-03-27:
Thanks a lot, I have been assuming that device support was relying on the base zigbee2mqtt and the hass-integration only was inheriting from that.

closed 2020-03-27 by knutasund