Koenkk/zigbee2mqtt #3183: Add support for ORVIBO ZIGBEE RGB RL804CZB

OXOTH1K, 2020-03-23:
Thats work for me:

```
    // Orvibo
    {
        zigbeeModel: ['8643db61de35494d93e72c1289b526a3'],
        model: 'RL804CZB',
        vendor: 'Orvibo',
        description: 'Zigbee LED controller RGB + CCT or RGBW',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
        supports: 'on/off, brightness, color temperature or white, color',
    },
```
Device:
https://secur.ua/rele-pitanija-orvibo-zigbee-rgb-rl804czb.html

Koenkk, 2020-03-23:
Thanks, will be supported in the dev branch in a few hours.

closed 2020-03-23 by Koenkk

cingolanifede, 2020-10-05:
Hi, I have this Orvibo rgb, it have 3 micro-switch and it gives 2 different zigbeemodels,
'e84c3be8030e494d8e29044f50ed45a4' & '6c155b46eb9a4dcbb64fb3dd858aa18e'. I 've added them to device.js, but I can't change the color. Only blue channel is working, i've tested and there is no mosfet burned. @OXOTH1K did you have this problem? Is there any kind of config to modify? thanks

OXOTH1K, 2020-10-16:
> Hi, I have this Orvibo rgb, it have 3 micro-switch and it gives 2 different zigbeemodels,
> 'e84c3be8030e494d8e29044f50ed45a4' & '6c155b46eb9a4dcbb64fb3dd858aa18e'. I 've added them to device.js, but [...]

I use this device only with white led strip, sorry

yuz123, 2021-12-12:
> Hi, I have this Orvibo rgb, it have 3 micro-switch and it gives 2 different zigbeemodels, 'e84c3be8030e494d8e29044f50ed45a4' & '6c155b46eb9a4dcbb64fb3dd858aa18e'. I 've added them to device.js, but [...]

I have the same issue. Did you resolve it?

OXOTH1K, 2022-02-19:
> Hi @OXOTH1K. Please share your experience about connecting Orvibo RL804CZB. My Zigbee2MQTT does not find this device. Maybe I'm doing something wrong, I just didn't find anything about it on the internet. I apologize for disturbing and offtopic.

Oh, it’s hard to switch in paring mode. I’ll go to find instruction tomorrow.

OXOTH1K, 2022-02-20:
[image]

lxndrrl, 2022-02-20:
@OXOTH1K Thank you very much.

tjunussov, 2023-09-30:
I have CC10ZW ( looks same as RL804CZB ) I successfully paired, and connected Warm+Cold White Led Strip to the W1 and W2 accordingly, but only Warm Channel is working, i could change brightness and turn on/off,, If i switch to Cool Channel nothing happens, no errors, only warm channel keeps on