Koenkk/zigbee2mqtt #3184: zigbee2mqtt stops running after days or hours

regissg, 2020-03-23:
This issue was covered in #2095 but was closed by the stale robot and I do not know how to reopen a closed case.

I keep having from time to time (could be weeks) the same problem where zigbee2mqtt is not working anymore and where I need to unplug/plug the CC2531 and then restart the zigbee2mqtt service on windows.
The issue is blocking me as I now use zigbee2mqtt to control my home heating system.

The CC2531 is running the 20190619 version and my windows 10 PC was running zigbee2mqtt version 1.7.1. I have just updated it to 1.12.0.
The CC2531 is connected to a powered usb hub with a 3 meter cable to the PC.

Am I the only one to have this stability problem?

Is there a way to recover it automatically (ie detect the failure situation and restart cc2531 and zigbee2mqtt service without human intervention)?

Koenkk, 2020-03-23:
Could you provide the debug log when it crashes?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

regissg, 2020-03-23:
Here is the file. Sorry it is rather big as it stopped working after a week.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4372100/log.txt)

thieskenl, 2020-03-23:
My CC2531 Zigbee stick stop working after reboot of my pi. After removing the stick and plug it back in it works again. Found post where you can put a usbhub with poweradapter between the stick and the pi but is not helping in my case.

Hubschra01, 2020-03-24:
Same like me...
[https://github.com/Koenkk/zigbee2mqtt/issues/3162](url)

Koenkk, 2020-03-24:
@regissg it looks that somehow the sticks disconnects, we need to dive deeper here, could you provide the log when running with zigbee herdsman debug logging?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

regissg, 2020-03-24:
hello I have the following error when I try to launch zigbee2mqtt with:
DEBUG=zigbee-herdsman* npm start

'DEBUG' is not recognized as an internal or external command,
operable program or batch file.

Koenkk, 2020-03-25:
I'm not sure how to do it on Windows,but you need to set the environment variable `DEBUG` to `zigbee-herdsman*`.

regissg, 2020-03-25:
I have added the environment variable and here is the log file just after zigbee2mqtt restarted. I do not know if the debug variable was taken into account?
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4379816/log.txt)

Koenkk, 2020-03-25:
No doesn't look like it, you should get a lot of logging after this.

regissg, 2020-03-25:
I have tried to set it by end or using the command setx DEBUG "zigbee-herdsman:*" 
As you could see it shows in the windows environment variable 
[image] but I could not see any effect in the log file 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4380043/log.txt)

Zagitta, 2020-03-26:
Seeing as you're running windows have you tried disabling power saving for the USB stick?

regissg, 2020-03-27:
@Zagitta good suggestion. I am trying it (though the PC I am using for zigbee2mqtt was set to never enter in sleep mode).

@Koenkk any hint why I could not enable herdsman debug logging?

Koenkk, 2020-03-27:
Can you try setting it as a system variable? 

Also if you run the service with NSSM I think you can also set environment variables.

regissg, 2020-03-27:
I have tried to set it up as a system as a user variable as you could see above. I have also tried to add it through NSSM that I am using to run the service and in all cases it does not work.

What is the proper value for the DEBUG variable? "zigbee-herdsman*" or "zigbee-herdsman:*" or "zigbee-herdsman *"?

Koenkk, 2020-03-27:
`zigbee-herdsman:*`

regissg, 2020-05-30:
@Zagitta> Seeing as you're running windows have you tried disabling power saving for the USB stick?

I had had no failure for the past 2 months. So this seems have fixed the problem.

Thanks

closed 2020-05-30 by regissg