Koenkk/zigbee2mqtt #3185: bridge/group/remove_all generates an error

crowbarz, 2020-03-23:
# Bug Report
## What happened
Attempt to remove_all a device from a group throws an error

## What did you expect to happen
Bulb should be removed from all groups per documentation.

## How to reproduce it (minimal and precise)
Send MQTT message:
> topic: zigbee2mqtt/bridge/group/remove_all
> payload: Attic Bedroom 3

Generated log messages:

> zigbee2mqtt:debug 2020-03-23 19:15:04: Received MQTT message on 'zigbee/bridge/group/remove_all' with data 'Attic Bedroom 3'
> zigbee2mqtt:info  2020-03-23 19:15:04: Removing 'Attic Bedroom 3' from all groups
> zigbee2mqtt:info  2020-03-23 19:15:04: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"device_group_remove_all","message":{"friendly_name":"Attic Bedroom 3"}}'
> zigbee2mqtt:info  2020-03-23 19:15:04: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"device_group_remove_all","message":{"friendly_name":"Attic Bedroom 3"}}'
> zigbee2mqtt:info  2020-03-23 19:15:04: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"device_group_remove_all","message":{"friendly_name":"Attic Bedroom 3"}}'
> zigbee2mqtt:info  2020-03-23 19:15:04: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"device_group_remove_all","message":{"friendly_name":"Attic Bedroom 3"}}'
> zigbee2mqtt:error 2020-03-23 19:15:04: Failed to call 'Groups' 'onMQTTMessage' (TypeError: group.devices is not iterable
>     at groupHasDevice (/app/lib/util/settings.js:550:32)
>     at Object.removeDeviceFromGroup (/app/lib/util/settings.js:580:17)
>     at Groups.onMQTTMessage (/app/lib/extension/groups.js:194:26)
>     at async Controller.callExtensionMethod (/app/lib/controller.js:340:21))

It's possible to work around this by manually identifying the groups that the bulb is currently a member of from groups.yaml and removing them manually from each group: that seems to work without any issues.

## Debug Info
zigbee2mqtt version: 1.11.0-dev
CC253X firmware version: 20190619

Koenkk, 2020-03-23:
Thanks, fixed in latest dev branch.

closed 2020-03-23 by Koenkk

insipiens, 2020-10-26:
Well, that was conclusive!

[image]