Koenkk/zigbee2mqtt #3187: Unable to start

unpack8866, 2020-03-24:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 
CC253X firmware version:
  
       
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.

[s6-init] ensuring user provided files have correct perms...exited 0.

[fix-attrs.d] applying ownership & permissions fixes...

[fix-attrs.d] done.

[cont-init.d] executing container initialization scripts...

[cont-init.d] done.

[services.d] starting services

[services.d] done.

./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory

[Info] Socat is DISABLED and not started

2020-03-24T10:01:21: PM2 log: Launching in no daemon mode

2020-03-24T10:01:21: PM2 log: App [npm:0] starting in -fork mode-

2020-03-24T10:01:21: PM2 log: App [npm:0] online

> zigbee2mqtt@1.12.0 start /zigbee2mqtt-1.12.0

> node index.js

zigbee2mqtt:info  2020-03-24 10:01:23: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-24.10-01-23' filename: log.txt

zigbee2mqtt:info  2020-03-24 10:01:23: Starting zigbee2mqtt version 1.12.0 (commit #unknown)

zigbee2mqtt:info  2020-03-24 10:01:23: Starting zigbee-herdsman...

zigbee2mqtt:error 2020-03-24 10:01:45: Error while starting zigbee-herdsman

zigbee2mqtt:error 2020-03-24 10:01:45: Failed to start zigbee

zigbee2mqtt:error 2020-03-24 10:01:45: Exiting...

zigbee2mqtt:error 2020-03-24 10:01:45: Error: AREQ - ZDO - activeEpRsp after 10000ms

    at Timeout._onTimeout (/zigbee2mqtt-1.12.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)

    at listOnTimeout (internal/timers.js:531:17)

    at processTimers (internal/timers.js:475:7)

npm

 ERR! code ELIFECYCLE

npm ERR! errno 1

npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`

npm ERR! Exit status 1



 

npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.

npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:

npm ERR!     /root/.npm/_logs/2020-03-24T02_01_45_963Z-debug.log

2020-03-24T10:01:48: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]

2020-03-24T10:01:48: PM2 log: App [npm:0] starting in -fork mode-

2020-03-24T10:01:48: PM2 log: App [npm:0] online

Koenkk, 2020-03-24:
What CC2531 firmware do you run?

unpack8866, 2020-03-26:
> What CC2531 firmware do you run?

20190608

zrainx, 2020-03-29:
I have the same problem, with the hardware CC2530 + CC2592 
The log is below:
> zigbee2mqtt@1.12.0 start D:\repo\Z2M\zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-29 21:48:19: Logging to console and directory: 'D:\repo\Z2M\zigbee2mqtt\data\log\2020-03-29.21-48-19' filename: log.txt
zigbee2mqtt:info  2020-03-29 21:48:20: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-03-29 21:48:20: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-03-29 21:48:27: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-29 21:48:27: Failed to start zigbee
zigbee2mqtt:error 2020-03-29 21:48:27: Exiting...
zigbee2mqtt:error 2020-03-29 21:48:27: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (D:\repo\Z2M\zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (D:\repo\Z2M\zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\zrainx\AppData\Roaming\npm-cache\_logs\2020-03-29T13_48_27_683Z-debug.log

Koenkk, 2020-03-29:
Can you provide the log when running with herdsman debug logging enabled?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

zrainx, 2020-03-30:
Hi Koenkk, thanks for your reply on time. 
The debug log is as below:

D:\repo\Z2M\zigbee2mqtt>set DEBUG=zigbee-herdsman*
D:\repo\Z2M\zigbee2mqtt>npm start

> zigbee2mqtt@1.12.0 start D:\repo\Z2M\zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-30 22:58:18: Logging to console and directory: 'D:\repo\Z2M\zigbee2mqtt\data\log\2020-03-30.22-58-18' filename: log.txt
zigbee2mqtt:debug 2020-03-30 22:58:18: Removing old log directory 'D:\repo\Z2M\zigbee2mqtt\data\log\2020-03-29.21-32-38'
zigbee2mqtt:debug 2020-03-30 22:58:19: Loaded state from file D:\repo\Z2M\zigbee2mqtt\data\state.json
zigbee2mqtt:info  2020-03-30 22:58:19: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-03-30 22:58:19: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-03-30 22:58:19: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"D:\\repo\\Z2M\\zigbee2mqtt\\data\\database.db","databaseBackupPath":"D:\\repo\\Z2M\\zigbee2mqtt\\data\\database.db.backup","backupPath":"D:\\repo\\Z2M\\zigbee2mqtt\\data\\coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"\\\\.\\COM3"}}'
  zigbee-herdsman:adapter:zStack:znp:error Failed to determine if path is valid: 'Error: EINVAL: invalid argument, lstat '\\.\COM3\'' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"\\\\.\\COM3"},"databasePath":"D:\\repo\\Z2M\\zigbee2mqtt\\data\\database.db","databaseBackupPath":"D:\\repo\\Z2M\\zigbee2mqtt\\data\\database.db.backup","backupPath":"D:\\repo\\Z2M\\zigbee2mqtt\\data\\coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with \\.\COM3 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +25ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +3ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
zigbee2mqtt:error 2020-03-30 22:58:26: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-30 22:58:26: Failed to start zigbee
zigbee2mqtt:error 2020-03-30 22:58:26: Exiting...
zigbee2mqtt:error 2020-03-30 22:58:26: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (D:\repo\Z2M\zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (D:\repo\Z2M\zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\zrainx\AppData\Roaming\npm-cache\_logs\2020-03-30T14_58_27_038Z-debug.log

Koenkk, 2020-03-30:
@zrainx this is a different issue, in your case the connection to the CC2531 is not ok, make sure to check (it's not a zigbee2mqtt bug) https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

closed 2020-06-06 by stale[bot]