Koenkk/zigbee2mqtt #3188: Fresh install

geeano, 2020-03-24:
Dear Koen,

I faced a serious number of issues after upgrading of the latest Domoticz version yesterday.
I finally decided to build up a new system by fresh install of OS and relevant software.
I was able to re-use a backup of my Domoticz database, so all informaton is restored.

My final step in installation of zigbee2mqtt. Is there any precausions ?
Do I have to connect the CC2531 USB before starting the installation or afterwards. 
Is there a risk of repairing. 

Please let me now.

Koenkk, 2020-03-24:
- Restore everything that was in the `data` directory (`database.db` and `configuration.yaml` are important here)
- Connect CC2531
- Start zigbee2mqtt

Good luck!

closed 2020-03-24 by Koenkk

geeano, 2020-03-24:
- disconnect cc2531
-  complete installation zigbee2mqtt
- stop zigbee2mqtt ( I assume ?)
- restore data directory
- connect cc2531
- start zigbee2mqtt

Please conform.
Thanks in advance