Koenkk/zigbee2mqtt #3189: device can join the network but interview_failed

fantasyjqc, 2020-03-24:
# Bug Report
## What happened
a device of model outlet16a-a1p5，when I try to pair the device , the device can successfully connect to coordinator and start interview but interview_failed 

[image]

[image]

`
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x00124b000724eb23', attempt 6 +10s
  zigbee-herdsman:controller:device Interview - failed for device '0x00124b000724eb23' with error 'Error: Interview failed because can not get node descriptor ('0x00124b000724eb23')
  zigbee-herdsman:controller:device     at Device.<anonymous> (/opt/zigbee2mqtt5/node_modules/zigbee-herdsman/dist/controller/model/device.js:265:23)
  zigbee-herdsman:controller:device     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device     at rejected (/opt/zigbee2mqtt5/node_modules/zigbee-herdsman/dist/controller/model/device.js:6:65)' +1ms
  zigbee-herdsman:controller:error Interview failed for '0x00124b000724eb23 with error 'Error: Interview failed because can not get node descriptor ('0x00124b000724eb23')' +0ms
zigbee2mqtt:error 2020-03-25 01:00:09: Failed to interview '0x00124b000724eb23', device has not successfully been paired
`

## What did you expect to happen

can successfully pair this device

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: "version":"1.10.0-dev","commit":"ad59ff5"
CC253X firmware version:cc2530-default

Koenkk, 2020-03-24:
- Could you provide an URL to the device?
- Can you sniff the traffic when pairing the device? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Westcott1, 2020-03-24:
Whenever this sort of thing happens to me, the following seems to overcome it -

Stop z2m
Update z2m if available
Unplug and re-einsert coordinator
Start z2m

New devices now interview and add correctly, at least for me!

fantasyjqc, 2020-03-26:
> * Could you provide an URL to the device?
> * Can you sniff the traffic when pairing the device? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

this is a 16A air conditioner socket produced by a small Chinese company, i cannot find its url, I can only find a introduction page in Chinese.

[http://www.znjj.tv/product/4970.html#lxwm](url)

currently I can't sniff traffic because I don't have a device like cc2531, but I have already purchased one, and I will test it when I receive the device

fantasyjqc, 2020-03-27:
> * Could you provide an URL to the device?
> * Can you sniff the traffic when pairing the device? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

i don't know why i cannot use zboss 
i use the packet sniffer to sniff traffic, when i try to pairing the device, the traffic is like this:

[image]
[image]
[image]

Koenkk, 2020-03-27:
Please use wireshark to sniff and provide the `.pcapng` file to me (this tool isn't useful as it doesn't show details about the packets).

fantasyjqc, 2020-03-27:
> Please use wireshark to sniff and provide the `.pcapng` file to me (this tool isn't useful as it doesn't show details about the packets).

this is the .pcapng file when I tried to pair the device until it failed, i tried to keep pressing the switch during pairing

[socket.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/4395777/socket.pcapng.zip)

closed 2020-06-06 by stale[bot]

countrysideboy, 2021-08-06:
Looking forward to support it.