Koenkk/zigbee2mqtt #3191: availability_timeout prevents start

Jogobo, 2020-03-25:
# Bug Report
## What happened
When I set "availability_timeout" to a value greater "0" zigbee2mqtt refuses to start with the following log messages:

Mar 25 11:08:40 xxxx npm[16584]: /usr/local/zigbee2mqtt/lib/extension/deviceAvailability.js:34
Mar 25 11:08:40 xxxx npm[16584]:             return settings.getEntity(e).ID;
Mar 25 11:08:40 xxxx npm[16584]:                                         ^
Mar 25 11:08:40 xxxx npm[16584]: TypeError: Cannot read property 'ID' of null
Mar 25 11:08:40 xxxx npm[16584]:     at DeviceAvailability.blacklist.settings.get.advanced.availability_blacklist.map(/usr/local/zigbee2mqtt/lib/extension/deviceAvailability.js:34:41)

Changing line 34 in deviceAvailability.js from
      return settings.getEntity(e).ID;
to:
      return ((settings.getEntity(e) === null) ? null : settings.getEntity(e).ID);
does the trick.

Line 39 needs to be changed the same way.

Though these changes get zigbee2mqtt working an unplugged device at startup time is shown as "online" as the logs show:

Mar 25 11:25:39 xxxx npm[18253]: ^[[32mzigbee2mqtt:info ^[[39m 2020-03-25 11:25:39: MQTT publish: topic 'zigbee2mqtt/<FRIENDLY_NAME>/availability', **payload 'online'**
Mar 25 11:25:49 xxxx npm[18253]: ^[[31mzigbee2mqtt:error^[[39m 2020-03-25 11:25:49: Failed to setup reporting for '<FRIENDLY_NAME>' - Error: Bind <FRIENDLY_NAME>/3 genOnOff from '<FRIENDLY_NAME>/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
Mar 25 11:25:49 xxxx npm[18253]:     at Endpoint.<anonymous> (/usr/local/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)

## What did you expect to happen
Startup without errors and handling deactivated devices correctly.

## How to reproduce it (minimal and precise)
Put "availability_timeout: 60" in configuration.yaml

## Debug Info
zigbee2mqtt version: 1.12.0

Koenkk, 2020-03-25:
The reason for this is that there are non existing entities on your `availability_whitelist` or `availiabilty_blacklist`. The latest zigbee2mqtt dev will now refuse to start and provide a clear error warning about this.

Regarding the online state right after start, at this time zigbee2mqtt assumes that all devices are offline, however after one iteration of the availability timeout you should see the correct state.

sjorge, 2020-03-25:
@Koenkk don't you mean online at startup?

Jogobo, 2020-03-26:
Does that mean that I have to put at least one ID/Friendly_Name in the whitelist and blacklist? As both lists try to get the ID of a NULL object in the original deviceAvailability.js and prevent zigbee2mqtt to start.

In my opinion these lists can exist independently without having impact on the functionality.

- availability_timeout without white- and blacklist: every ID has an availability_topic
- only whitelist filled: the listed IDs are the only ones which have an availability topic
- only blacklist filled: all but the blacklisted IDs have an availability topic
- blacklist and whitelist filled: makes no sense as everything is covered by the above 3 configurations

@sjorge I see "online" as payload when starting a subscriber on a disconnected device

Koenkk, 2020-03-26:
@Jogobo can you share your `configuration.yaml`? (without sensitive information)

Jogobo, 2020-03-26:
Here you go...
```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.xxx.yyy.zzz'
  user: ...
  password: ...
serial:
  port: /dev/ttyZZZZ
advanced:
  availability_timeout: 60
  availability_blacklist:
    - ''
  availability_whitelist:
    - 'mqtt_Licht_Flur_klein'
  report: true
  log_level: debug
  log_output:
    - console
  network_key: <some private key>
experimental:
  output: json
devices:
  '0x7zzzzzzzzzzzzzzz':
    friendly_name: 'Testlicht'
  '0x8zzzzzzzzzzzzzzz':
    friendly_name: 'mqtt_Licht_Flur_klein'

```
I'm still testing with little devices before I start changing from Osram Lightify Gateway to zigbee2mqtt.

Koenkk, 2020-03-26:
I expect it to work when removing this from your configuration:

```yaml
  availability_blacklist:
    - ''
```

Jogobo, 2020-03-26:
Switched back to original deviceAvailability.js and removed 
```
availability_blacklist: 
  - ''
```
from configuration.yaml. Surprisingly zigbee2mqtt starts normal. Even when I remove 
```
availability_whitelist: 
  - '...'
```
from configuration.yaml zigbee2mqtt starts. That was not the case yesterday. Only difference is that I now have 2 devices configured instead of only 1 device until yesterday.

Strange...

Koenkk, 2020-03-27:
It's expected that it works after that (I only added a check to make sure the configuration is valid).

Can I consider this issue fixed?

Jogobo, 2020-03-27:
I just checked with only availability_timeout set, no whitelist, no blacklist. It works now.

closed 2020-03-27 by Jogobo