Koenkk/zigbee2mqtt #3192: Can anyone confirm out-of-the-box compatibility with ITEAD CC2531 USB-dongle?

Hedda, 2020-03-25:
You have probably already heard, but first want to inform that ITEAD has started selling this CC2531 USB-dongle which they write is pre-flashed with an (unknown) coordinator firmware for only $4 (US).

- https://www.itead.cc/cc2531-usb-dongle.html

Wondering if anyone working on Zigbee2mqtt has already got one and can confirm for sure that the pre-flashed firmware it comes with is plug-and-play compatible with Zigbee2mqtt out-of-the-box?

I'm not a native English speaker and clearly neither are they as their page about this Zigbee USB-dongle is very poorly written, however taking their meaning, they claim on their website that it is plug-and-play compatible with Zigbee2mqtt and should work with it directly out-of-the-box without first having to flash/reflash the dongle.

And if someone who has one can also somehow check if ITEAD is using their own firmware config or if they are just flashed Koenkk's default firmware? https://github.com/Koenkk/Z-Stack-firmware

If you contact ITEAD then maybe they would be willing to donate a bunch of them to both your team-developers and team-testers since they are even marketing it as Zigbee2mqtt compatible?

It's a good price for a preflashed dongle so I'm asking because wondering if I should recommend this to online friends (who's technical skills warry) so that they won't need to buy a CC-debugger.

Somewhat related is that someone has created a compatibility list for Zigbee coordinators and devices at zigbee.blakadder.com and they have already listed that as Zigbee2mqtt compatible here:

- https://zigbee.blakadder.com/coordinators.html

Not sure if the people who run zigbee.blakadder.com actually tested it or just read ITEAD's page?

johras, 2020-03-25:
Looks like any other CC2531 stick.

CC-debugger is not necessary for flashing. There are many other easy ways to do that.
I did the flashing by connecting the CC2531 to Raspberry's USB and GPIO headers.

[image]

https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html

Hedda, 2020-03-31:
@johras sorry but that does not actually answer my question. What I am specifically asking if someone who bought one of these from ITEAD can confirm if they already pre-flashed with compatible firmware. And as a follow-up question to that, if they are already preflashed with a compatible firmware then I'm also asking if someone who bought one of these from ITEAD can also verify if they are just using the default firmware from Koenkk or if they have made a custom config of their own.

johras, 2020-03-31:
Just wanted to point out that no CC-debugger needed so it's cheap to try it out.

ITEAD may have updated their product description because they are now claiming that
"Yup, it is a ready-to-go CC2531 USB dongle for zigbee2mqtt application"

and

 "As ITEAD has already done the firmware flash of CC2531ZNP-Prod for CC2531 USB dongle, you can directly start with the section of 'Running Zigbee2mqtt'"

Hedda, 2020-03-31:
OK but suppose =! confirm or verify. I already have several myself so I'm "asking for a friend" now :P

> It's a good price for a preflashed dongle so I'm asking because wondering if I should recommend this to online friends (who's technical skills warry) so that they won't need to buy a CC-debugger.

I don't want to recommend this to a bunch of online friends without CC-debuggers until it has been confirmed to definitely work out-of-the-box with Zigbee2mqtt, even if is considered to be dirt cheap by some.

blakadder, 2020-03-31:
CC debugger isn't the only way to flash a CC253x... You can do it on the cheap with a Wemos or NodeMCU and CCLoader (https://zigbee.blakadder.com/flashing_ccloader) and there's the CClib route described in zigbee2mqtt docs

Hedda, 2020-06-04:
FYI, more than one person now confirmed that it works but ships with old firmware from "20180507"

- https://github.com/zha-ng/zigpy-znp/issues/14

- https://community.home-assistant.io/t/itead-of-sonoff-fame-now-sells-a-very-inexpensive-zigbee-usb-dongle-that-should-work-out-of-the-box-with-both-zigbee2mqtt-and-the-zha-integration/181085/5

closed 2020-07-26 by stale[bot]

createcandle, 2021-03-30:
It works, but only to a degree. In my case not all properties came through, so I wouldn't recommend it without flashing a new firmware.

Also, keep an eye on the new Zigbee 3.0 Itead USB stick.
https://www.itead.cc/zigbee-3-0-usb-dongle.html