Koenkk/zigbee2mqtt #3194: Could not locate the bindings file.

Frenkie64, 2020-03-25:
I'm trying to get zigbe2mqtt working. It hasn't been easy so far, but slowly I'm getting there. MQTT broker is up, node-red is up, lan interface in domoticz is up, but I can't seem to get the zigbee2mqtt running.
I tried installing and reinstalling a few times without success. To get the picture clear, I'm runninc Domoticz on a Synology with a CC2351 stick and a AEOTEC stick. Z-Wave is working as a charm, but I want to expand my network with zigbee devices.
As I am running on a Synology, I ran into issues with Python, Git, Npm, NodeJS, etc.
Last problem I have is that zigbee2mqtt wants to run with NodeJS version 10 or higher. As soon as I install that, I have issues with zlib (and I'm unsure how to update that on Synology), so I stay on NodeJS 8 for now. Installation seems to work, though it takes about 20 minutes.
When I run it, I get this log:

`zigbee2mqtt:info  2020-03-25 14:33:50: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-25.14-33-49' filename: log.txt
zigbee2mqtt:info  2020-03-25 14:33:53: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-03-25 14:33:53: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-03-25 14:33:53: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-25 14:33:53: Failed to start zigbee
zigbee2mqtt:error 2020-03-25 14:33:53: Exiting...
zigbee2mqtt:error 2020-03-25 14:33:53: Error: Could not locate the bindings file. Tried:
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/out/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/default/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/compiled/12.14.0/linux/arm/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/binding/node-v72-linux-arm/bindings.node
    at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/index.js:14:22)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-03-25T13_33_54_037Z-debug.log`

How can I fix the bindings.node issue?

romaint2b, 2020-03-30:
Hi,
I've got the same issue.
I found another topic here and try to install indicated npm packages but i've got the same error.
https://github.com/Koenkk/zigbee2mqtt/issues/2412

Cmd :
npm install bindings --save
npm install node-gyp
npm install libxmljs --> Install hasn't worked
npm install
npm start
...
 did you find the solution?

romaint2b, 2020-03-31:
after spending many hours, I reinstall g++ which cause a compilation error for bindings module.using npm start works for now.next step : correct the service to use node 12 (it use node 10 for the moment).

closed 2020-06-06 by stale[bot]