Koenkk/zigbee2mqtt #3195: Aurora GU10 Bulb Support

mantis64, 2020-03-25:
I bought some GU10 bulbs from Aurora Lighting.  Can anyone please help me to make these work with Zigbee2mqtt and home assistant?

## What happened
Device cannot pair.  
zigbee2mqtt:warn  2020-03-25 17:08:46: Received message from unsupported device with Zigbee model 'FWGU10Bulb50AU'
zigbee2mqtt:warn  2020-03-25 17:08:46: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

## What did you expect to happen
Device to pair

## How to reproduce it (minimal and precise)
For it to work.

## Debug Info
zigbee2mqtt version: 1:12
CC253X firmware version:

mantis64, 2020-03-25:
Also I did try and follow the link to support new devices, however it does not make much sense to me.  Please can you help.  I don't seem to have a devices.js file that the link talks about.  I am using Hassio on a RPi3b.  Please help

sjorge, 2020-03-25:
Can you post the database.db entry for the device?

mantis64, 2020-03-25:
Hi, thanks for coming back to me:
Here is the database entry:

{"id":30,"type":"Router","ieeeAddr":"0x90fd9ffffe52a2b6","nwkAddr":31916,"manufId":4456,"manufName":"Aurora","powerSource":"Mains (single phase)","modelId":"FWGU10Bulb50AU","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"FWGU10Bulb50AU","manufacturerName":"Aurora","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180103","swBuildId":"2.0"}}},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180103","swBuildId":"2.0","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1585168398371}

sjorge, 2020-03-25:
Can't figure out if it is a WS or RGB bulb... ginve that there website says my country (belgium) is banned :(

If it is RGB try adding this to devices.js
```
    {
        zigbeeModel: ['FWGU10Bulb50AU'],
        model: 'FWGU10Bulb50AU',
        vendor: 'Aurora',
        description: 'Auorora Lights GU10',
        extend: generic.light_onoff_brightness_colorxy,
    },
```

If it is a WS one try with
```
    {
        zigbeeModel: ['FWGU10Bulb50AU'],
        model: 'FWGU10Bulb50AU',
        vendor: 'Aurora',
        description: 'Auorora Lights GU10',
        extend: generic.light_onoff_brightness_colortemp,
    },
```

mantis64, 2020-03-25:
Hi, yes I figured that out, its a white spectrum light and I copied a similar GU10 bulb I have from Ikea.  I can get it to pair and it says in is supported.  However now I don't know how I can get it to show up in my entities and devices.  It does not show up in MQTT broker discovery.  Please can you help me with this?

sjorge, 2020-03-25:
You need to add them to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L987 too I think, just do the same and copy an ikea one for a WS bulb.

The key is the 'model' entry from devices.js

mantis64, 2020-03-25:
Hi, thanks for this, could you give me a little more info on what I need to do.  Do I need to copy this file and place it somewhere like I did with the devices.js

mantis64, 2020-03-26:
Hi, can you please clarify on what I need to do.  I can control the bulb if I manually add this to my HA configuration.yaml, and it turned out that it was only a white bulb so can only dim it.  However I have another bulb which in tunable.  

This is what I did so perhaps you can help me to add it into MQTT Discovery so it is a proper entity that I can control with automations etc..

1) I paired the bulb.  
2) From Zigbee2Mqtt I got the model name 'TWGU10Bulb50AU'
3) I added devices.js file from Github to my \\hassio\share\zigbee2mqtt location
4) I opened devices.js and copied a similar bulb and changed the model to my bulb model 'TWGU10Bulb50AU'
5) I then added this to my HA configuration.yaml
light:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/<FRIENDLY_NAME>"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    color_temp: true
    schema: "json"
    command_topic: "zigbee2mqtt/<FRIENDLY_NAME>/set"

So now I can add this device in Lovelace and can control it via the lovelace UI. But I cannot do any automations with it since it is not a properly discovered entity.  

Please can you explain on what I need to do to make it a proper entity that I can use in HA for automations.  

Many thanks

Mantis

mantis64, 2020-03-30:
Can someone please help me?

Koenkk, 2020-03-30:
What does this bulb support? 

- brightness?
- color temperature?
- color (RGB)?

mantis64, 2020-03-30:
Hi, thanks for responding :)

It supports brightness and colour temp only.  I have another 2 types of GU10 bulbs that are unsupported too.  How can I add support for them?

The one that I am testing now 'TWGU10Bulb50AU', if I add to my configuration yaml I am able to control it via lovelace.  But I cannot add it to automations etc....  

How can I add this to the MQTT broker so I am able to use it please?

Koenkk, 2020-03-30:
For all devices that you want to add:
- Provide an url (or picture) of the device
- Provide the entry in `database.db`
- Supported features (https://github.com/Koenkk/zigbee2mqtt/issues/3195#issuecomment-606142187)

mantis64, 2020-03-30:
Thats great, thank you. I will this over the next few days.  I have around 2-4 unsupported GU10 bulbs. 

Presumably you are adding to the master database so I will not have to do anything more than just update Z2M?

For the future for one-off devices, how can I add these myself to my installation of HA on my RPi3 running hassio? 

Many thanks again and hope you are keeping safe in these crazy times.

mantis64, 2020-04-09:
Dear @Koenkk here is my list of bulbs, I would really appreciate if you could support these.  

If you need further info do let me know and thanks in advance.  

-------------------------------------------------------------------

**Description**
Leedarson 7ZB-M350ST-W1Z GU10 White Tuneable
2700-6500K

**Supported Features**
White Spectrum Tuneable and Dimmable

**Pairing**
Off On 7 times
Will flash twice to indicate in pairing mode.  Will flash once to confirm paired

**Zigbee2Mqtt**
Device '0x00158d000210435c' with Zigbee model 'TWGU10Bulb02UK' is NOT supported

**Database**
{"id":41,"type":"Router","ieeeAddr":"0x00158d000210435c","nwkAddr":19457,"manufId":4456,"manufName":"Aurora","powerSource":"Mains (single phase)","modelId":"TWGU10Bulb02UK","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,4,3,6,8,5,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TWGU10Bulb02UK","manufacturerName":"Aurora","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180418","swBuildId":"2.0"}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180418","swBuildId":"2.0","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1586362317331}

**Weblink & Picture**
[https://www.leedarson.com/iot.php/Product/detail/id/208](https://www.leedarson.com/iot.php/Product/detail/id/208)

-------------------------------------------------------------------------------------

**Description**
Aurora AU-A1GUZBCX5 GU10 White Tuneable
2200-5000K

**Supported Features**
White Spectrum Tuneable and Dimmable

**Pairing**
Off On 7 times
Will flash twice to indicate in pairing mode.  Will flash once to confirm paired

**Zigbee2Mqtt**
Device ' 0x90fd9ffffe65186c' with Zigbee model ' TWGU10Bulb50AU' is NOT supported

**Database**
{"id":43,"type":"Router","ieeeAddr":"0x90fd9ffffe65186c","nwkAddr":41395,"manufId":4456,"manufName":"Aurora","powerSource":"Mains (single phase)","modelId":"TWGU10Bulb50AU","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096,65244],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"TWGU10Bulb50AU","manufacturerName":"Aurora","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180821","swBuildId":"2.2"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180821","swBuildId":"2.2","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1586363297446}":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180418","swBuildId":"2.0"}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180418","swBuildId":"2.0","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1586362317331}

**Weblink & Picture**
[https://auroralighting.com/gb/ProductDetail/AU-A1GUZBCX5](https://auroralighting.com/gb/ProductDetail/AU-A1GUZBCX5)

---------------------------------------------------------------------------------------

**Description**
Aurora AU-A1GUZB5/30 GU10 Warm White Dimmable
3000k

**Supported Features**
White and Dimmable

**Pairing**
Off On 7 times
Will flash twice to indicate in pairing mode.  Will flash once to confirm paired

**Zigbee2Mqtt**
Device ' 0x90fd9ffffe64f994' with Zigbee model ' FWGU10Bulb50AU' is NOT supported

**Database**
{"id":45,"type":"Router","ieeeAddr":"0x90fd9ffffe64f994","nwkAddr":32362,"manufId":4456,"manufName":"Aurora","powerSource":"Mains (single phase)","modelId":"FWGU10Bulb50AU","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096,65244],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"FWGU10Bulb50AU","manufacturerName":"Aurora","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180821","swBuildId":"2.2"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180821","swBuildId":"2.2","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1586375867685}

**Weblink & Picture**
[https://www.smartthings.com/gb/products/aurora-aone-48w-smart-dimmable-gu10-lamp-3000k](https://www.smartthings.com/gb/products/aurora-aone-48w-smart-dimmable-gu10-lamp-3000k)

[... 60 lines cut ...]

**Weblink & Picture**
[https://auroralighting.com/gb/ProductDetail/AU-A1GSZ9RGBW](https://auroralighting.com/gb/ProductDetail/AU-A1GSZ9RGBW)

----------------------------------------------------------------------------------

**Description**
Aurora 9.5W SMART RGBW GLS LAMP E27
2700k - 6500k

**Supported Features**
RGB Tuneable and Dimmable

**Pairing**
Off On 8 times
Will flash twice to indicate in pairing mode.  Will flash once to confirm paired

**Zigbee2Mqtt**
Device '0x00158d000180178c' with Zigbee model 'RGBBulb02UK' is NOT supported

**Database**
{"id":40,"type":"Router","ieeeAddr":"0x00158d000180178c","nwkAddr":4977,"manufId":4456,"manufName":"Aurora","powerSource":"Mains (single phase)","modelId":"RGBBulb02UK","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,4,3,6,8,5,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"RGBBulb02UK","manufacturerName":"Aurora","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20181207","swBuildId":"2.0"}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20181207","swBuildId":"2.0","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1586427872447}

**Weblink & Picture**
[https://auroralighting.com/gb/ProductDetail/AU-A1GSZ9RGBW](https://auroralighting.com/gb/ProductDetail/AU-A1GSZ9RGBW)

Koenkk, 2020-04-10:
@mantis64 thanks for the good information. All are supported now in the latest dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

mantis64, 2020-04-10:
Dear @Koenkk thanks so much for supporting these devices.  I have enabled the edge version on Hassio but still get this error:

Received message from unsupported device with Zigbee model 'TWGU10Bulb50AU'

I will try restarting the RPi3.  

Will it be supported in the main branch in the next release.

Thanks again and stay safe. :)

sjorge, 2020-04-10:
The Hassio Edge version lags behind a few hours I think.

mantis64, 2020-04-10:
@sjorge ok thanks for that.  It's still not working and I get the same error.  I will try tomorrow.  :)

mantis64, 2020-04-11:
All working now,  thanks again @Koenkk and @sjorge

closed 2020-04-11 by mantis64