Koenkk/zigbee2mqtt #3196: 205 error, but reachable via broadcast (groups)

tricoos, 2020-03-25:
# Bug Report
## What happened
Having suffered from 205 errors for a long time which just have almost completely disappeared after switching to groups I now had it happen that the same device was controllable via the group but returned a 205 error in Zigbee2MQTT.

## What did you expect to happen
No 205 errors at all at anytime.

## How to reproduce it (minimal and precise)
Probably has to to with the IKEA lamps (FLOALT in this case). This is a known issue that has been reported here several times. I have 50+ zigbee devices.

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: 20190608 (source routing)

The group 'light_hallway' consist only of the devices 'ikeatradfri01' and 'ikeatradfri03'!
What's interesting: even if ikeatradfri03 was reported as not being in the network (205) it correctly reported its status after being turned on/off via the group:
```
zigbee2mqtt:info  2020-03-25 21:07:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ikeatradfri03' failed: 'Error: Command 0x000d6ffffe164546/1 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ikeatradfri03"}}'

zigbee2mqtt:error 2020-03-25 21:07:09: Publish 'set' 'state' to 'ikeatradfri03' failed: 'Error: Command 0x000d6ffffe164546/1 genOnOff.on({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'

zigbee2mqtt:info  2020-03-25 21:07:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ikeatradfri03' failed: 'Error: Command 0x000d6ffffe164546/1 genOnOff.on({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ikeatradfri03"}}'

zigbee2mqtt:info  2020-03-25 21:07:15: MQTT publish: topic 'zigbee2mqtt/ikeatradfri01', payload '{"state":"ON","brightness":178.5,"color_temp":295,"update_available":false}'

zigbee2mqtt:info  2020-03-25 21:07:15: MQTT publish: topic 'zigbee2mqtt/light_hallway', payload '{"state":"ON","brightness":178.5,"color_temp":295}'

zigbee2mqtt:info  2020-03-25 21:07:15: MQTT publish: topic 'zigbee2mqtt/ikeatradfri03', payload '{"state":"ON","brightness":178.5,"color_temp":295,"update_available":false}'

zigbee2mqtt:info  2020-03-25 21:07:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ikeatradfri03' failed: 'Error: Command 0x000d6ffffe164546/1 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ikeatradfri03"}}'

zigbee2mqtt:info  2020-03-25 21:08:13: MQTT publish: topic 'zigbee2mqtt/ikeatradfri01', payload '{"state":"OFF","brightness":178.5,"color_temp":295,"update_available":false}'

zigbee2mqtt:info  2020-03-25 21:08:13: MQTT publish: topic 'zigbee2mqtt/light_hallway', payload '{"state":"OFF","brightness":178.5,"color_temp":295}'

zigbee2mqtt:info  2020-03-25 21:08:13: MQTT publish: topic 'zigbee2mqtt/ikeatradfri03', payload '{"state":"OFF","brightness":178.5,"color_temp":295,"update_available":false}'
```
Also, ikeatradfri01 reports its status before the cause of the status change 'light_hallway' has been logged.

sjorge, 2020-03-25:
Out of interest, did you start experiencing these error after updating your IKEA devices?

tricoos, 2020-03-26:
No, there never were any updates available for the IKEA FLOALT lamps.

turshija, 2020-03-26:
I have same error, but I don't have IKEA devices.

`zigbee2mqtt:error 2020-03-26 19:37:44: Publish 'set' 'state' to '0x00124b001d3ab041' failed: 'Error: Command 0x00124b001d3ab041/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Data request failed with error: 'No network route' (205))'`

My device: [Gledopto LED strip](https://www.zigbee2mqtt.io/devices/GL-C-007_GL-C-008.html)

I started getting these errors since 2 days ago. I switched my cc2531 adapter with new one (with built-in antenna), flashed same firmware on it, and while I was already doing the switch I've updated hassio-zigbee2mqtt in Home Assistant to 1.12.0. Unfortunately I'm not sure what was my previous version but I'm sure I haven't updated it in a month or two (I don't know how to check, I have 5 week old snapshot of entire Home assistant, maybe I can dig it out).

tricoos, 2020-03-27:
I've had these errors actually for months before, but I created this ticket specifically to point of the difference between lights being reachable via groups and even reporting their status and non-reachability via the normal network *at the same time*. Maybe @Koenkk has an idea?

Koenkk, 2020-03-27:
- Groups messages are broadcasted over the whole network, meaning that every device will receive it and check if it's in that particular group. If it is it applies the command.
- Commands to a specific device are send to a device directly, or when it's not a range via a few devices (= a route). If this route is not available it will lead to this error.
- In case you have this error the following things can be done
    - Make sure you are on zigbee2mqtt 1.12.0, it contains some improvements related to this
    - Switch to the source routing firmware, this firmware is able to remember routes meaning it doesn't have to query it each time. In your OP you mention that you are (`CC253X firmware version: 20190608 (source routing)`, but `20190608` is not a source routing firmware [`20190619`](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing) is. So make sure that you actually are.
   - But even with this firmware, due to hardware limitations of the CC2531, it is only able to store up to 40 routes. With so much devices I would recommend upgrading to a more powerful adapter (e.g. [CC26XR1](https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc26x2r1)) as it will easily handle your Zigbee network (it can store up to 200 routes).

closed 2020-06-06 by stale[bot]