Koenkk/zigbee2mqtt #3199: Crash after upgrading to 1.12.0-dev

cbrandlehner, 2020-03-25:
# Bug Report
## What happened
I had 0.11.0 installed, working fine. After the upgrade to 0.12.0-dev the service fails to start.

## What did you expect to happen
Service startup

## How to reproduce it (minimal and precise)
Upgrade from 0.11.0 to 0.12.0-dev

## LOG
[LOG](https://hastebin.com/usigahoxom.coffeescript)

## Debug Info
zigbee2mqtt version: 0.12.0-dev
CC253X firmware version: 20190608

cbrandlehner, 2020-03-25:
I downgraded to 0.12.0 (not 0.12.0-dev) and the service starts so the problem seems to be with the last dev version only.

Koenkk, 2020-03-26:
`Error: Non-existing entity 'DEVICE_FRIENDLY_NAME' specified in 'availability_blacklist'`

It looks that there is a non-existing entity in your `availability_blacklist`?

Jogobo, 2020-03-26:
Should be the same reason as [here](https://github.com/Koenkk/zigbee2mqtt/issues/3191). If you disable 'availability_timeout' in your configuration.yaml the service should start.
You can apply the changes described in the other thread to your 'deviceAvailability.js' and your going to be fine for now.

cbrandlehner, 2020-03-26:
I am already fine as my configuration currently works with version 0.12.0.

As @Koenkk mentioned it seems to be related to 'availability_blacklist'. I tried to change the value from friendly name to ID and also tried to completely remove the parameter but still had issues. Probably my fault, I am not familiar with yaml syntax.

This is my configuration.yaml

`homeassistant: false
permit_join: false
mqtt:
  base_topic: rpi3
  server: 'mqtt://diskstation.c2home.brandlehner.at:1883'
  include_device_information: true
serial:
  port: /dev/ttyACM0
  disable_led: false
advanced:
  pan_id: 6754
  ext_pan_id:
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
  channel: 15
  cache_state: true
  log_level: info
  log_directory: data/log/%TIMESTAMP%
  baudrate: 115200
  rtscts: true
  soft_reset_timeout: 0
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  last_seen: ISO_8601_local
  elapsed: true
  availability_timeout: 0
  availability_blacklist:
    - DEVICE_FRIENDLY_NAME
  report: true
map_options:
  graphviz:
    colors:
      fill:
        enddevice: '#fff8ce'
        coordinator: '#e04e5d'
        router: '#4ea3e0'
      font:
        coordinator: '#ffffff'
        router: '#ffffff'
        enddevice: '#000000'
      line:
        active: '#009900'
        inactive: '#994444'
queue:
  delay: 250
  simultaneously: 5
devices:
  '0x00158d0003408603':
    friendly_name: Temp-Schlafzimmer
    retain: false
  '0x00158d0003eef902':
    friendly_name: Door-Terrasse-rechts
    retain: false
  '0x00158d0003f114a8':
    friendly_name: Temp-Gast
    retain: false
  '0x00158d00044a02a5':
    friendly_name: Temp-Kueche
    retain: false
  '0x00158d0003cd224b':
    friendly_name: Door-Balkon
    retain: false
  '0x00158d0003cd3e37':
    friendly_name: Door-Terrasse-links
    retain: false
  '0x00158d0003a2896c':
    friendly_name: Temp-Buero
    retain: false
  '0x00158d0002879994':
    friendly_name: Wassersensor
    retain: false
  '0x00158d00040175cd':
    friendly_name: Fenster-Buero
  '0x00158d000446f6b5':
    friendly_name: Door-Haustuer`

Koenkk, 2020-03-26:
You should remove this from your `configuration.yaml`

```yaml
  availability_blacklist:
    - DEVICE_FRIENDLY_NAME
```

closed 2020-06-02 by stale[bot]