Koenkk/zigbee2mqtt #3200: No status updates for switch TUYATEC GDKES-03TZXD

IronButterfly, 2020-03-26:
Hi,

z2m 1.12.0
USB stick CC2538

I have added these wall switches https://www.zigbee2mqtt.io/devices/GDKES-03TZXD.html without any problem.
The problem in that no status changes after switching light manually. No switches status updates in HomeAssistant.
Any ideas how this can be fixed?

Thanks.

IronButterfly, 2020-03-27:
Repairing switch solved the issue.

alting, 2020-04-24:
I'm have the same issue with GDKES-03TZXD and GDKES-02TZXD
Repairing helps, but not for long

jlweston, 2020-06-02:
Possibly related but can move this to a separate issue if appropriate.

I'm using the 2-gang version (GDKES-02TZXD) and it often fails to respond to commands sent to either topic (`/left/set` or `/right/set`). However, when the switch is toggled manually, the JSON correctly updates to show the current `left_state` and `right_state`.

As commented above, re-pairing works but only for a short while (a few hours until the problem is apparent again).

Koenkk, 2020-06-03:
Let's first fix the OP issue, do you get state updates when updating the definition in `devices.js` to:

```js
{
    zigbeeModel: ['TS0013'],
    model: 'GDKES-03TZXD',
    vendor: 'TUYATEC',
    description: 'Smart light switch - 3 gang without neutral wire',
    supports: 'on/off',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    meta: {configureKey: 2, multiEndpoint: true},
    configure: async (device, coordinatorEndpoint) => {
        for (const ID of [1, 2, 3]) {
            const endpoint = device.getEndpoint(ID);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint);
        }
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to get to `devices.js`

IronButterfly, 2020-06-08:
> Let's first fix the OP issue, do you get state updates when updating the definition in `devices.js` 

Tried this in z2m docker of my Hassio. Works fine. I didn't notice any difference.

Koenkk, 2020-06-09:
So you get status updates now when toggling manually?

IronButterfly, 2020-06-10:
Yes, works fine.

rpaluba, 2020-06-11:
Can anyone tell me how to add GDKES-02TZXD to domoticz? I am using zigbee usb cc2531.
In the zigbee2tmqq logs blow events when I press the button manually, but I don't know how to use it in domotics.

I added devices.js to the file
"
{
        zigbeeModel: ['TS0012'],
        model: 'GDKES-02TZXD',
        vendor: 'TUYATEC',
        description: 'Smart light switch - 2 gang without neutral wire',
        supports: 'on / off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'left': 1, 'right': 2};
        }
        meta: {configureKey: 1, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint) => {
            await bind (device.getEndpoint (1), coordinatorEndpoint, ['genOnOff']);
            await bind (device.getEndpoint (2), coordinatorEndpoint, ['genOnOff']);
        }
    }

"
I get bugs
Jun 11 13:10:59 raspberrypi npm [1151]: zigbee2mqtt: error 2020-06-11 13:10:59: Failed to configure '0xec1bbdfffec94ff9', attempt 1 (Error: Bind 0xec1bbdfffec94ff9 / 1 genBasic from '0x00124b001cce3380 / 1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
Jun 11 13:10:59 raspberrypi npm [1151]: at Endpoint. <anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
Jun 11 13:10:59 raspberrypi npm [1151]: at Generator.throw (<anonymous>)
Jun 11 13:10:59 raspberrypi npm [1151]: at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
Jun 11 13:10:59 raspberrypi npm [1151]: zigbee2mqtt: info 6/6/2020 13:10:59: Configuring '1234567'
Jun 11 13:11:09 raspberrypi npm [1151]: zigbee2mqtt: error 2020-06-11 13:11:09: Failed to configure '1234567', attempt 1 (Error: Bind 0xec1bbdfffe91e8ad / 1 genBasic from '0x00124b001cce3380 / 1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
Jun 11 13:11:09 raspberrypi npm [1151]: at Endpoint. <anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
Jun 11 13:11:09 raspberrypi npm [1151]: at Generator.throw (<anonymous>)
Jun 11 13:11:09 raspberrypi npm [1151]: at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

I can't configure the device in Domoticz, I can only see it in my own zigbemqtt.

Please advise, I can pay.