Koenkk/zigbee2mqtt #3201: Support for repeated click/press on remotes

ellnic, 2020-03-26:
I've not long been using Z2MQTT, so firstly thanks for all your work on this. 

I was just setting up a couple of Ikea remotes and, coming from deCONZ, I immediately noticed that there was not the option for continuous clicks. I'm not sure if it's different for other switches, but in deCONZ for a 2 button Ikea remote I get:

1. Single click
2. Press and hold
3. Repeated click

So 6 possible actions

In Z2MQTT, I get:

1. On hold
2. On release
3. Single click

So 4 possible actions - because the hold and release action can't both be used.

Are there any plans to add continuous clicks? It would be great to have more button combinations. :-)

sjorge, 2020-03-26:
I think these are the only events the remote sends itself, you can probably handle the repeated click in node-red or home assistant (not sure, don't use the later)

ellnic, 2020-03-26:
Yeah I suppose NR is an option, but I thought it might be an easy addition from within Z2MQTT itself?

sjorge, 2020-03-26:
I guess you could fake this in the converter, but for consistency you then need to do it for all remotes... which is extra maintenance.

Now z2m supports what the device support without fancy stuff on top. It also might break existing work flows that say count clicks and do different actions based on the amount etc...

I guess you could emit both mqtt messages to avoid that I suppose.

~ sjorge

ellnic, 2020-03-26:
Ok, thanks. I guess this is answered then. :)

closed 2020-03-26 by ellnic