Koenkk/zigbee2mqtt #3206: Support for Osram/Sylvania Gardenspot RBG Lights

Botched1, 2020-03-27:
OK, I am new to zigbee2mqtt, but am tryin to add my garden lights to the system. They were listed as an unsupported device on pairing.

So, I copied one of the other Osram Garden RGB lights, and renamed it to my device. That made it show up as supported - which is good. STATE and BRIGHTNESS work, buy annoyingly corlar cahnging with COLOR_TEMP does not.

Is there anything I can do to try and get color changing working?

Here is what the device looks like in my datrabase.db:
{"id":6,"type":"Router","ieeeAddr":"0xf0d1b80000078a36","nwkAddr":9762,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"Outdoor Accent Light RGB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768,2821,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Outdoor Accent Light RGB","manufacturerName":"LEDVANCE","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":0,"dateCode":"Jan  8 2019\u0000\u0000\u0000\u0000\u0000"}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":0,"dateCode":"Jan  8 2019\u0000\u0000\u0000\u0000\u0000","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1585271772427}

This is the device - or at least that is what it looks like. Maybe the model is slightly different (mine was listed as 75541 as it came in a 'starter kit', not 73699)?
https://zigbee.blakadder.com/OSRAM_73699.html

jrc1883, 2020-03-27:
I put this in devices.js

`    {
        zigbeeModel: ['Outdoor Accent RGB'],
        model: '75541',
        vendor: 'Sylvania',
        description: 'SMART+ Outdoor Accent RGB lighting kit',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },`

cleared the database and restarted. paired right up.

jrc1883, 2020-03-27:
I got mine working just now. Here are my relevant settings. Sorry for the formatting of the HSBtoRGB.js. 

my devices.js:
`    {
        zigbeeModel: ['Outdoor Accent RGB'],
        model: '75541',
        vendor: 'Sylvania',
        description: 'SMART+ Outdoor Accent RGB lighting kit',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },`

my .things:
`    Thing mqtt:topic:Garden "Garden" (mqtt:broker:Home) {
        Type dimmer : dimmer "Garden Dimmer" [ stateTopic = "zigbee2mqtt/Garden/dimmer", commandTopic = "zigbee2mqtt/Garden/set/brightness", min=0, max=255, step=1 ]
Type colorHSB : color "Garden Color" [ stateTopic = "zigbee2mqtt/Garden/dumb", commandTopic = "zigbee2mqtt/Garden/set", transformationPatternOut = "JS:HSBtoRGB.js", onBrightness = 100 ]
    }`

my .items:
`Dimmer Garden_Dimmer "Garden [%s]" { channel="mqtt:topic:Garden:dimmer" }
Color  Garden_color "Garden Color" {channel="mqtt:topic:Garden:color"}`

my configuration.yaml:
`experimental:
  output: attribute`

my HSBtoRGB.js:
`
(function (x) {
    if (x=="ON" || x=="INCREASE") {
        return '{"state":"ON","brightness":250}'
    } else if (x=="OFF" || x=="DECREASE") {
        return '{"state":"OFF"}'
    } else {
        var tex = x.split(',');
        s = parseInt(tex[1]);
        v = parseInt(tex[2]);
        h = parseInt(tex[0]);
    
        h=(!h ? 0 : h/360.0);
        s=(!s ? 0 : s/100.0);
        v=(!v ? 0 : v/100.0);

        i = Math.floor(h * 6);
        f = h * 6 - i;
        p = v * (1 - s);
        q = v * (1 - f * s);
        t = v * (1 - (1 - f) * s);
        switch (i % 6) {
            case 0: r = v, g = t, b = p; break;
            case 1: r = q, g = v, b = p; break;
            case 2: r = p, g = v, b = t; break;
            case 3: r = p, g = q, b = v; break;
            case 4: r = t, g = p, b = v; break;
            case 5: r = v, g = p, b = q; break;
        }
        return '{"brightness":'+ Math.round(v * 255)+',"color":{"rgb":"'+ Math.round(r * 255)+','+ Math.round(g * 255)+','+Math.round(b * 255)+'"}}'
    }
})(input)
`

Koenkk, 2020-03-27:
@Botched1 your zigbeeModel was slightly different, I've added it now. Will be supported in the latest zigbee2mqtt dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-03-27 by Koenkk

Botched1, 2020-03-27:
> @Botched1 your zigbeeModel was slightly different, I've added it now. Will be supported in the latest zigbee2mqtt dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Thanks!

Quick question I've been googling this morning, but haven't found an answer for yet. When new updates come out do you need to manually git clone the herdsman files over again (I'm using docker - latest-dev)? 

If so, what do people do to maintain devices.js customizations?

Koenkk, 2020-03-27:
Yes, you need to merge the changes every time. Therefore it's recommend to deliver back the changes so it's supported officially.

JerLarge, 2020-04-19:
looks like there are a few different variations of the zigbee model for the same thing.   just got around to adding mine to this, and see its unsupported.

{"id":18,"type":"Router","ieeeAddr":"0x841826000000d3bb","nwkAddr":40958,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Gardenspot RGB","epList":[3],"endpoints":{"3":{"profId":260,"epId":3,"devId":258,"inClusterList":[0,3,4,5,6,8,768,2820,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Gardenspot RGB","manufacturerName":"OSRAM","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20140331CNEF****"}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20140331CNEF****","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1587236756628}