Koenkk/zigbee2mqtt #3207: 'zcl' data is from unknown device with address '28430', skipping...

errolsancaktar, 2020-03-27:
# Bug Report
## What happened
no Zigbee2mqtt devices are reporting any info all of them show similar log error:

```
zigbee2mqtt    | 2020-03-27T03:44:41.931Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":47,"manufacturerCode":4660,"commandIdentifier":10},"Payload":[{"attrId":65282,"dataType":76,"attrData":[{"elmType":16,"elmVal":0},{"elmType":33,"elmVal":2985},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":163},{"elmType":32,"elmVal":97}],"structElms":[{"elmType":16,"elmVal":0},{"elmType":33,"elmVal":2985},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":163},{"elmType":32,"elmVal":97}],"numElms":6}]},"address":55335,"endpoint":1,"linkquality":86,"groupID":0}'
zigbee2mqtt    | 2020-03-27T03:44:41.931Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '55335', skipping...
```

## What did you expect to happen

was working before

## How to reproduce it (minimal and precise)
always happening
## Debug Info
zigbee2mqtt version:1.12.0-dev
CC253X firmware version:1.12.0-dev

Logs:

https://paste.sancaktar.net/?8e132fd86e5e6727#AHdXAzxiCLPyspdpSv7M5JXLvF97BzWERFJudhYwBRjH

Koenkk, 2020-03-27:
Your link is dead, is your `database.db` empty?

errolsancaktar, 2020-03-27:
sorry something up w/ my paster:
https://pastebin.com/9Wga5Cud

there is nothing in database.db. how would that have been emptied?  any way to resolve without rejoining every device?

Koenkk, 2020-03-27:
Did you change the channel, pan_id or network_key in your configuration.yaml?

errolsancaktar, 2020-03-27:
nope, no changes just pulled 1.12-dev

errolsancaktar, 2020-03-27:
attempted a repair of one device:  got some errors:
```
zigbee2mqtt    | zigbee2mqtt:debug 2020-03-27 14:07:15: Received Zigbee message from 'Basement_Motion', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | zigbee2mqtt:error 2020-03-27 14:07:15: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
zigbee2mqtt    |     at /app/lib/extension/deviceReceive.js:106:42
zigbee2mqtt    |     at Array.filter (<anonymous>)
zigbee2mqtt    |     at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:105:52)
zigbee2mqtt    |     at Controller.callExtensionMethod (/app/lib/controller.js:346:44)
zigbee2mqtt    |     at Controller.onZigbeeEvent (/app/lib/controller.js:231:14)
zigbee2mqtt    |     at Zigbee.emit (events.js:311:20)
zigbee2mqtt    |     at Controller.<anonymous> (/app/lib/zigbee.js:58:52)
zigbee2mqtt    |     at Controller.emit (events.js:311:20)
zigbee2mqtt    |     at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:463:22)
zigbee2mqtt    |     at Generator.next (<anonymous>))
zigbee2mqtt    | 2020-03-27T20:07:15.553Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

After that the logs show receiving data but mqtt not showing a new topic

Koenkk, 2020-03-27:
Did you ever run with a modified herdsman converters? or touched `devices.js` to support new devices?

errolsancaktar, 2020-03-28:
i did a long time ago.  seems as though readding devices is working though some are not getting recognized.  going to try to add them right next to the controller

closed 2020-06-06 by stale[bot]