Koenkk/zigbee2mqtt #3208: Problem with HEIMAN HS1CA-M - sais it is not supported

tonipetrovic, 2020-03-27:
I just recieved HEIMAN HS1CA-M CO detector. I wnted to pair it with my network and it sais it is not supported, bu it is in the list of supported devices (https://www.zigbee2mqtt.io/devices/HS1CA-M.html).

I am using Texas Instruments CC1352P-2 as coordinator and setting for joining new devices in configuration file is set to false (permit_join: false). The coordinator is attached to raspberry pi 4 running zigbee2mqtt version 1.12.0 (commit #840b9d9).

Output:

Mar 27 09:42:24 marspi npm[25254]: zigbee2mqtt:info  2020-03-27 09:42:24: Device '0x005043c953292988' joined
Mar 27 09:42:24 marspi npm[25254]: zigbee2mqtt:info  2020-03-27 09:42:24: MQTT publish: topic 'aihome/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x005043c953292988"}}'
Mar 27 09:42:24 marspi npm[25254]: zigbee2mqtt:info  2020-03-27 09:42:24: Starting interview of '0x005043c953292988'
Mar 27 09:42:24 marspi npm[25254]: zigbee2mqtt:info  2020-03-27 09:42:24: MQTT publish: topic 'aihome/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x005043c953292988"}}'
Mar 27 09:42:24 marspi npm[25254]: zigbee2mqtt:info  2020-03-27 09:42:24: MQTT publish: topic 'aihome/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x005043c953292988"}}'
Mar 27 09:42:31 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:31: Received message from unsupported device with Zigbee model 'CO_YDLV10'
Mar 27 09:42:31 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:31: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Mar 27 09:42:32 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:32: Received message from unsupported device with Zigbee model 'CO_YDLV10'
Mar 27 09:42:32 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Mar 27 09:42:33 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:33: Received message from unsupported device with Zigbee model 'CO_YDLV10'
Mar 27 09:42:33 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:33: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Mar 27 09:42:34 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:34: Received message from unsupported device with Zigbee model 'CO_YDLV10'
Mar 27 09:42:34 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:34: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Mar 27 09:42:36 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:36: Received message from unsupported device with Zigbee model 'CO_YDLV10'
Mar 27 09:42:36 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:36: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Mar 27 09:42:37 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:37: Received message from unsupported device with Zigbee model 'CO_YDLV10'
Mar 27 09:42:37 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:37: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Mar 27 09:42:40 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:40: Received message from unsupported device with Zigbee model 'CO_YDLV10'
Mar 27 09:42:40 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:40: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Mar 27 09:42:40 marspi npm[25254]: zigbee2mqtt:info  2020-03-27 09:42:40: Successfully interviewed '0x005043c953292988', device has successfully been paired
Mar 27 09:42:40 marspi npm[25254]: zigbee2mqtt:warn  2020-03-27 09:42:40: Device '0x005043c953292988' with Zigbee model 'CO_YDLV10' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devic          es.html
Mar 27 09:42:40 marspi npm[25254]: zigbee2mqtt:info  2020-03-27 09:42:40: MQTT publish: topic 'aihome/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x005043c953292988","su          pported":false}}'

I am new to this so if you need any additional info, please point me how to get it :).

Thank you and regards,
Toni

Kryzek, 2020-03-27:
Probably your device is different version of HS1CA-M since old entry is for model 'CO_V15' and your device identifies itself as 'CO_YDLV10'.

I would try adding model to existing device entry.
If you're using bare metal installation (not docker), edit node_modules/zigbee-herdsman-converters/devices.js and find existing entry and replace `zigbeeModel: ['CO_V15'], ` to `zigbeeModel: ['CO_V15', 'CO_YDLV10'],` - resulting:
```
    {
        zigbeeModel: ['CO_V15', 'CO_YDLV10'],
        model: 'HS1CA-M',
        description: 'Smart carbon monoxide sensor',
        supports: 'carbon monoxide',
        vendor: 'HEIMAN',
        fromZigbee: [fz.heiman_carbon_monoxide, fz.battery],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
            await configureReporting.batteryAlarmState(endpoint);
        },
    },
```

tonipetrovic, 2020-03-27:
Dear @Kryzek ,

thank you for this. Now it identified and paired it successfully, but I don't get any messages from it to zgbe2mqtt, also when I restart zgbe2mqtt (I get messages from all other sensors like Xiaomi Aqara temperature, humidity and pressure sensor, body sensor, etc.). Is there any way to check if the sensor is publishing something to the network?

Regards,
Toni

tonipetrovic, 2020-03-27:
Dear @Kryzek ,

thank you for this. Now it identified and paired it successfully. Will you add this model also to the code on git?

Thank you and kind regards,
Toni

closed 2020-03-27 by tonipetrovic

Kryzek, 2020-03-27:
@tonipetrovic - I made a pull request to add new device model for this device, so it should be supported out of the box in the future.