Koenkk/zigbee2mqtt #3209: Change the CC2531 coordinator to CC2530 with usb adapter. In synology nas docker the usb ttyACM0 disappered.

spider7611, 2020-03-27:
#3172  Debug Info
zigbee2mqtt version: latest

If I push in the new CC2530 coordinator the synology nas not show up this: ttyACM0. 
/dev ls not see
After the "lsusb" I see the new CC2530 coordinator. "12MBit/s 100mA 1IF  (Silicon Labs CP2102 USB to UART Bridge Controller 0001)" How can I solve this problem?

If I put back the old CC2531 coordinator the ttyACM0 come back and working good.

spider7611, 2020-03-28:
On my synology install USB drivers from here: http://www.jadahl.com/drivers_6.2/?arch=apollolake# and now see this: ttyUSB0
Setup everything and start the zigbee2mqtt and the log show this:

> info  2020-03-28 16:32:55: Starting zigbee2mqtt version 1.12.0-dev (commit #ac7ac1a)
> info  2020-03-28 16:32:55: Starting zigbee-herdsman...
> error 2020-03-28 16:32:58: Failed to call 'DeviceEvent' 'stop' (TypeError: Cannot read property 'getEntries' of null
>     at Function.loadFromDatabaseIfNecessary (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:200:55)
>     at Function.all (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:220:16)
>     at Controller.getDevices (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:215:31)
>     at Zigbee.getClients (/app/lib/zigbee.js:116:30)
>     at DeviceEvent.stop (/app/lib/extension/deviceEvent.js:18:42)
>     at Controller.callExtensionMethod (/app/lib/controller.js:346:44)
>     at Controller.stop (/app/lib/controller.js:152:20)
>     at process.handleQuit (/app/index.js:21:20)
>     at process.emit (events.js:311:20))
> error 2020-03-28 16:32:58: Not connected to MQTT server!
> error 2020-03-28 16:32:58: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
> info  2020-03-28 16:32:58: Disconnecting from MQTT server
> error 2020-03-28 16:33:02: Error while starting zigbee-herdsman

What is the problem?

Koenkk, 2020-03-28:
Can you provide the herdsman debug logging when starting?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

spider7611, 2020-03-28:
This is what you need?

> 19:01:58
>   zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +0ms
> stdout
> 19:01:58
>   zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
> stdout
> 19:01:58
>   serialport/stream _write 1 bytes of data +4ms
> stdout
> 19:01:58
>   serialport/binding-abstract write 1 bytes +3ms
> stdout
> 19:01:58
>   serialport/bindings/unixRead Starting read +0ms
> stdout
> 19:01:58
>   serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
> stdout
> 19:01:58
>   serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
> stdout
> 19:01:58
>   errno: -11,
> stdout
> 19:01:58
>   code: 'EAGAIN',
> stdout
> 19:01:58
>   syscall: 'read'
> stdout
> 19:01:58
> } +2ms
> stdout
> 19:01:58
>   serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
> stdout
> 19:01:58
>   serialport/bindings/poller Polling for "readable" +11ms
> stdout
> 19:01:58
>   serialport/bindings/unixWrite write returned: wrote 1 bytes +4ms
> stdout
> 19:01:58
>   serialport/bindings/unixWrite Finished writing 1 bytes +0ms
> stdout
> 19:01:58
>   serialport/stream binding.write write finished +7ms
> stdout
> 19:01:59
>   zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
> stdout
> 19:01:59
>   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
> stdout
> 19:01:59
>   serialport/stream _write 5 bytes of data +1s
> stdout
> 19:01:59
>   serialport/binding-abstract write 5 bytes +1s
> stdout
> 19:01:59
>   serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +1s
> stdout
> 19:01:59
>   serialport/bindings/unixWrite write returned: wrote 5 bytes +1ms
> stdout
> 19:01:59
>   serialport/bindings/unixWrite Finished writing 5 bytes +0ms
> stdout
> 19:01:59
>   serialport/stream binding.write write finished +2ms
> stdout
> 19:02:05
> zigbee2mqtt:error 2020-03-28 20:02:05: Error while starting zigbee-herdsman
> stdout
> 19:02:05
> winston:file written true false
> stdout
> 19:02:05
> zigbee2mqtt:error 2020-03-28 20:02:05: Failed to start zigbee
> stdout
> 19:02:05
> winston:file written true false
> stdout
> 19:02:05
> zigbee2mqtt:error 2020-03-28 20:02:05: Exiting...
> stdout
> 19:02:05
> winston:file written true false
> stdout
> 19:02:05
> zigbee2mqtt:error 2020-03-28 20:02:05: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
> stdout
> 19:02:05
>     at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
> stdout
> 19:02:05
>     at Generator.throw (<anonymous>)
> stdout
> 19:02:05
>     at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
> stdout
> 19:02:05
> winston:file written true false
> stdout
> 19:02:05
> npm ERR! code ELIFECYCLE
> stdout
> 19:02:05
> npm ERR! errno 1
> stdout
> 19:02:05
> npm ERR! zigbee2mqtt@1.12.0-dev start: `node index.js`
> stdout
> 19:02:05
> npm ERR! Exit status 1
> stdout
> 19:02:05
> npm ERR! 
> stdout
> 19:02:05
> npm ERR! Failed at the zigbee2mqtt@1.12.0-dev start script.
> stdout
> 19:02:05
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> stdout
> 19:02:05
> stdout
> 19:02:05
> npm ERR! A complete log of this run can be found in:
> stdout
> 19:02:05
> npm ERR!     /root/.npm/_logs/2020-03-28T19_02_05_573Z-debug.log

spider7611, 2020-03-28:
And this is the log.txt

> info  2020-03-28 20:01:57: Logging to console and directory: '/app/data/log/2020-03-28.20-01-57' filename: log.txt
> debug 2020-03-28 20:01:57: Removing old log directory '/app/data/log/2020-03-28.19-57-56'
> debug 2020-03-28 20:01:58: Loaded state from file /app/data/state.json
> info  2020-03-28 20:01:58: Starting zigbee2mqtt version 1.12.0-dev (commit #fb4d86c)
> info  2020-03-28 20:01:58: Starting zigbee-herdsman...
> debug 2020-03-28 20:01:58: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyUSB0"}}'
> error 2020-03-28 20:02:05: Error while starting zigbee-herdsman

spider7611, 2020-03-28:
Sorry my fault. Connect wrong pin from the usb serial to cc2530. Closed!

closed 2020-03-28 by spider7611