Koenkk/zigbee2mqtt #3212: Seemingly random hangs of coordinator \ mesh ?

atxbyea, 2020-03-27:
# Bug Report
## What happened
mesh appears to hang, have to restart \ replug coordinator to solve it
## What did you expect to happen
mesh to heal itself, not hang, not sure if issue is in coordinator firmware or in the mesh itself, if I have to replug the adapter it says  Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms) in the logs
## How to reproduce it (minimal and precise)
patience... 2-6 days between each hang
## Debug Info
zigbee2mqtt version: version 1.12.0-dev (commit #5cd1af5)
CC1352-P2 firmware version: 20191106

Used to run on a docker host with 12GB RAM \ SSD \ i5 CPU, now running on a RPI 3b+ with the same frequency of issue.

Logs attached here (at info level), will update issue with debug cases on next occurance :

https://filebin.net/5k21ho2x57h7xa62

Koenkk, 2020-03-27:
I indeed need the log of the moment that it happens.

atxbyea, 2020-04-01:
Luckily the error happened quickly, see logs here https://filebin.net/ok4t4wqxd50hbreq

Koenkk, 2020-04-01:
It seems to happen when controlling many devices at once. 

Can you update to the latest zigbee2mqtt dev branch and add the following to your `configuration.yaml`:

```yaml
advanced:
  adapter_concurrent: 2
```

atxbyea, 2020-04-04:
Sorry for the late reply, updated and added the variable now.

I do indeed have a few devices, slowly closing in on 100 with currently 84 total devices in zigbee2mqtt and still some not moved yet from the hue hub.

atxbyea, 2020-04-09:
No crashes yet @Koenkk so looking good, is there a limit for how high I can set it on the 1352 ? Docs only mention 2531 and 26x2, and would it be even more beneficial to have it further increased considering the size of my network?

Koenkk, 2020-04-10:
We initially thought the limit would be 16, but this doesn't seem to be true. I would start increasing this limit now and see if it keeps stable. As a next step try `6`.

atxbyea, 2020-04-18:
Seems very much more stable so far, been running with concurrent 6 for 5 days without a single hickup so far @Koenkk

atxbyea, 2020-04-27:
Appears to not have been fully resolved.....
Woke up this morning no communication with the mesh, so I restarted the pi, however, the issue returned again this afternoon, so two hangs within 12 hours.

Both logs attached @Koenkk 
[logs.tgz.zip](https://github.com/Koenkk/zigbee2mqtt/files/4541389/logs.tgz.zip)

Koenkk, 2020-04-28:
It seems that the coordinator still crashes. A few things can be done:
- Update to the latest dev firmware (https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin) (would try this for sure as it contains a new SDK version form Texas Instruments)
- Disable device availibilty
- Don't execute network scans anymore
- Set `adapter_concurrent: 4`

atxbyea, 2020-04-29:
Dev firmware updated
Device avaliablility retained because I need it for an automation currently
Disabled network scans
Changed concurrent to 4.

Monitoring.

atxbyea, 2020-05-02:
Another hang after 3 days, replugging USB, restarting docker with avalibility disabled now
[logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/4568403/logs.zip)
[config.log](https://github.com/Koenkk/zigbee2mqtt/files/4568418/config.log)

Current config and last logs attached.

atxbyea, 2020-05-03:
Wondering if my gledopto devices might be involved? I see in Zigbee2mqttassistant that they are "unresponsive" for multiple weeks

Koenkk, 2020-05-04:
I still see `debug 2020-05-02 17:02:36: Successfully pinged 'Kjokken Ventilator'` which is only logged by the availability feature, but you config shows it's disabled. Any idea what happened there?

atxbyea, 2020-05-24:
Still looking good after 2 weeks, I also did migrate to a new mqtt server, and added a cc2531 repeater .

Will try readding avaliability timeout and see if that makes the issue reappear

closed 2020-07-26 by stale[bot]