Koenkk/zigbee2mqtt #3213: Adding a new device based on TuYa

way2many, 2020-03-27:
Hi guys,

I'm trying to add following device I bought on [aliexpress](https://www.aliexpress.com/item/4000401001285.html?spm=a2g0s.9042311.0.0.75434c4d5mBSuS)

It should be Tuya based. It joined the network but was not supported. I modified therefore the device.js file in order to add support for it. I copied the data from "TS0002" (Tuya based 2 gang switch) and modified the zigbeemodel, model, vendor (info based on database.db) and removed the whitelabel part. I did a restart of zigbee2mqtt and removed the device by sending the mqtt command, it removed the device and rejoined almost instantly but this time with "supported". Unfortunately that is how much I got from it. It doesn't respond to any command so I tried to dig down in the debug log but I don't understand half of it. Not sure what I should do now.

Here is the debug log:

```
info  2020-03-27 17:22:34: Device '0xec1bbdfffeddb36f' joined
info  2020-03-27 17:22:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xec1bbdfffeddb36f"}}'
info  2020-03-27 17:22:34: Starting interview of '0xec1bbdfffeddb36f'
info  2020-03-27 17:22:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xec1bbdfffeddb36f"}}'
debug 2020-03-27 17:22:34: Device '0xec1bbdfffeddb36f' announced itself
info  2020-03-27 17:22:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xec1bbdfffeddb36f"}}'
debug 2020-03-27 17:22:35: Received Zigbee message from '0xec1bbdfffeddb36f', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0601","manufacturerName":"_TZE200_nkjintbl"}' from endpoint 1 with groupID 0
debug 2020-03-27 17:22:35: No converter available for 'TS0601' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TS0601","manufacturerName":"_TZE200_nkjintbl"}'
debug 2020-03-27 17:22:35: Received Zigbee message from '0xec1bbdfffeddb36f', type 'readResponse', cluster 'genBasic', data '{"powerSource":1,"zclVersion":3}' from endpoint 1 with groupID 0
debug 2020-03-27 17:22:35: No converter available for 'TS0601' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1,"zclVersion":3}'
debug 2020-03-27 17:22:36: Received Zigbee message from '0xec1bbdfffeddb36f', type 'readResponse', cluster 'genBasic', data '{"appVersion":82,"stackVersion":0}' from endpoint 1 with groupID 0
debug 2020-03-27 17:22:36: No converter available for 'TS0601' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":82,"stackVersion":0}'
debug 2020-03-27 17:22:36: Received Zigbee message from '0xec1bbdfffeddb36f', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":""}' from endpoint 1 with groupID 0
debug 2020-03-27 17:22:36: No converter available for 'TS0601' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":""}'
debug 2020-03-27 17:22:36: Received Zigbee message from '0xec1bbdfffeddb36f', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2020-03-27 17:22:36: No converter available for 'TS0601' with cluster 'genBasic' and type 'readResponse' and data '{}'
info  2020-03-27 17:22:36: Successfully interviewed '0xec1bbdfffeddb36f', device has successfully been paired
info  2020-03-27 17:22:36: Device '0xec1bbdfffeddb36f' is supported, identified as: _TZE200_nkjintbl 2 gang switch (TS0601)
info  2020-03-27 17:22:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffeddb36f","model":"TS0601","vendor":"_TZE200_nkjintbl","description":"2 gang switch","supported":true}}'
debug 2020-03-27 17:22:41: No converter available for 'TS0601' with cluster 'genTime' and type 'read' and data '["localTime"]'
debug 2020-03-27 17:22:47: Received Zigbee message from '0xec1bbdfffeddb36f', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
debug 2020-03-27 17:22:47: No converter available for 'TS0601' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":21,"appVersion":82}'
debug 2020-03-27 17:22:47: Received Zigbee message from '0xec1bbdfffeddb36f', type 'attributeReport', cluster 'genBasic', data '{"65503":"��\u0010&\u0012��\u0010&\u0013"}' from endpoint 1 with groupID 0
debug 2020-03-27 17:22:47: No converter available for 'TS0601' with cluster 'genBasic' and type 'attributeReport' and data '{"65503":"��\u0010&\u0012��\u0010&\u0013"}'
debug 2020-03-27 17:22:50: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,10,2,0,9,2,1,0,1,0]}' from endpoint 1 with groupID 0
debug 2020-03-27 17:22:50: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,10,2,0,9,2,1,0,1,0]}'
```

What I think I understand is that it is missing the "translation" part so that zigbee2mqtt can understand the data that is being submitted from the device. I had a look at fromZigbee file and tried to understand but this is way over my head.

Any help is greatly appreciated.

Thank you.

Koenkk, 2020-03-27:
With the following definition:

```js
    {
        zigbeeModel: ['TS0601'],
        model: 'TS0601',
        vendor: 'TuYa',
        description: '2 gang switch',
        supports: 'on/off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {configureKey: 1, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint) => {
            await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        },
    },
```

What do you see in the log/happens to the switch:
- Publish to `zigbee2mqtt/0xec1bbdfffeddb36f/l1/set` payload `{"state": "ON"}` or (`OFF`)
- Publish to `zigbee2mqtt/0xec1bbdfffeddb36f/l2/set` payload `{"state": "ON"}` or (`OFF`)
- Press the left button on the switch
- Press the right button on the switch

way2many, 2020-03-29:
> * Publish to `zigbee2mqtt/0xec1bbdfffeddb36f/l1/set` payload `{"state": "ON"}` or (`OFF`)

```
debug 2020-03-29 19:40:39: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffeddb36f/l1/set' with data '{"state": "ON"}'
debug 2020-03-29 19:40:39: Publishing 'set' 'state' to '0xec1bbdfffeddb36f'
info  2020-03-29 19:40:39: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffeddb36f', payload '{"state_l1":"ON"}'
```

> * Publish to `zigbee2mqtt/0xec1bbdfffeddb36f/l2/set` payload `{"state": "ON"}` or (`OFF`)

```
debug 2020-03-29 19:40:55: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffeddb36f/l2/set' with data '{"state": "ON"}'
error 2020-03-29 19:40:55: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'ID' of undefined
    at /app/lib/zigbee.js:192:106
    at Array.find (<anonymous>)
    at Zigbee.resolveEntity (/app/lib/zigbee.js:192:76)
    at EntityPublish.onMQTTMessage (/app/lib/extension/entityPublish.js:59:36)
    at Controller.callExtensionMethod (/app/lib/controller.js:346:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:242:14)
    at MQTT.emit (events.js:311:20)
    at MQTT.onMessage (/app/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:311:20)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1162:12))
```

> * Press the left button on the switch

```
Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:06: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}'
debug 2020-03-29 19:41:06: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:06: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}'
debug 2020-03-29 19:41:06: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:06: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}'
debug 2020-03-29 19:41:06: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:06: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}'
debug 2020-03-29 19:41:06: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:06: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,14,2,0,9,1,1,0,1,1]}'
```

> * ress the right button on the switch

```
debug 2020-03-29 19:41:19: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:19: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}'
debug 2020-03-29 19:41:19: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:19: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}'
debug 2020-03-29 19:41:19: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:19: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}'
debug 2020-03-29 19:41:19: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:19: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}'
debug 2020-03-29 19:41:20: Received Zigbee message from '0xec1bbdfffeddb36f', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}' from endpoint 1 with groupID 0
debug 2020-03-29 19:41:20: No converter available for 'TS0601' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[9,15,2,0,10,2,1,0,1,1]}'
```

Some extra log, this is right after removing electricity and switch it back on:

```
[... cut ...]
    at async DeviceConfigure.configure (/app/lib/extension/deviceConfigure.js:95:13))
error 2020-03-29 19:29:08: Failed to configure '0xec1bbdfffeddb36f', attempt 3 (TypeError: Cannot read property 'bind' of undefined
    at bind (/app/node_modules/zigbee-herdsman-converters/devices.js:39:24)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices.js:1074:19)
    at async DeviceConfigure.configure (/app/lib/extension/deviceConfigure.js:95:13))
```

Thanks for having a look at it :)

Koenkk, 2020-03-30:
Can you share your `database.db`?

way2many, 2020-03-31:
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b000be8a770","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1585337401847}
{"id":2,"type":"Group","groupID":10,"members":[{"deviceIeeeAddr":"0x90fd9ffffe2a208d","endpointID":1},{"deviceIeeeAddr":"0x90fd9ffffe2a3f57","endpointID":1}],"meta":{}}
{"id":3,"type":"Group","groupID":20,"members":[{"deviceIeeeAddr":"0xd0cf5efffeda28a4","endpointID":1},{"deviceIeeeAddr":"0xd0cf5efffe226944","endpointID":1}],"meta":{}}
{"id":4,"type":"Group","groupID":21,"members":[{"deviceIeeeAddr":"0x000d6ffffe054c2c","endpointID":1},{"deviceIeeeAddr":"0x14b457fffe48254d","endpointID":1},{"deviceIeeeAddr":"0x14b457fffe744bae","endpointID":1},{"deviceIeeeAddr":"0x14b457fffe3f5dd7","endpointID":1}],"meta":{}}
{"id":5,"type":"Group","groupID":30,"members":[{"deviceIeeeAddr":"0x90fd9ffffe2a601f","endpointID":1},{"deviceIeeeAddr":"0x000d6ffffe5b1f7b","endpointID":1}],"meta":{}}
{"id":6,"type":"Router","ieeeAddr":"0xd0cf5efffe226944","nwkAddr":43476,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 CWS opal 600lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":512,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 CWS opal 600lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"lightingColorCtrl":{"attributes":{"currentX":32178,"currentY":25559}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b000be8a770","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b000be8a770","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b000be8a770","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1585583483191}
{"id":7,"type":"Router","ieeeAddr":"0x0017880104b8d410","nwkAddr":60711,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LWB010","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":256,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWB010","manufacturerName":"Philips","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170908","swBuildId":"1.29.0_r21169"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b000be8a770","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b000be8a770","endpointID":1}]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170908","swBuildId":"1.29.0_r21169","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1585543663708}
{"id":8,"type":"Router","ieeeAddr":"0x90fd9ffffe2a601f","nwkAddr":48505,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 W 400lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb GU10 W 400lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b000be8a770","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b000be8a770","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1585574070359}
[... cut ...] {"attributes":{"modelId":"GL-MC-001","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.1"}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.1","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1585478693062}
```

It is ID 40.

Koenkk, 2020-03-31:
You mention that the device has 2 gangs, but I see that the device has only 1 endpoint. Endpoint 1 currently control gang 1, so I don't know how they control the other gang. Is this indeed a 2 gang switch?

way2many, 2020-03-31:
This is indeed a 2 gang switch. There are two physical buttons.

Koenkk, 2020-03-31:
Very strange, normally these devices also have 2 endpoints in this case. I have no clue now how to control the other gang. Unfortunately I think we need to sniff the traffic with the original gateway on how to do this.

way2many, 2020-03-31:
Ah damn, I was afraid of this. I will program one of the stick with the sniffing firmware and post back the results.

As always, thank you for your time.

dzegarra, 2020-05-09:
I recently wrote an article explaining how to decode the payload of tuya-base devices. I may help you figure out what these numbers mean. 

[Zigbee2mqtt: How to add support for a new Tuya-based device (part 1)](https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-1-b20227251d46)

tabletka-ao, 2020-06-09:
Hi @way2many I have the same problem. I bought wall switches X711A (1 gang) , X712A (2 gang).
Did you manage to deal with him?

tabletka-ao, 2020-06-17:
linked #2778

closed 2020-07-26 by stale[bot]