Koenkk/zigbee2mqtt #3216: Zemismart ZM-CSW032-D Curtain/roller blind switch set position

slashbv, 2020-03-28:
I have installed this switch for a test and it works very well and planning to buy more. Checking logs of zigbee2mqtt I have seen that it is also sending the position: 
MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":50,"linkquality":73}'
However this is not available in Home Assistant so an intermediate position cannot be set. Is it possible to have this working in Home Assistant ?

Koenkk, 2020-03-28:
Should be fixed in the latest dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

slashbv, 2020-03-28:
great thanks. I will check it. when it will get to stable release?

Koenkk, 2020-03-28:
on next zigbee2qmtt release (1.5 weeks from now).

slashbv, 2020-03-29:
not sure if it's related but I realised that when using the buttons on device the status is not updated in HA. Now the shutter is open but in HA it shows as closed.

Koenkk, 2020-03-29:
Could you provide the debug logging when controlling it manually?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

slashbv, 2020-03-30:
see below all messages related to this switch:
error 2020-03-30 08:06:03: Failed to setup reporting for '0xec1bbdfffe946658' - Error: ConfigureReporting 0xec1bbdfffe946658/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":3,"maximumReportInterval":300,"reportableChange":1},{"attribute":"currentPositionTiltPercentage","minimumReportInterval":3,"maximumReportInterval":300,"reportableChange":1}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 19527 - 1 - 2 - 258 - 7 after 6000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:316:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
debug 2020-03-30 08:06:15: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,11,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-03-30 08:06:15: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,11,10,1,0,32,66,226,255,32,19]}'
debug 2020-03-30 08:07:01: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-03-30 08:07:01: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":73}'
debug 2020-03-30 08:08:02: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
info  2020-03-30 08:08:02: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":50,"linkquality":70}'
debug 2020-03-30 08:08:26: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100}' from endpoint 1 with groupID 0
info  2020-03-30 08:08:26: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":0,"linkquality":70}'
debug 2020-03-30 08:08:38: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,15,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-03-30 08:08:38: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,15,10,1,0,32,66,226,255,32,19]}'
debug 2020-03-30 08:08:50: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
info  2020-03-30 08:08:50: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":50,"linkquality":70}'

slashbv, 2020-03-30:
status is unknown in HA

Koenkk, 2020-03-30:
I see, zigbee2mqtt seems to publish the correct position, but are you on the latest dev branch otherwise it won't work

slashbv, 2020-03-30:
I tried to load dev but I get this error:
ERROR: pull access denied for koenkk/zigbee2mqtt-dev, repository does not exist or may require 'docker login': denied: requested access to the resource is denied

Koenkk, 2020-03-30:
It should be `koenkk/zigbee2mqtt:dev`

slashbv, 2020-03-31:
thanks, seems to be working I can see the status and position now. The problem is the position doesn't correspond with reality, it is opened but it shows 50%. Is there a way to fix it ?
info  2020-03-31 08:08:38: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":50,"linkquality":70}'

Koenkk, 2020-03-31:
So it seems that it reports the wrong position? What does it report when setting it to e.g. 50%?

slashbv, 2020-03-31:
I have done more tests, it seems it is always reporting 50% no matter of position. If for example I press the close or open buttons it reports 100 and 0 and then 50. 
info  2020-03-31 21:31:05: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":0,"linkquality":70}'
info  2020-03-31 21:31:08: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":50,"linkquality":70}'
info  2020-03-31 21:31:13: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":70}'
info  2020-03-31 21:32:13: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":50,"linkquality":70}'

Even if I closed it from the switch it was still showing opened in HA and the slide of course at 50. When trying to control it from the slide I have the following errors in log:

error 2020-03-31 21:30:45: No converter available for 'position' (56)
error 2020-03-31 21:30:45: No converter available for 'position' (54)
error 2020-03-31 21:30:46: No converter available for 'position' (80)
error 2020-03-31 21:30:46: No converter available for 'position' (80)
error 2020-03-31 21:30:46: No converter available for 'position' (91)
error 2020-03-31 21:30:47: No converter available for 'position' (90)
error 2020-03-31 21:30:47: No converter available for 'position' (67)
error 2020-03-31 21:30:47: No converter available for 'position' (100)

Koenkk, 2020-04-01:
- `error 2020-03-31 21:30:45: No converter available for 'position' (54)`
    - Should be fixed in the latest dev branch (please check in 4 hours from now)
- Regarding the wrong position, please provide the **debug** logs when controlling it the device with the buttons. Clearly indicate in the logs what hapend after which button.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

slashbv, 2020-04-01:
Here is the log:

--This error appears every time I restart zigbee2mqtt

error 2020-04-01 20:24:55: Failed to setup reporting for '0xec1bbdfffe946658' - Error: ConfigureReporting 0xec1bbdfffe946658/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":3,"maximumReportInterval":300,"reportableChange":1},{"attribute":"currentPositionTiltPercentage","minimumReportInterval":3,"maximumReportInterval":300,"reportableChange":1}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 19527 - 1 - 2 - 258 - 7 after 6000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
debug 2020-04-01 20:25:20: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,22,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-01 20:25:20: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,22,10,1,0,32,66,226,255,32,19]}'
debug 2020-04-01 20:27:43: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,23,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-01 20:27:43: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,23,10,1,0,32,66,226,255,32,19]}'

-- The shutters were opened, here I pressed the close button

debug 2020-04-01 20:28:43: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-04-01 20:28:43: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":73}'
debug 2020-04-01 20:29:44: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
info  2020-04-01 20:29:44: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":50,"linkquality":73}'

--Here I pressed the open button 

debug 2020-04-01 20:30:05: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100}' from endpoint 1 with groupID 0
info  2020-04-01 20:30:05: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":0,"linkquality":73}'
debug 2020-04-01 20:30:06: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'genBasic', data '{"65506":19,"appVersion":66}' from endpoint 1 with groupID 0
debug 2020-04-01 20:31:06: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
info  2020-04-01 20:31:06: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":50,"linkquality":73}'

--Here I pressed the close button again

debug 2020-04-01 20:31:20: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-04-01 20:31:20: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":73}'
debug 2020-04-01 20:32:21: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
info  2020-04-01 20:32:21: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":50,"linkquality":73}'
debug 2020-04-01 20:32:29: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,31,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-01 20:32:29: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,31,10,1,0,32,66,226,255,32,19]}'

Koenkk, 2020-04-01:
Thanks, can you check again with the latest dev branch?

slashbv, 2020-04-02:
Morning,
 I have checked and the behaviour seems similar like with stable version just that when it's opened it shows closed and when it's closed it shows open. There is no intermediate position anymore, not even 50%. The slide is HA does nothing. One you pres a button it will set that position (closed or open) no matter if you stop it in the middle or not, there is no 25, 50 or 75 %.
Here is the log:

-- this error is still there

error 2020-04-02 08:39:35: Failed to setup reporting for '0xec1bbdfffe946658' - Error: ConfigureReporting 0xec1bbdfffe946658/1                   closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":3,"maximumReportInterval":300,"r                  eportableChange":1},{"attribute":"currentPositionTiltPercentage","minimumReportInterval":3,"maximumReportInterval":300,"report                  ableChange":1}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 19527 - 1 -                   2 - 258 - 7 after 6000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)

--close button

debug 2020-04-02 08:40:44: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-04-02 08:40:44: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":73}'
debug 2020-04-02 08:40:45: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-04-02 08:40:45: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":70}'
debug 2020-04-02 08:40:45: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-04-02 08:40:45: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":73}'
debug 2020-04-02 08:40:50: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
debug 2020-04-02 08:40:50: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
debug 2020-04-02 08:40:50: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
debug 2020-04-02 08:41:34: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,97,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-02 08:41:34: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,97,10,1,0,32,66,226,255,32,19]}'

--stop in the middle
--close button 

debug 2020-04-02 08:43:57: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,98,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-02 08:43:57: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,98,10,1,0,32,66,226,255,32,19]}'
debug 2020-04-02 08:44:46: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-04-02 08:44:46: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":70}'
debug 2020-04-02 08:44:46: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-04-02 08:44:46: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":70}'
debug 2020-04-02 08:45:47: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
debug 2020-04-02 08:46:20: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,101,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-02 08:46:20: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,101,10,1,0,32,66,226,255,32,19]}'
debug 2020-04-02 08:48:43: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,102,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-02 08:48:43: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,102,10,1,0,32,66,226,255,32,19]}'

--open button

[... cut ...]
debug 2020-04-02 08:52:48: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
debug 2020-04-02 08:53:29: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,108,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-02 08:53:29: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,108,10,1,0,32,66,226,255,32,19]}'

Koenkk, 2020-04-02:
- Error should be fixed in latest dev
- Inverted position should be fixed in latest dev
- Regarding intermediate position, it seems that the device doesn't send this position. I've also checked the `raw` `genBasic` but it also does not seem to contain the position. So unfortunately this cannot be added.

slashbv, 2020-04-02:
thank you. 
I see in the log there is some intermediate position sent: 

debug 2020-04-02 08:52:48: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0

Is this something else ?

Koenkk, 2020-04-02:
That position is always reported after it posted 0 or 100 so this value is ignored.

slashbv, 2020-04-03:
inverted position and error are fixed but now it cannot be controlled from HA anymore when I press buttons I have these errors:
error 2020-04-03 13:44:03: No converter available for 'state' (CLOSE)
error 2020-04-03 13:44:04: No converter available for 'state' (STOP)

When I use the slide nothing happens either and nothing is shown in the log.  I used to have the errors below when using the slide but not there is nothing. Maybe you can make it from the slide to go to position. I have contacted the seller and he says in smartthings it should show intermediate position. I don't have smartthings to test it.

error 2020-03-31 21:30:45: No converter available for 'position' (56)
error 2020-03-31 21:30:45: No converter available for 'position' (54)
error 2020-03-31 21:30:46: No converter available for 'position' (80)
error 2020-03-31 21:30:46: No converter available for 'position' (80)
error 2020-03-31 21:30:46: No converter available for 'position' (91)
error 2020-03-31 21:30:47: No converter available for 'position' (90)
error 2020-03-31 21:30:47: No converter available for 'position' (67)
error 2020-03-31 21:30:47: No converter available for 'position' (100)

Koenkk, 2020-04-03:
No converter for state should be fixed in latest dev branch, 

Could the seller point you to the SmartThings groovy code? Based on this we can add the same to zigbee2mqtt, I tried to find it but failed to.

slashbv, 2020-04-03:
thank you. What should I ask for ? SmartThings groovy code?

Koenkk, 2020-04-03:
Yes, it like the zigbee-herdsman-converters (devices.js) for SmartThings.

slashbv, 2020-04-03:
I have asked for groovy code. If it helps he said earlier in Smart things to use type: zigbee window shade, like in the image below.
[image]

Koenkk, 2020-04-03:
I found the source code of smartthings (https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/zigbee-window-shade.src/zigbee-window-shade.groovy) and adjusted the code of zigbee2mqtt to match that.  Please check with the latest dev branch.

slashbv, 2020-04-04:
the error message is back but with different values than before and appears every 1 or 2 minutes. After a few minutes it dissapears and it's replaced with an actual message (see highlighted messages). It seems it's trying to configure something which finally succeeds. Everything works except the slider to set position and getting intermediate position when using the switch. It is receiving something but then there is this "No converter available" message. Now in the logs you can see it is sending position and there is a new message Configuring roller_kitchen that was not there before. I believe you are close .

error 2020-04-04 08:49:31: Failed to configure 'roller_kitchen', attempt 2 (Error: ConfigureReporting 0xec1bbdfffe946658/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":62000,"reportableChange":1}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 19527 - 1 - 5 - 258 - 7 after 6000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
debug 2020-04-04 08:50:33: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data '{ "position": 78 }'
debug 2020-04-04 08:50:33: Publishing 'set' 'position' to 'roller_kitchen'
debug 2020-04-04 08:50:44: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data '{ "position": 0 }'
debug 2020-04-04 08:50:44: Publishing 'set' 'position' to 'roller_kitchen'
debug 2020-04-04 08:51:09: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data '{ "position": 49 }'
debug 2020-04-04 08:51:09: Publishing 'set' 'position' to 'roller_kitchen'
debug 2020-04-04 08:51:39: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,67,10,1,0,32,66,226,255,32,19]}'
info  2020-04-04 08:51:39: Configuring 'roller_kitchen'
error 2020-04-04 08:51:54: Failed to configure 'roller_kitchen', attempt 3 (Error: ConfigureReporting 0xec1bbdfffe946658/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":62000,"reportableChange":1}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 19527 - 1 - 9 - 258 - 7 after 6000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
**debug 2020-04-04 08:54:02: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,68,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-04 08:54:02: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,68,10,1,0,32,66,226,255,32,19]}'**

--control from switch
--close -- stop -- open
debug 2020-04-04 09:14:59: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-04-04 09:14:59: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":0,"linkquality":73}'
debug 2020-04-04 09:14:59: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{**"currentPositionLiftPercentage":0**}' from endpoint 1 with groupID 0
info  2020-04-04 09:14:59: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":0,"linkquality":70}'
debug 2020-04-04 09:15:01: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"**currentPositionLiftPercentage":50**}' from endpoint 1 with groupID 0
debug 2020-04-04 09:15:03: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{**"currentPositionLiftPercentage":100**}' from endpoint 1 with groupID 0
info  2020-04-04 09:15:03: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":73}'

Koenkk, 2020-04-04:
It seems that the configure reporting is not working for this device (this is the same what the SmartThings driver does so I assume it also doesn't work there).

I already checked this `raw` `genBasic` message, but it does not seem to contain the position (It's also not mentioned in the SmartThings driver).

Regarding intermediate position, it does send some position, but as you can see only 0, 50 or 100, so actually it doesn't report it.

I'm afraid I'm out of ideas how to fix this.

The position slider should be fixable though. What do you see in the log now when setting the position via HA?

slashbv, 2020-04-04:
I understand it might not work there either and I can't test. However the seller insists it's working. 
0, 50, 100 should be good enough, I mean if it's anywhere between 0 and 100 and it sends 50 at least you know it's not closed or opened completely.
See below the log while using the slider first to some intermediate position and then to 100 completely open:
debug 2020-04-04 20:49:58: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data '{ "position": 39 }'
debug 2020-04-04 20:49:58: Publishing 'set' 'position' to 'roller_kitchen'
debug 2020-04-04 20:50:17: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data '{ "position": 100 }'
debug 2020-04-04 20:50:17: Publishing 'set' 'position' to 'roller_kitchen'

same with the slider , if you put it somewhere between 0 and 100 and it sends 50 it's better than nothing, what do you think ? Do you know any other zigbee switch ?

slashbv, 2020-04-06:
did you add any changes ?

Koenkk, 2020-04-06:
Just pushed it, position will now always be returned when setted.

slashbv, 2020-04-07:
I don't see any difference, the slide does nothing as before even if I set it to 0 or 100.  See below the log while using the slide. II ave moved the slide from 100 to 49 -> 0 ->50 ->100. Nothing happens:
debug 2020-04-07 11:03:00: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data '{ "position": 49 }'
debug 2020-04-07 11:03:00: Publishing 'set' 'position' to 'roller_kitchen'
debug 2020-04-07 11:03:03: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data '{ "position": 0 }'
debug 2020-04-07 11:03:03: Publishing 'set' 'position' to 'roller_kitchen'
debug 2020-04-07 11:03:13: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data '{ "position": 50 }'
debug 2020-04-07 11:03:13: Publishing 'set' 'position' to 'roller_kitchen'
debug 2020-04-07 11:03:17: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data '{ "position": 100 }'
debug 2020-04-07 11:03:17: Publishing 'set' 'position' to 'roller_kitchen'

this message still appears ever few minutes:
debug 2020-04-07 11:25:53: Received Zigbee message from 'roller_kitchen', type 'raw', cluster 'genBasic', data '{"type":"Buffer","data":[9,59,10,1,0,32,66,226,255,32,19]}' from endpoint 1 with groupID 0
debug 2020-04-07 11:25:53: No converter available for 'ZM-CSW032-D' with cluster 'genBasic' and type 'raw' and data '{"type":"Buffer","data":[9,59,10,1,0,32,66,226,255,32,19]}'

consider the following scenario: 
- close the shutter from switch but stop somewhere in the middle
- later you want to close it completely from HA
- in HA is showing that it's down already it doesn't know it's somewhere between 0 and 100
- you have to open it first and then to close it 
In this situation would be very good to have and intermediate position, no matter where exactly just somewhere between 0 and 100 so HA will know it's not completely closed

slashbv, 2020-04-07:
I found this in Smartthings groovy code, don't know if it helps:

def levelEventHandler(currentLevel) {
	def lastLevel = device.currentValue("level")
	log.debug "levelEventHandle - currentLevel: ${currentLevel} lastLevel: ${lastLevel}"
	if (lastLevel == "undefined" || currentLevel == lastLevel) { //Ignore invalid reports
		log.debug "Ignore invalid reports"
	} else {
		sendEvent(name: "level", value: currentLevel)
		if (currentLevel == 0 || currentLevel == 100) {
			sendEvent(name: "windowShade", value: currentLevel == 0 ? "closed" : "open")
		} else {
			if (lastLevel < currentLevel) {
				sendEvent([name:"windowShade", value: "opening"])
			} else if (lastLevel > currentLevel) {
				sendEvent([name:"windowShade", value: "closing"])
			}
			runIn(1, "updateFinalState", [overwrite:true])
		}
	}
}

def updateFinalState() {
	def level = device.currentValue("level")
	log.debug "updateFinalState: ${level}"
	if (level > 0 && level < 100) {
		sendEvent(name: "windowShade", value: "partially open")
	}
}

Koenkk, 2020-04-08:
That makes me doubt if this device actually supports setting a position. It would also explain why it doesn't send an intermediate position.

slashbv, 2020-05-10:
I don't know if it's helpful or not but I have just realised that position 50 is sent every time the circuit is stopped. If I press close physical button it sends 0, then I press stop it sends 50. If I press close it I let it go all the way down, I don't press stop button, the circuit is stopped after a while automatically, it sends again 50.

--here I press OPEN

debug 2020-05-10 **18:56:23**: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100}' from endpoint 1 with groupID 0
info  2020-05-10 18:56:23: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":76}'

--here I press STOP

debug 2020-05-10 **18:56:27**: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0

--here I press CLOSE
debug 2020-05-10 **18:56:29**: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-05-10 **18:56:29**: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":0,"linkquality":73}'

--Here the switch stops by itself

debug 2020-05-10 **18:57:30**: Received Zigbee message from 'roller_kitchen', type 'attributeReport', 
cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0

--Now from HA - same behaviour

debug 2020-05-10 19:20:02: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data 'CLOSE'
debug 2020-05-10 19:20:02: Publishing 'set' 'state' to 'roller_kitchen'
debug 2020-05-10 19:20:02: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
info  2020-05-10 19:20:02: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":0,"linkquality":76}'
debug 2020-05-10 19:20:04: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data 'STOP'
debug 2020-05-10 19:20:04: Publishing 'set' 'state' to 'roller_kitchen'
debug 2020-05-10 19:20:04: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0
debug 2020-05-10 19:20:32: Received MQTT message on 'zigbee2mqtt/roller_kitchen/set' with data 'OPEN'
debug 2020-05-10 19:20:32: Publishing 'set' 'state' to 'roller_kitchen'
debug 2020-05-10 **19:20:32**: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100}' from endpoint 1 with groupID 0
info  2020-05-10 19:20:32: MQTT publish: topic 'zigbee2mqtt/roller_kitchen', payload '{"position":100,"linkquality":76}'
debug 2020-05-10 **19:21:33**: Received Zigbee message from 'roller_kitchen', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50}' from endpoint 1 with groupID 0

ullrik, 2020-06-19:
In advance, sorry for my English.
Wouldn't it be possible to just ignore the "currentPositionLiftPercentage = 50" which comes around 30 sec after each up or down programming?
Because now if I press stop, when my roller shutter is not at the maximum or the minimum, I no longer have the information.

In addition, on the Zemismart WiFi model, since the last version of the Framework, the position seems well indicated now. However, you must connect to a local web page to adjust the real positions of the roller shutter.

ullrik, 2020-06-21:
I have create a solution and for me it's work : 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/1336
juste add two properties in your configuration.yaml

slashbv, 2020-07-09:
@ullrik how can I test this ? what should I add to configuration.yaml ?

ullrik, 2020-07-10:
I indicated on the discussion here : Koenkk/zigbee-herdsman-converters#1336
But I will try to provide more information.
So, if you get the last version (or my file update), you must also modify your configuration.yaml exemple : 
> devices:
>   '0xec1bbdfffe900000': 
>     friendly_name: MyCover 
>     time_close: 27 
>     time_open: 28

*For set time_close and time_open, you must close completely by recording the time (start, end) and the same thing to open.

slashbv, 2020-07-10:
thanks, do I have to switch to dev to test this ?

ullrik, 2020-07-10:
I don't know, but for me my update is on Master now
Maybe you can update manually your file fromZigbee.js (in zigbee-herdsman-converters/blob/master/converters/)

Koenkk, 2020-07-10:
> thanks, do I have to switch to dev to test this ?

yes (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html), until next zigbee2mqtt.

slashbv, 2020-07-12:
seems to be good so far, thank you very much both. There is still one issue, the position slider moves to show the position but if you want to use it to move the shutter to some desired position it does nothing.

ullrik, 2020-07-12:
Yes, I tried, but it's not in fromZigbee.js, but it's in toZigbee.js. And unfortunately for the moment I have not found a solution. 
You should be able to launch a first action and after X seconds, you would need a second action to stop (apart from 0 and 100%).

@Koenkk It is possible to set 100 for up cover and after X seconds set 50 for stop with frame MQTT ?

ullrik, 2020-07-12:
Finally, I looked a little more side of the file toZigbee.js. I have modify for have this : 
`            await entity.write('closuresWindowCovering', {position: value});
`
Because I would like to send this MQTT : 
`MQTT publish: topic 'zigbee2mqtt/InterVolet_BaieVitree_Salon', payload '{"position":3,"linkquality":21}'
`
But with my update or other trying, I have always this error : 
```
Publish 'set' 'position' to 'InterVolet_BaieVitree_Salon' failed: 'Error: Unknown attribute 'position', specify either an existing attribute or a number'
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'position' to 'InterVolet_BaieVitree_Salon' failed: 'Error: Unknown attribute 'position', specify either an existing attribute or a number'","meta":{"friendly_name":"InterVolet_BaieVitree_Salon"}}'
```
Any idea for set the good frame MQTT ? maybe with "await entity.write" or "await entity.command"

ullrik, 2020-07-13:
@slashbv, in Home Assistant, do you know what is sent with the up or down button?
I think that by understanding the command which is sent by Home Assistant for the three actions, should not be able to send the same command for toZigbee.js

ullrik, 2020-07-15:
Finally, I found the solution (sorry for the previous spam questions/messages :) )
I will post my change when I have cleaned up my code.

slashbv, 2020-08-12:
I was in vacation, and now that I am back my zigbee network is completly down, I am not able to connect anything to coordinator. I suspect it may be damaged. I will receive a new CC2531 usb stick in the next days and I will be able to test. Is this in the stable build or dev ?

slashbv, 2020-08-19:
everything looks fine until I restart HA. Then the status is unknown.

slashbv, 2020-09-04:
don't know if it's related but cover.set_cover_position  service doesn't work.  cover.close_cover and cover.open_cover work fine

barsurfer, 2020-09-30:
@ullrik great work done! For `time_close` / `time_open` properties to work, should both `fromZigbee` / `toZigbee` use `ZMCSW032D_cover_position_tilt`? I can see changes on master for `fz.ZMCSW032D_cover_position_tilt`, but it looks like the PR for the `tz.ZMCSW032D_cover_position_tilt` was not approved and merged.

slashbv, 2020-10-14:
@ullrik would it be possible to send stop command after time_close / time_open has passed ?

ullrik, 2020-10-19:
I don't know in "fromZigbee.js" but in "toZigbee.js" you can use this command : 
`await entity.command()`
Exemple :
`await entity.command('closuresWindowCoveringZMCSW032D', "open", {}, getOptions(meta.mapped));`
You can change open by "stop" or "close". 
For "closuresWindowCoveringZMCSW032D" : I had duplicate "closuresWindowCovering" in zigbee-herdsman\dist\zcl\definition\cluster.js

closed 2020-11-26 by github-actions[bot]

KaHeMu, 2021-03-11:
@ullrik Did your changes referred above ever make it into the main trunk? I'm using Z2M v.1.17 using your time_close, time_open seetings described above but still have issues with the ZM-CSW032-D reporting:
```
Zigbee2MQTT:info  2021-03-11 16:56:51: MQTT publish: topic 'zigbee2mqtt/BWM_EG_Flur', payload '{"battery":100,"last_seen":"2020-03-02T10:17:11+00:00","linkquality":47,"occupancy":false,"voltage":3025}'
Zigbee2MQTT:debug 2021-03-11 16:56:54: Received Zigbee message from 'SchlaZi_Rollo', type 'raw', cluster 'genBasic', data '{"data":[9,47,10,1,0,32,66,226,255,32,19],"type":"Buffer"}' from endpoint 1 with groupID 0

```
(P.S.: bare with me if this is the wrong approach. I'm bloody new to github and this is my first post ever. Sorry if wrong).

ullrik, 2021-03-21:
@KaHeMu I have update my previous code in this commit : 
[https://github.com/Koenkk/zigbee-herdsman-converters/pull/2405](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2405)