Koenkk/zigbee2mqtt #3217: Kwikset Lock 10_L Variant support

Botched1, 2020-03-28:
Running current latest-dev Docker from 27/03/2020.

Kwikset 914 Zigbee Lock paired as unsupported.

Added the following to devices.js, and it works.

devices.js:
```
    {
        zigbeeModel: ['SMARTCODE_DEADBOLT_10_L'],
        model: 'unknown',
        vendor: 'Kwikset',
        description: 'SmartCode 10 Touch',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
        toZigbee: [tz.generic_lock],
        meta: {configureKey: 3},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(2);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

database.db:
```
{"id":56,"type":"EndDevice","ieeeAddr":"0x002446fffd036946","nwkAddr":5462,"manufId":4242,"manufName":"Kwikset","powerSource":"Battery","modelId":"SMARTCODE_DEADBOLT_10_L","epList":[2],"endpoints":{"2":{"profId":260,"epId":2,"devId":10,"inClusterList":[0,1,3,4,5,9,32,257,1026,2821,64957],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"SMARTCODE_DEADBOLT_10_L","manufacturerName":"Kwikset","powerSource":3,"zclVersion":1,"appVersion":7,"stackVersion":2,"hwVersion":3}}},"binds":[{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca16309","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca16309","endpointID":1}]}},"appVersion":7,"stackVersion":2,"hwVersion":3,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":3},"lastSeen":1585423269153}
```

Thanks!

Koenkk, 2020-03-28:
Could you provide a link (url) to the device?

Botched1, 2020-03-28:
> Could you provide a link (url) to the device?

My appologies:
https://www.kwikset.com/products/detail/914-smartcode-traditional-electronic-deadbolt-with-zigbee-technology

Koenkk, 2020-03-28:
Added, thanks!

closed 2020-03-28 by Koenkk