Koenkk/zigbee2mqtt #3218: SmartThings Leak Sensor (2016) Not Reporting Temperature

Botched1, 2020-03-28:
Running latest-dev Docker from 27/03/2020.

I noticed that the Smartthings 2016 Leak Sensor does not report temperature on Z2M, but does on other hubs.

Is this type of issue something you like reports on? Because there are a few other devices I have that have additional sensor values not reported in Z2M, but appear in other hubs. If you are interested, I'll file other issue tickets on those devices.

database.db:
```
{"id":10,"type":"EndDevice","ieeeAddr":"0x24fd5b00010c9991","nwkAddr":51915,"manufId":4362,"manufName":"SmartThings","powerSource":"Unknown","modelId":"moisturev4","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,15,32,1026,1280],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"moisturev4","manufacturerName":"SmartThings","powerSource":0,"zclVersion":1,"appVersion":27,"swBuildId":"R860_#11"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001ca16309","zoneState":1}},"genPowerCfg":{"attributes":{"batteryVoltage":28}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca16309","endpointID":1}]}},"appVersion":27,"swBuildId":"R860_#11","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1585381038461}

```

Koenkk, 2020-03-28:
Can you try with the following definition in devices.js (repair the device after this). See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to reach devices.js

```js
{
    zigbeeModel: ['moisturev4'],
    model: 'STS-WTR-250',
    vendor: 'SmartThings',
    description: 'Water leak sensor (2016 model)',
    supports: 'water leak',
    fromZigbee: [fz.ias_water_leak_alarm_1, fz.battery_3V, fz.temperature],
    toZigbee: [],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'msTemperatureMeasurement']);
        await configureReporting.batteryVoltage(endpoint);
        await configureReporting.temperature(endpoint);
    },
},
```

Botched1, 2020-03-28:
> Can you try with the following definition in devices.js (repair the device after this). See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to reach devices.js
>
> ```js
> { [...]

Yes, that worked. Temperature is reporting now.

Koenkk, 2020-03-29:
Thanks, added, will be available in the latest dev branch in a few hours.

closed 2020-03-29 by Koenkk