Koenkk/zigbee2mqtt #3219: NYCE Ceiling Motion Sensor Not Reporting Humidity or Temperature

Botched1, 2020-03-28:
Running latest-dev Docker from 27/03/2020.

NYCE Ceiling Motion sensor pairs correctly, but only shows motion. The device also reports Humidity and Temperature on other hubs.

database.db:
```
{"id":50,"type":"EndDevice","ieeeAddr":"0x000d6f00132aa990","nwkAddr":25047,"manufId":4281,"manufName":"NYCE","powerSource":"Battery","modelId":"3043","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1030,1280,32,1026,1029],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"3043","manufacturerName":"NYCE","powerSource":3,"zclVersion":1,"appVersion":21,"stackVersion":0,"hwVersion":0,"swBuildId":"b80"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001ca16309","zoneState":0}}},"binds":[]}},"appVersion":21,"stackVersion":0,"hwVersion":0,"swBuildId":"b80","zclVersion":1,"interviewCompleted":false,"meta":{},"lastSeen":1585421101117}
```

Koenkk, 2020-03-28:
Can you try with the following definition in devices.js (repair the device after this). See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to reach devices.js

```js
{
    zigbeeModel: ['3043'],
    model: 'NCZ-3043-HA',
    vendor: 'Nyce',
    description: 'Ceiling motion sensor',
    supports: 'motion, humidity and temperature',
    fromZigbee: [
        fz.occupancy, fz.humidity, fz.temperature, fz.ignore_basic_report,
        fz.ignore_genIdentify, fz.ignore_poll_ctrl,
        fz.battery_not_divided, fz.ignore_iaszone_report,
        fz.ias_occupancy_alarm_2,
    ],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const bindClusters = ['msTemperatureMeasurement', 'msRelativeHumidity'];
        await bind(endpoint, coordinatorEndpoint, bindClusters);
        await configureReporting.temperature(endpoint);
        await configureReporting.humidity(endpoint);
    },
},
```

Botched1, 2020-03-28:
Unfortunately that didn't work. Various errors both after re-load and on pairing retry. Unfortunately this is on a 7m high ceiling, so is not very easy to get it...

```
error 2020-03-28 15:38:06: Failed to configure 'Master Bathroom Shower Motion', attempt 1 (Error: Bind 0x000d6f00132aa990/1 msTemperatureMeasurement from '0x00124b001ca16309/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

```
error 2020-03-28 15:39:29: Failed to configure 'Master Bathroom Shower Motion', attempt 1 (Error: Bind 0x000d6f00132aa990/1 msTemperatureMeasurement from '0x00124b001ca16309/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```
```
error 2020-03-28 15:41:05: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /app/lib/extension/deviceReceive.js:106:42
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:105:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:346:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:231:14)
    at Zigbee.emit (events.js:311:20)
    at Controller.<anonymous> (/app/lib/zigbee.js:58:52)
    at Controller.emit (events.js:311:20)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:463:22)
    at Generator.next (<anonymous>))
```

```
error 2020-03-28 15:41:16: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /app/lib/extension/deviceReceive.js:106:42
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:105:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:346:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:231:14)
    at Zigbee.emit (events.js:311:20)
    at Controller.<anonymous> (/app/lib/zigbee.js:58:52)
    at Controller.emit (events.js:311:20)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:463:22)
    at Generator.next (<anonymous>))
```

closed 2020-06-06 by stale[bot]

kailampi, 2021-04-24:
Can we continue in this thread. 

I have similar problem. Koenkk's solution worked in 1.14.x installation, but somewhere around 1.17.x that config started to generate errors and only default config let zigbee2mqtt to start. 

The strange thing is that three of my NYCEs work (ie. send temp and humidity reports) with the new default config and other three don't. Any ideas how to debug what is different in these devices and how sensors should be configured with the current devices.js config?