Koenkk/zigbee2mqtt #3220: Delay after switching ON not available device

wizu, 2020-03-28:
# Bug Report
## What happened
After update to the new version my night lamp started to turn on after a delay - about 2-3 seconds
I use a flow: Node-RED > Home Assistant > zigbee2mqtt and had some IKEA sockets defined in "switch on" task which are no longer in use. 

## What did you expect to happen
It worked immediatele

## How to reproduce it (minimal and precise)
It looks like, when you try to turn on https://www.zigbee2mqtt.io/devices/E1603_E1702.html and it is not available, then error is thrown and that generates a delay in executing next tasks.

## Debug Info
zigbee2mqtt      | zigbee2mqtt:error 2020-03-28 19:26:44: Publish 'set' 'state' to 'ikea_socket_b' failed: 'Error: Command 0x086bd7fffe42c0c1/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'

zigbee2mqtt version: 1.12.0
CC253X firmware version: 20190608

Koenkk, 2020-03-28:
The reason for this is that Zigbee2mqtt has a builtin protection mechanism for the CC2531 to prevent it from being ddosed (resulting in a crash). It allows max 2 commands to be active at a time. There are 2 ways to work around this:
- Don't control switches which are not available (and thus the queue won't get blocked)
- Use a more powerful adapter, e.g. CC26X2R1 which allows for 16 commands to be active at a time (https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc26x2r1)

wizu, 2020-03-30:
Thanks @Koenkk for your work. I've removed unused controls from my flow and of course everything works great now.
It's time to upgrade my adapter ;)

closed 2020-03-30 by wizu