Koenkk/zigbee2mqtt #3221: MoesHouse Zigbee Thermostatic Radiator Valve

Westcott1, 2020-03-28:
Just got one of these from AliExpress -
[https://www.aliexpress.com/item/4000738309287.html?spm=a2g0s.9042311.0.0.14594c4dXTky2m](url)
It claims to be "Tuya ZigBee 3.0"
z2m does not see anything from it, and there does not seem to be a reset mechanism.
The "instructions" suggest that it will pair automatically with a Moes Tuya hub.

Has anyone else seen one of these?

sjorge, 2020-03-29:
There is another issues somewhere about another TuYa ZB3 TRV but it looks different, it is not supported either though.

Before you try and pair again, stop z2m and physically unplug and replay the usb stick.

Then try again, hopefully you will atleast get the unsupported device message. Then we might have some usable info from database.db

kirovilya, 2020-04-11:
I have such. It have almost the same control as Siterwell https://github.com/Koenkk/zigbee2mqtt/issues/3113
I plan to do PR in the near future.

kirovilya, 2020-04-12:
@Westcott1 For pair you need to press and hold "Auto mode" and "Comfort mode" two buttons at the same time. Then "signal" start blinking
[image]

Westcott1, 2020-04-12:
Thanks Kirovilya for the pairing info! I must have tried just about every other combination...
Thanks to you both (Koenkk) for everything z2m
As expected, using v1.12.2 I now get -
`Received message from unsupported device with Zigbee model 'kud7u2l'`

Here's the entry from database.db -
Like some other devices, it contains a few spurious "\u0000"

{"id":24,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe9466af","nwkAddr":22617,"manufId":0,"manufName":"_TYST11_ckud7u2l","powerSource":"Battery","modelId":"kud7u2l\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"kud7u2l\u0000","manufacturerName":"_TYST11_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727"}}},"binds":[]}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1586685369907}

kirovilya, 2020-04-12:
switch to dev-branch

Westcott1, 2020-04-12:
Thanks, it now adds correctly.
Is this important?
zigbee-herdsman-converters:siterwell_gs361: NOT RECOGNIZED DP #366 with data {"type":"Buffer","data":[0]}

kirovilya, 2020-04-13:
@Westcott1 normal. I do not found what 366 mean

closed 2020-06-12 by Westcott1

pedroke, 2020-06-26:
Hi guys, is this device going to be supported in official release anytime soon? Thanks.

kirovilya, 2020-06-26:
already supported

pedroke, 2020-06-26:
Thanks Ilya and of course many others! Also for working on this device and other supportive stuff for zigbee community ;)