Koenkk/zigbee2mqtt #3222: Networkmap grphviz return incomplete map

Botched1, 2020-03-28:
Maybe I just have too many devices/too big of a network, since adding my last batch of devices networkmap returns a topic message that is incomplete.

It returns a lot, but it is like the bottom part of the message is cut off.

Any thoughts?

Koenkk, 2020-03-29:
Could you provide the zigbee2mqtt log when the message is posted?

MQTT should be able to handle it as the limit of the payload is 256MB (https://stackoverflow.com/questions/41738818/how-much-data-can-hold-mqtt-in-single-message0

Botched1, 2020-03-29:
> Could you provide the zigbee2mqtt log when the message is posted?
> 
> MQTT should be able to handle it as the limit of the payload is 256MB (https://stackoverflow.com/questions/41738818/how-much-data-can-hold-mqtt-in-single-message)

Here you go:
```
zigbee2mqtt:info  2020-03-29 14:39:41: Starting network scan (includeRoutes 'true')
zigbee2mqtt:error 2020-03-29 14:39:53: Failed to execute LQI for 'Landscape Lights Left 2'
zigbee2mqtt:error 2020-03-29 14:40:05: Failed to execute LQI for 'Storage Hall Outlet'
zigbee2mqtt:error 2020-03-29 14:40:15: Failed to execute routing table for 'Storage Hall Outlet'
zigbee2mqtt:error 2020-03-29 14:40:25: Failed to execute LQI for 'Living Room Outlet'
zigbee2mqtt:error 2020-03-29 14:40:35: Failed to execute routing table for 'Living Room Outlet'
zigbee2mqtt:error 2020-03-29 14:40:45: Failed to execute LQI for 'Media Room Outlet'
zigbee2mqtt:error 2020-03-29 14:40:55: Failed to execute routing table for 'Media Room Outlet'
zigbee2mqtt:error 2020-03-29 14:41:05: Failed to execute LQI for 'Media Room Speakers'
zigbee2mqtt:error 2020-03-29 14:41:15: Failed to execute routing table for 'Media Room Speakers'
zigbee2mqtt:info  2020-03-29 14:41:16: Network scan finished
zigbee2mqtt:info  2020-03-29 14:41:16: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];
  "0x00124b001ca16309" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001ca16309 (0)|2020-03-29T14:41:16-05:00}"];
  "0x00124b001ca16309" -> "0xd0cf5efffedda73c" [penwidth=0.5, weight=0, color="#994444", label="132"]
  "0x00124b001ca16309" -> "0x000d6ffffea03f4b" [penwidth=0.5, weight=0, color="#994444", label="148"]
  "0x00124b001ca16309" -> "0xf0d1b80000078a36" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0x00124b001ca16309" -> "0xf0d1b80000078a9c" [penwidth=0.5, weight=0, color="#994444", label="232"]
  "0x00124b001ca16309" -> "0xf0d1b800000789a1" [penwidth=0.5, weight=0, color="#994444", label="235"]
  "0x00124b001ca16309" -> "0x000d6ffffe2edc19" [penwidth=0.5, weight=0, color="#994444", label="107"]
  "0x00124b001ca16309" -> "0x000b57fffed5026a" [penwidth=0.5, weight=0, color="#994444", label="102"]
  "0x00124b001ca16309" -> "0x000d6ffffea05bd4" [penwidth=0.5, weight=0, color="#994444", label="164"]
  "0x00124b001ca16309" -> "0x24fd5b00010afced" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0x086bd7fffee97848" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{zz Extra Humidity and Temperature|0x086bd7fffee97848 (58280)|Konke Temperature and humidity sensor (2AJZ4KPFT)|2020-03-29T14:21:36-05:00}"];
  "0x086bd7fffee97848" -> "0x00124b001ca16309" [penwidth=1, weight=0, color="#994444", label="98"]
  "0x000d6f0013393f63" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{MCT-350 Contact Sensor|0x000d6f0013393f63 (7347)|Visonic Magnetic door & window contact sensor (MCT-350 SMA)|2020-03-29T10:53:29-05:00}"];
  "0x000d6f0013393f63" -> "0x000d6ffffe2edc19" [penwidth=1, weight=0, color="#994444", label="136"]
  "0xd0cf5efffedda73c" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Dining Room Ikea Repeater|0xd0cf5efffedda73c (3903)|IKEA TRADFRI signal repeater (E1746)|2020-03-29T14:28:40-05:00}"];
  "0xd0cf5efffedda73c" -> "0x00124b001ca16309" [penwidth=2, weight=1, color="#009900", label="80 (routes: 3903,60295,9945,57652,13804,32751)"]
  "0xd0cf5efffedda73c" -> "0x000d6ffffea03f4b" [penwidth=2, weight=1, color="#009900", label="121 (routes: 54874)"]
  "0xd0cf5efffedda73c" -> "0xf0d1b80000078a36" [penwidth=0.5, weight=0, color="#994444", label="255"]
  "0xd0cf5efffedda73c" -> "0xf0d1b80000078a9c" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0xd0cf5efffedda73c" -> "0xf0d1b800000789a1" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0xd0cf5efffedda73c" -> "0x000d6ffffe2edc19" [penwidth=0.5, weight=0, color="#994444", label="136"]
  "0xd0cf5efffedda73c" -> "0x000b57fffed5026a" [penwidth=2, weight=1, color="#009900", label="139 (routes: 34338)"]
  "0xd0cf5efffedda73c" -> "0x000d6ffffea05bd4" [penwidth=0.5, weight=0, color="#994444", label="96"]
  "0xd0cf5efffedda73c" -> "0x24fd5b00010afced" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0x000d6ffffea03f4b" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Living Room Ikea Repeater|0x000d6ffffea03f4b (28434)|IKEA TRADFRI signal repeater (E1746)|2020-03-29T14:40:48-05:00}"];
  "0x000d6ffffea03f4b" -> "0x00124b001ca16309" [penwidth=2, weight=1, color="#009900", label="89 (routes: 5462,23939,8838,45953,43908,9587,4338,6546,13818,35036)"]
  "0x000d6ffffea03f4b" -> "0xd0cf5efffedda73c" [penwidth=0.5, weight=0, color="#994444", label="120"]
  "0x000d6ffffea03f4b" -> "0xf0d1b80000078a36" [penwidth=0.5, weight=0, color="#994444", label="255"]
  "0x000d6ffffea03f4b" -> "0xf0d1b80000078a9c" [penwidth=0.5, weight=0, color="#994444", label="255"]
  "0x000d6ffffea03f4b" -> "0xf0d1b800000789a1" [penwidth=0.5, weight=0, color="#994444", label="255"]
  "0x000d6ffffea03f4b" -> "0x000d6ffffe2edc19" [penwidth=0.5, weight=0, color="#994444", label="136"]
  "0x000d6ffffea03f4b" -> "0x000b57fffed5026a" [penwidth=0.5, weight=0, color="#994444", label="128"]
  "0x000d6ffffea03f4b" -> "0x000d6ffffea05bd4" [penwidth=0.5, weight=0, color="#994444", label="136"]
[... cut ...]
  "0x086bd7fffee982b6" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Music Room Humidity and Temperature|0x086bd7fffee982b6 (23570)|Konke Temperature and humidity sensor (2AJZ4KPFT)|2020-03-29T14:30:07-05:00}"];
  "0x086bd7fffee982b6" -> "0x000b57fffed5026a" [penwidth=1, weight=0, color="#994444", label="171"]
  "0x002446fffd036946" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Front Door Lock|0x002446fffd036946 (5462)|Kwikset SmartCode 10 Touch (unknown)|2020-03-29T14:15:00-05:00}"];
  "0x002446fffd036946" -> "0x000d6ffffea03f4b" [penwidth=1, weight=0, color="#994444", label="184"]
}'

```

Botched1, 2020-03-29:
The MQTT topic, however only has:
```
digraph G {
node[shape=record];
  "0x00124b001ca16309" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001ca16309 (0)|2020-03-29T14:41:16-05:00}"];
  "0x00124b001ca16309" -> "0xd0cf5efffedda73c" [penwidth=0.5, weight=0, color="#994444", label="132"]
  "0x00124b001ca16309" -> "0x000d6ffffea03f4b" [penwidth=0.5, weight=0, color="#994444", label="148"]
  "0x00124b001ca16309" -> "0xf0d1b80000078a36" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0x00124b001ca16309" -> "0xf0d1b80000078a9c" [penwidth=0.5, weight=0, color="#994444", label="232"]
  "0x00124b001ca16309" -> "0xf0d1b800000789a1" [penwidth=0.5, weight=0, color="#994444", label="235"]
  "0x00124b001ca16309" -> "0x000d6ffffe2edc19" [penwidth=0.5, weight=0, color="#994444", label="107"]
  "0x00124b001ca16309" -> "0x000b57fffed5026a" [penwidth=0.5, weight=0, color="#994444", label="102"]
  "0x00124b001ca16309" -> "0x000d6ffffea05bd4" [penwidth=0.5, weight=0, color="#994444", label="164"]
  "0x00124b001ca16309" -> "0x24fd5b00010afced" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0x086bd7fffee97848" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{zz Extra Humidity and Temperature|0x086bd7fffee97848 (58280)|Konke Temperature and humidity sensor (2AJZ4KPFT)|2020-03-29T14:21:36-05:00}"];
  "0x086bd7fffee97848" -> "0x00124b001ca16309" [penwidth=1, weight=0, color="#994444", label="98"]
  "0x000d6f0013393f63" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{MCT-350 Contact Sensor|0x000d6f0013393f63 (7347)|Visonic Magnetic door & window contact sensor (MCT-350 SMA)|2020-03-29T10:53:29-05:00}"];
  "0x000d6f0013393f63" -> "0x000d6ffffe2edc19" [penwidth=1, weight=0, color="#994444", label="136"]
  "0xd0cf5efffedda73c" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Dining Room Ikea Repeater|0xd0cf5efffedda73c (3903)|IKEA TRADFRI signal repeater (E1746)|2020-03-29T14:28:40-05:00}"];
  "0xd0cf5efffedda73c" -> "0x00124b001ca16309" [penwidth=2, weight=1, color="#009900", label="80 (routes: 3903,60295,9945,57652,13804,32751)"]
  "0xd0cf5efffedda73c" -> "0x000d6ffffea03f4b" [penwidth=2, weight=1, color="#009900", label="121 (routes: 54874)"]
  "0xd0cf5efffedda73c" -> "0xf0d1b80000078a36" [penwidth=0.5, weight=0, color="#994444", label="255"]
  "0xd0cf5efffedda73c" -> "0xf0d1b80000078a9c" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0xd0cf5efffedda73c" -> "0xf0d1b800000789a1" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0xd0cf5efffedda73c" -> "0x000d6ffffe2edc19" [penwidth=0.5, weight=0, color="#994444", label="136"]
  "0xd0cf5efffedda73c" -> "0x000b57fffed5026a" [penwidth=2, weight=1, color="#009900", label="139 (routes: 34338)"]
  "0xd0cf5efffedda73c" -> "0x000d6ffffea05bd4" [penwidth=0.5, weight=0, color="#994444", label="96"]
  "0xd0cf5efffedda73c" -> "0x24fd5b00010afced" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0x000d6ffffea03f4b" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Living Room Ikea Repeater|0x000d6ffffea03f4b (28434)|IKEA TRADFRI signal repeater (E1746)|2020-03-29T14:40:48-05:00}"];
  "0x000d6ffffea03f4b" -> "0x00124b001ca16309" [penwidth=2, weight=1, color="#009900", label="89 (routes: 5462,23939,8838,45953,43908,9587,4338,6546,13818,35036)"]
  "0x000d6ffffea03f4b" -> "0xd0cf5efffedda73c" [penwidth=0.5, weight=0, color="#994444", label="120"]
  "0x000d6ffffea03f4b" -> "0xf0d1b80000078a36" [penwidth=0.5, weight=0, color="#994444", label="255"]
  "0x000d6ffffea03f4b" -> "0xf0d1b80000078a9c" [penwidth=0.5, weight=0, color="#994444", label="255"]
  "0x000d6ffffea03f4b" -> "0xf0d1b800000789a1" [penwidth=0.5, weight=0, color="#994444", label="255"]
  "0x000d6ffffea03f4b" -> "0x000d6ffffe2edc19" [penwidth=0.5, weight=0, color="#994444", label="136"]
  "0x000d6ffffea03f4b" -> "0x000b57fffed5026a" [penwidth=0.5, weight=0, color="#994444", label="128"]
  "0x000d6ffffea03f4b" -> "0x000d6ffffea05bd4" [penwidth=0.5, weight=0, color="#994444", label="136"]
  "0x000d6ffffea03f4b" -> "0x24fd5b00010afced" [penwidth=0.5, weight=0, color="#994444", label="254"]
  "0xf0d1b80000078a36" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Landscape Lights Left 1|0xf0d1b80000078a36 (9762)|Sylvania SMART+ Outdoor Accent RGB lighting kit (75541)|2020-03-29T12:41:04-05:00}"];
  "0xf0d1b80000078a36" -> "0x00124b001ca16309" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0xf0d1b80000078a36" -> "0xd0cf5efffedda73c" [penwidth=0.5, weight=0, color="#994444", label="112"]
  "0xf0d1b80000078a36" -> "0x000d6ffffea03f4b" [penwidth=0.5, weight=0, color="#994444", label="52"]
  "0xf0d1b80000078a36" -> "0xf0d1b80000078a9c" [penwidth=0.5, weight=0, color="#994444", label="238"]
  "0xf0d1b800000789de" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Landscape Lights Left 2|0xf0d1b800000789de (53862)failed: lqi|Sylvania SMART+ Outdoor Accent RGB lighting kit (75541)|2020-03-29T13:05:34-05:00}"];
  "0xf0d1b800000789de" -> "0xd0cf5efffedda73c" [penwidth=0.5, weight=0, color="#994444", label="88"]
  "0xf0d1b800000789de" -> "0xf0d1b80000078a36" [penwidth=2, weight=1, color="#009900", label="254 (routes: 53862)"]
  "0xf0d1b800000789de" -> "0xf0d1b80000078a9c" [penwidth=0.5, weight=0, color="#994444", label="219"]
[... cut ...]
  "0x90fd9ffffe329c99" -> "0x00124b001ca16309" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0xd0cf5efffe7aceb1" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Media Room Speakers|0xd0cf5efffe7aceb1 (45953)failed: lqi,routingTable|IKEA TRADFRI control outlet (E1603/E1702)|2020-03-28T17:20:11-05:00}"];
  "0xd0cf5efffe7aceb1" -> "0x00124b001ca16309" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x086bd7fffe915a36" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Boys Bathr
```

Koenkk, 2020-03-30:
Is there any way to find out what is received by mosquitto? (to determine where the issues comes from)

vikinghts, 2020-06-05:
I have the same problem. Let me look into if i can see what mosquitto says

essera, 2020-06-20:
Hi,
my network map is also incomplete. I work wirh Fhem internal MQTT Broker. But the map information i copy from the Z2M Log is the same i receive in my Broker. There are all device listed but no connection informations.

digraph G {node[shape=record];

  "0x00124b001ca604ba" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001ca604ba (0)|2020-06-20T12:03:47+02:00}"];

  "0x00124b001ca604ba" -> "0x7cb03eaa0a068f59" [penwidth=0.5, weight=0, color="#994444", label="18"]

  "0x00158d00023102e2" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bizTemp|0x00158d00023102e2 (33877)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-06-20T12:02:02+02:00}"];

  "0x00158d0002e9f93c" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{kellerAg|0x00158d0002e9f93c (30183)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2020-06-20T11:29:11+02:00}"];

  "0x0017880108641bc3" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bzMove2|0x0017880108641bc3 (49846)|Philips Hue motion sensor (9290012607)|2020-06-20T12:01:40+02:00}"];

  "0x00158d000257d9f9" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{akz|0x00158d000257d9f9 (11356)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2020-06-20T12:02:32+02:00}"];

  "0x00158d00023a7fb7" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bzTemp|0x00158d00023a7fb7 (43665)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-06-20T11:56:30+02:00}"];

  "0x00158d0002adc401" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{azVibration1|0x00158d0002adc401 (58194)|Xiaomi Aqara vibration sensor (DJT11LM)|2020-06-20T12:03:03+02:00}"];

  "0x00158d000238f9b7" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{wzTemp|0x00158d000238f9b7 (14009)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-06-20T11:34:29+02:00}"];

  "0x00158d000238fb9c" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{szTemp|0x00158d000238fb9c (21159)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-06-20T11:32:26+02:00}"];

  "0x00158d0002327a09" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{aussenTemp|0x00158d0002327a09 (35869)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-06-20T11:54:44+02:00}"];

  "0x00158d0002327a09" -> "0x7cb03eaa0a068f59" [penwidth=1, weight=0, color="#994444", label="48"]

  "0x00158d000232279a" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{defekt|0x00158d000232279a (42963)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-06-20T12:01:48+02:00}"];

  "0x00158d000273008a" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{azTemp|0x00158d000273008a (10993)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-06-20T12:02:37+02:00}"];

  "0x00158d00029ac648" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{azCube1|0x00158d00029ac648 (21166)|Xiaomi Mi/Aqara smart home cube (MFKZQ01LM)|2020-06-20T11:05:55+02:00}"];

  "0x00158d000257d9e6" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{wk1|0x00158d000257d9e6 (11984)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2020-06-20T11:47:27+02:00}"];

  "0x00158d000241790d" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{wk2|0x00158d000241790d (4896)|Xiaomi MiJia human body movement sensor (RTCGQ01LM)|2020-06-20T11:13:18+02:00}"];

  "0x00158d00032b8722" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{azWindow1|0x00158d00032b8722 (50855)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|2020-06-20T11:56:27+02:00}"];

  "0x00158d0002886aa9" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{azWasser2|0x00158d0002886aa9 (64028)|Xiaomi Aqara water leak sensor (SJCGQ11LM)|2020-04-26T15:04:27+02:00}"];

  "0x00158d0001eb7f10" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{szBedAndy|0x00158d0001eb7f10 (56822)|Xiaomi Aqara wireless switch (with gyroscope) (WXKG12LM)|2020-06-20T11:12:08+02:00}"];

  "0x00158d00031484ac" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bzMove|0x00158d00031484ac (19861)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2020-06-20T12:03:23+02:00}"];

  "0x00158d000424ccc5" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{szBedSteffi|0x00158d000424ccc5 (4369)|Xiaomi Aqara wireless switch (WXKG11LM)|2020-06-20T11:15:00+02:00}"];

  "0x00158d0002ec30c0" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{kellerDg|0x00158d0002ec30c0 (25287)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2020-06-20T11:39:41+02:00}"];

  "0x00158d0002ec30c0" -> "0x00124b001ca604ba" [penwidth=1, weight=0, color="#994444", label="77"]

[... cut ...] 0}"];

  "0x0017880104ad4092" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{HueSchalt1|0x0017880104ad4092 (28060)|Philips Hue dimmer switch (324131092621)|2020-06-20T11:49:46+02:00}"];

}

Sometimes are more infos for the single device, but i never had the situation that all device are shown with their correct info.s.

This situation is the same since the start with Z2M and i now working with version 1.13 and will chenge to 1.14 in the next day.

Any ideas ?

closed 2020-07-27 by stale[bot]

Bruesefran, 2020-08-23:
Just in case somebodystumbles accross this... This error may occur when using MQTT Explorer which cannot handlelarge amounts of data. Thus appears incomplete in MQTT explorer while complete elsewhere (MQTTfix or within openHAB in my case)