Koenkk/zigbee2mqtt #3223: Smartthings outlet reporting power every 5s - even when off

Botched1, 2020-03-28:
I have a smartthings outletv4. Everything works correctly on it, but the power reporting seems VERY frequent. It reports EVERY 5s, even when the outlet is off (due to incoming power fluctuation).

Is there any way to configure that to be less "chatty"?

database.db:
```
{"id":45,"type":"Router","ieeeAddr":"0x24fd5b00010afced","nwkAddr":4200,"manufId":4362,"manufName":"SmartThings","powerSource":"Unknown","modelId":"outletv4","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2,"inClusterList":[0,3,4,5,6,9,15,2820],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"outletv4","manufacturerName":"SmartThings","powerSource":0,"zclVersion":1,"appVersion":18,"hwVersion":10,"swBuildId":"R880_#162"}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":10,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":10,"rmsVoltage":1179,"rmsCurrent":5,"activePower":0}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca16309","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca16309","endpointID":1}]}},"appVersion":18,"hwVersion":10,"swBuildId":"R880_#162","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1585422905238}
```

Koenkk, 2020-03-29:
Try changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L4429 to `await configureReporting.rmsVoltage(endpoint, {reportableChange: 10});` and increment the `configureKey` by one (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L4421). If that doesn't work try with `reportableChange: 100`.

Botched1, 2020-03-29:
> Try changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L4429 to `await configureReporting.rmsVoltage(endpoint, {reportableChange: 10});` and increment the [...]

"reportableChange: 10" seemed to quiet it down considerably. Thanks!

Koenkk, 2020-03-30:
Added, thanks!

closed 2020-03-30 by Koenkk