Koenkk/zigbee2mqtt #3224: IKEA LED1623G12 reporting breaks after power cycle

ofalvai, 2020-03-28:
# Bug Report
## What happened
I control a LED1623G12 bulb with a E1743 remote via binding. I also have reporting enabled, which is working correctly after I set up everything. However I noticed that reporting breaks if I physically turn off the bulb, and then turn on again. This is what I saw in the log when I turned the bulb on:

```
zigbee2mqtt:info  2020-03-28 18:41:03: MQTT publish: topic 'zigbee2mqtt/light_bedroom/availability', payload 'online'
zigbee2mqtt:error 2020-03-28 18:41:13: Failed to setup reporting for '0x000b57fffe8f3e58' - Error: Bind 0x000b57fffe8f3e58/1 genOnOff from '0x00124b000fea91a5/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
``` 

Another error a few hours later:
```
zigbee2mqtt:error 2020-03-28 20:41:04: Failed to read state of 'light_bedroom' after reconnect
```

I couldn't fix reporting, had to remove the device from zigbee2mqtt, do a hardware reset on the bulb, then set up everything again.

## Debug Info
zigbee2mqtt version: 1.12.0
CC253X firmware version: rev. 20190608

Koenkk, 2020-03-29:
Are there any OTA updates available for the bulb? (https://www.zigbee2mqtt.io/information/ota_updates.html)

ofalvai, 2020-03-29:
No updates for the bulb, but there was an update for the switch that I installed previously.

ofalvai, 2020-03-30:
Tried to reproduce the problem again and definitely not every power cycle causes the failure in reporting.
Another observation that might be useful:
After the reporting fails, it's enough to remote the bulb from the network by the MQTT command, then power cycle it to join again (no need to hardware reset as I stated before). Reporting is working again after this.

Koenkk, 2020-03-30:
Could it be that the connection between zigbee2mqtt is not good?

ofalvai, 2020-03-30:
The link quality for the bulb is between 40 and 70 usually. Not sure if this is considered bad or just average, I don't have reliability problems with my Zigbee setup.

I started logging the link quality a few days ago, so I'll be able to notice if and when the reporting stops, and what the last link quality value was.
[image]

connesc, 2020-04-12:
I'm also facing this with my IKEA LED1624G9 (latest firmware version, excellent signal strength).
However, it happens randomly.

I have first seen this error after upgrading my CC2531 from 20190425 to 20190608. It's probably a clue, but could be a coincidence as well.

Koenkk, 2020-04-13:
It may be because the source routing firmware is more stable: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

connesc, 2020-04-13:
You're right, I'm not able to reproduce the error after switching to source routing firmware.
I still have `Failed to read state of 'my_bulb' after reconnect`, but it doesn't seem to be problematic.

ofalvai, 2020-04-26:
This happened again a few days ago:

```
info  2020-04-23 22:53:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"light_bedroom"}}'
info  2020-04-23 22:53:36: MQTT publish: topic 'zigbee2mqtt/light_bedroom/availability', payload 'online'
info  2020-04-23 22:53:36: MQTT publish: topic 'zigbee2mqtt/group_bedroom', payload '{"state":"ON","brightness":1}'
info  2020-04-23 22:53:36: MQTT publish: topic 'zigbee2mqtt/light_bedroom', payload '{"update_available":false,"state":"ON","linkquality":49,"brightness":1}'
info  2020-04-23 22:53:37: Successfully setup reporting for '0x000b57fffe8f3e58' - 1 - genOnOff
info  2020-04-23 22:53:37: MQTT publish: topic 'zigbee2mqtt/group_bedroom', payload '{"state":"ON","brightness":1}'
info  2020-04-23 22:53:37: MQTT publish: topic 'zigbee2mqtt/light_bedroom', payload '{"update_available":false,"state":"ON","linkquality":44,"brightness":1}'
info  2020-04-23 22:53:37: Successfully setup reporting for '0x000b57fffe8f3e58' - 1 - genLevelCtrl
info  2020-04-23 22:53:38: MQTT publish: topic 'zigbee2mqtt/group_bedroom', payload '{"state":"ON","brightness":1}'
info  2020-04-23 22:53:38: MQTT publish: topic 'zigbee2mqtt/light_bedroom', payload '{"update_available":false,"state":"ON","linkquality":47,"brightness":1}'
info  2020-04-23 22:53:40: MQTT publish: topic 'zigbee2mqtt/group_bedroom', payload '{"state":"ON","brightness":1}'
info  2020-04-23 22:53:40: MQTT publish: topic 'zigbee2mqtt/light_bedroom', payload '{"update_available":false,"state":"ON","linkquality":47,"brightness":1}'
info  2020-04-23 22:54:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"light_bedroom"}}'
info  2020-04-23 22:54:49: Successfully setup reporting for '0x000b57fffe8f3e58' - 1 - genOnOff
info  2020-04-23 22:54:49: MQTT publish: topic 'zigbee2mqtt/group_bedroom', payload '{"state":"ON","brightness":1}'
info  2020-04-23 22:54:49: MQTT publish: topic 'zigbee2mqtt/light_bedroom', payload '{"update_available":false,"state":"ON","linkquality":70,"brightness":1}'
error 2020-04-23 22:54:59: Failed to setup reporting for '0x000b57fffe8f3e58' - Error: Bind 0x000b57fffe8f3e58/1 genLevelCtrl from '0x00124b000fea91a5/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
```

Light was turned on, reporting was set up successfully. Then, a minute later the same happens again, except reporting setup fails for `genLevelCtrl`

ofalvai, 2020-04-26:
I'm not sure it's relevant, but maybe helps: When I removed the light to reconfigure, the logs tell me that it was removed twice:

```
info  2020-04-26 10:20:09: Removing 'light_bedroom'
info  2020-04-26 10:20:09: Clearing Home Assistant discovery topic for '0x000b57fffe8f3e58'
info  2020-04-26 10:20:09: MQTT publish: topic 'homeassistant/light/0x000b57fffe8f3e58/light/config', payload 'null'
info  2020-04-26 10:20:09: MQTT publish: topic 'homeassistant/sensor/0x000b57fffe8f3e58/linkquality/config', payload 'null'
info  2020-04-26 10:20:09: MQTT publish: topic 'homeassistant/binary_sensor/0x000b57fffe8f3e58/update_available/config', payload 'null'
info  2020-04-26 10:20:09: Successfully removed light_bedroom
info  2020-04-26 10:20:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"light_bedroom"}'
warn  2020-04-26 10:20:10: Device '0x000b57fffe8f3e58' left the network
info  2020-04-26 10:20:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x000b57fffe8f3e58"}}'
warn  2020-04-26 10:20:10: Device '0x000b57fffe8f3e58' left the network
info  2020-04-26 10:20:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x000b57fffe8f3e58"}}'
```

closed 2020-07-03 by stale[bot]

ofalvai, 2020-07-09:
Let's reopen the issue, it's still happening after a few days of usage.