Koenkk/zigbee2mqtt #3226: Failed to pair Xiaomi DJT11LM

nateww, 2020-03-29:
# Bug Report
## What happened
Attempted to pair a device, and it failed with:
```
info  2020-03-28 20:05:21: MQTT publish: topic 'zigbee2mqtttest/bridge/config', payload '{"version":"1.12.0","commit":"840b9d9","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
info  2020-03-28 20:05:32: Device '0x00158d0002c5b501' joined
info  2020-03-28 20:05:32: MQTT publish: topic 'zigbee2mqtttest/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0002c5b501"}}'
info  2020-03-28 20:05:32: Starting interview of '0x00158d0002c5b501'
info  2020-03-28 20:05:32: MQTT publish: topic 'zigbee2mqtttest/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0002c5b501"}}'
error 2020-03-28 20:06:32: Failed to interview '0x00158d0002c5b501', device has not successfully been paired
info  2020-03-28 20:06:32: MQTT publish: topic 'zigbee2mqtttest/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00158d0002c5b501"}}'
```

## What did you expect to happen
I expected the pairing to succeed.

## How to reproduce it (minimal and precise)
Start a stock version of zigbee2mqtt
```
$  npm ci
...
$ DEBUG=zigbee-herdsman* npm start
[ Detailed provided at https://pastebin.com/2EFxTd3p ]
```

Also, despite the fact that it failed, the `configuration.yaml` file is updated with the following contents:
```
devices:
   '0x00158d0002c5b501':
    friendly_name: '0x00158d0002c5b501'
```

## Debug Info
zigbee2mqtt version: 1.12.0
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}`

nateww, 2020-03-29:
Note, getting a debug log that can be added to pastebin was more difficult than it should be because of output colorization, which is not easily disabled despite all of my attempts (`npm config set color false` as well as setting a number of different environment variables).  I also attempted to disable colorization by editing `lib/util/logger.js`, but it only disabled some of the colorization.

nateww, 2020-05-30:
Because the issue was ignored does not make it irrelevant...

Koenkk, 2020-05-30:
- First make sure you are on the latest zigbee2mqtt (1.13.1)
- Now try the steps from https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails
- If that doesn't fix it please provide the herdsman debug logging. (the pastebin link from OP is expired)

closed 2020-07-26 by stale[bot]