Koenkk/zigbee2mqtt #3229: FR: send individual mqtt msg on event "device online" and "device offline"

M0ebiu5, 2020-03-29:
I would like to request the following feature:

If a device comes online, zigbee2mqtt should send a predefined msg (config file).
The same should also be possible for the "offline" event.

Use case: one could initialize devices on startup. eg it is possible to set the brightness of the tradfri lamps to 0 - if they have a transition time >= 3sec - before it emmits light. So if the device gets power, it stays dark and the software has full control off the startup timing and values.

sjorge, 2020-03-29:
https://www.zigbee2mqtt.io/information/availability.html Is pretty close, you can also use the log messages from the bridge endpoint, they have a bit more information.

Sadly by the time zigbee2mqtt sees the device announce the bulbs is already lit (at least for hue (unless confined to be off after a power failure), innr, osram, Müller light, and Trådfri)

M0ebiu5, 2020-03-30:
Thanks for the info about availability - my current workaround uses availability together with node-red - as soon as it gets an "online" msg, node-red immediately sends the "turn of" command.

As written before this works here with tradfri lamps, which i prefer because they perform well and only cost half of the hue lamps (i guess it is possible for hue lamps, that they stay off).

sjorge, 2020-03-31:
@Koenkk I think something like this might work for @M0ebiu5 and I think it might be useful for other people too. But I wrote this extension in ~1h while waiting for my backup to finish, it is not well tested.

```javascript
const logger = require('../util/logger');
const settings = require('../util/settings');
const utils = require('../util/utils');
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');
const BaseExtension = require('./baseExtension');

/**
 * Apply an initial state to a device imediatly after deviceAnnounces
 */
class DeviceInitialize extends BaseExtension {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus) {
        super(zigbee, mqtt, state, publishEntityState, eventBus);
    }

    async onDeviceAnnounce(announcedDevice) {
        // if we just pass `announcedDevice` we are missing entity.endpoint
	const entity = this.zigbee.resolveEntity(announcedDevice.ieeeAddr);
        const device = entity.device;
        const mapped = entity.mapped;
        const target = entity.endpoint;
        const converters = entity.mapped.toZigbee;
        const options = entity.settings;

        // abort early if we are interviewing
        if (!device.interviewCompleted) return;

        // abort early if we don't have initialize settings
        const json = options.hasOwnProperty("initialize") ? options.initialize : null;
        if (!json) return;

        logger.debug(`Applying initial state '${JSON.stringify(json)}' to '${entity.name}'`);
        const entries = Object.entries(json);
        const usedConverters = [];
        for (let [key, value] of entries) {
            let actualTarget = target;
        
            let postfix = null;
            if (key.includes('_')) {
                const underscoreIndex = key.lastIndexOf('_');
                const possiblePostfix = key.substring(underscoreIndex + 1, key.length);
                if (utils.getPostfixes().includes(possiblePostfix)) {
                    postfix = possiblePostfix;
                    key = key.substring(0, underscoreIndex);
                    const device = target.getDevice();
                    actualTarget = device.getEndpoint(mapped.endpoint(device)[postfix]);
                }
            }
        
            const converter = converters.find((c) => c.key.includes(key));
        
            if (usedConverters.includes(converter))
                continue;
        
            if (!converter || !converter.convertSet) {
                logger.error(`No converter available for '${key}'`);
                continue;
            }
        
            const meta = {
                endpoint_name: postfix,
                options,
                message: json,
                logger,
                device,
                state: this.state.get(settings.ID) || {},
                mapped,
            };
            await converter.convertSet(actualTarget, key, value, meta);
        
            usedConverters.push(converter);
        }
    }

    onZigbeeEvent(type, data, mappedDevice, settingsDevice) {
	if (data.device && type === 'deviceAnnounce')
            this.onDeviceAnnounce(data.device);
    }
}

module.exports = DeviceInitialize;
```

It's very rough around the edge and I am not 100% happy with this code. But it works, it successfully turns my innr bulb that I have at my desk to red after restoring mains power.

I have this in configuration.yaml for the bulb:
```yaml
devices:
  '0x14b457fffe2bd760':
    friendly_name: bedroom/desk_lamp/bulb
    initialize:
      color: "#FF0000"
```

At the very least I think the name needs to change `initialize` can mean a lot of things and I don't think it fits here. Perhaps `intial_state` or something is better.

Anyway, sorry for the random code drop. I was kind of procrastinating on #3177 and koenkk/zigbee-herdsman-converter#1086. Because those are a lot more work and I need to move my laptop to the tiny server closet.

sjorge, 2020-04-04:
@Koenkk been thinking about this for a bit more during my walks...

Is there a way to call entityPublish's onMQTTMessage from inside another extension?
That would make implementing this a lot cleaner, ...

Things currently bad:
- naming
- cannot select an endpoint
- unable to specify a delay, imediately works for for my innr bulb but maybe not universally so
- too much code duplication

Koenkk, 2020-04-04:
I'm wondering if this should be implemented in Zigbee2mqtt after all. The problem with it that this will hard to do 100% right (bulbs will flash shortly) and currently this seems to be the only user interested in this.

I thinks its relatively easy to respond to the device announce which is published to the MQTT log topic. 

Main point here is that I want to avoid adding features which are only used by very few people.

sjorge, 2020-04-04:
True... I do it with node-red and it works fine... I just did it for procrastination reason (and to get my hands dirty with the extension api)

But it does work way better like this, my innr doesn't even turn on (well it did 1 time) using it via the extension vs all the time when using node-red. Then again i did have the bulb just 'freeze' one time too, probably due to baraging it with zigbee messages super early.

M0ebiu5, 2020-04-04:
I thought it would be nice to have a way to initialize devices (it could be any device, not only lights), but if you think it's not useful, dont lose time with it.
Thanks for your efforts with zigbee2mqtt!

closed 2020-04-04 by M0ebiu5

Koenkk, 2020-04-04:
I think for these use cases it would be better to allow users to write own extensions. E.g. create an extension directory in the data directory which zigbee2mqtt will automatically load. This will also allow users to write automations directly in Zigbee2mqtt.

sjorge, 2020-04-04:
> I think for these use cases it would be better to allow users to write own extensions. E.g. create an extension directory in the data directory which zigbee2mqtt will automatically load. This will also allow users to write automations directly in Zigbee2mqtt.

<3 I'd love that, I might dutch node-red and write one that does HAP integration then haha. But I never felt it was a good fit for z2m so never bother.

Koenkk, 2020-04-05:
@sjorge good, I'm currently doing some internal z2m refactors and after this I will add it.

sjorge, 2020-04-05:
Excellent, I'll clean this up some more then and it can be a example module.

Perhaps we can move some of the on/off toggles to load extensions to a list where we list what we want to load. (Keeping the old behavior for ha and availability) That would remove the need to add a new field for each extension to the internal configuration file definition.

```yaml
extensions:
  - name: initialize
    configuration:
      config: option
      another: option here too
  - availability
```

Not sure how we can easily allow and extension to add per device config though 🤔

sjorge, 2020-04-22:
I wrote a minimal version of this using the new framework:
https://gist.github.com/sjorge/5018d939402f35b57a1f9b9849f19d91