Koenkk/zigbee2mqtt #3231: Remote serial supported?

dvbit, 2020-03-29:
Not really an issue so feel free to close and handle as low priority.
I am trying to use a dongle remotely through remserial (remotization of serial through tcp/ip socket) as the server is far from most of the devices.

Serial is correctly remotized as the setup is completely equal to another remote port I am using (zwave dongle).

However the application does not start 
Question : is the application relying on physical colocation with the dongle somehow?
@Koenkk Just wanted to rule out basic incompatibility before digging further  

PS: I searched for such a setup but could not find anyone adopted it. I also posted for help on the forum.

Thanks 

the TI 1532p dongle is running on a raspberry PI.
All software (pi OS, PC OS, application ) releases are latest.

`node index.js

zigbee2mqtt:info 2020-03-28 22:45:10: Logging to console and directory: ‘/opt/zigbee2mqtt/data/log/2020-03-28.22-45-10’ filename: log.txt
zigbee2mqtt:info 2020-03-28 22:45:10: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info 2020-03-28 22:45:10: Starting zigbee-herdsman…
zigbee2mqtt:error 2020-03-28 22:45:17: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-28 22:45:17: Failed to start zigbee
zigbee2mqtt:error 2020-03-28 22:45:17: Exiting…
zigbee2mqtt:error 2020-03-28 22:45:17: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
at ZStackAdapter. (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
at Generator.throw ()
at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: node index.js
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! /home/reinan/.npm/_logs/2020-03-28T21_45_17_992Z-debug.log

0 info it worked if it ends with ok
1 verbose cli [ ‘/usr/bin/node’, ‘/usr/bin/npm’, ‘start’ ]
2 info using npm@6.13.4
3 info using node@v12.16.1
4 verbose run-script [ ‘prestart’, ‘start’, ‘poststart’ ]
5 info lifecycle zigbee2mqtt@1.12.0~prestart: zigbee2mqtt@1.12.0
6 info lifecycle zigbee2mqtt@1.12.0~start: zigbee2mqtt@1.12.0
7 verbose lifecycle zigbee2mqtt@1.12.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.12.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/home/reinan/.local/bin:/opt/intel/openvino_2020.1.023/deployment_tools/model_optimizer:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.12.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.12.0~start: Args: [ ‘-c’, ‘node index.js’ ]
11 silly lifecycle zigbee2mqtt@1.12.0~start: Returned: code: 1 signal: null
12 info lifecycle zigbee2mqtt@1.12.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.12.0 start: node index.js
13 verbose stack Exit status 1
13 verbose stack at EventEmitter. (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack at EventEmitter.emit (events.js:311:20)
13 verbose stack at ChildProcess. (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack at ChildProcess.emit (events.js:311:20)
13 verbose stack at maybeClose (internal/child_process.js:1021:16)
13 verbose stack at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.12.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.0-8-amd64
17 verbose argv “/usr/bin/node” “/usr/bin/npm” “start”
18 verbose node v12.16.1
19 verbose npm v6.13.4`

sjorge, 2020-03-29:
This might interest you: https://github.com/Koenkk/zigbee-herdsman/pull/80

@Koenkk do we have any docs on the above? I had a look but couldn’t find any, but I remembered the PR

Koenkk, 2020-03-30:
It's documented here: https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

dvbit, 2020-03-30:
Thanks! Was looking for the other tool . Will try this too!

dvbit, 2020-03-31:
Well After setting it up I have a
Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
at startup

Same error I have got while trying on x86 laptop while trying first times your app.

I will upload details log tomorrow.

jmgarciamari, 2020-04-13:
I am in the same situation, also changed config to:
`127.0.0.1,15000:raw:0:/dev/ttyUSB0:9600 8DATABITS NONE 1STOPBIT`
for testing in local and fails too, I have z2m in a docker container in a rPi3.

dvbit, 2020-05-24:
I finally got it working but only when I run it manually online with npm start.
If I try using the service (systemd) it fails 
this is my service file:

[Unit]
Description=zigbee2mqtt
After=network.target
StartLimitInterval=200
StartLimitBurst=5
[Service]
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
RestartSec=30
User=reinan

[Install]
WantedBy=multi-user.target

dvbit, 2020-05-24:
and this is the debug log on node
`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.4
3 info using node@v12.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.13.1~prestart: zigbee2mqtt@1.13.1
6 info lifecycle zigbee2mqtt@1.13.1~start: zigbee2mqtt@1.13.1
7 verbose lifecycle zigbee2mqtt@1.13.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.13.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.13.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.13.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.13.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.13.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.13.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:310:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:310:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.13.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.0-9-amd64
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.16.3
19 verbose npm  v6.14.4
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.13.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.13.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
`

dvbit, 2020-05-24:
So . next try will be (if it exists as option) allow more time for systemd for the application to starting from cli takes some sime before getting OK from npm .Any suggestion of a Systemd parameter?

dvbit, 2020-05-24:
tried 
Timeousec option
no luck

if I run manually 8 times out of 10 it starts
via systemctl 10 on 10 No

dvbit, 2020-05-24:
this is the detailed debug log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.4
3 info using node@v12.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.13.1~prestart: zigbee2mqtt@1.13.1
6 info lifecycle zigbee2mqtt@1.13.1~start: zigbee2mqtt@1.13.1
7 verbose lifecycle zigbee2mqtt@1.13.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.13.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.13.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.13.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.13.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.13.1~start: Failed to exec start script
13 VErbose stack Error: zigbee2mqtt@1.13.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:310:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:310:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@10 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.4
3 info using node@v12.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.13.1~prestart: zigbee2mqtt@1.13.1
6 info lifecycle zigbee2mqtt@1.13.1~start: zigbee2mqtt@1.13.1
7 verbose lifecycle zigbee2mqtt@1.13.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.13.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.13.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.13.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.13.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.13.1~start: Failed to exec start script
13 VErbose stack Error: zigbee2mqtt@1.13.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:310:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:310:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.13.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.0-9-amd64
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.16.3
19 verbose npm  v6.14.4
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.13.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.13.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
~                                                                                                                                         
~                            .13.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.0-9-amd64
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.16.3
19 verbose npm  v6.14.4
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.13.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.13.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
~                                                                                                                                         
~

dvbit, 2020-05-24:
I see an error "12 info lifecycle zigbee2mqtt@1.13.1start: Failed to exec start script"

toxuin, 2020-05-26:
Is there a microcontroller-based ser2net analogue? Has anyone tested this with esp8266 yet? Sounds like a perfect application for it.

closed 2020-07-26 by stale[bot]