Koenkk/zigbee2mqtt #3233: Failed to start zigbee

zrainx, 2020-03-29:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:  1.12.0 (commit #840b9d9)
CC253X firmware version:   20190608

I have the problem of  "Failed to start zigbee".
The hardware is  CC2530 + CC2592  and CH340 (usb to serial)

It's on window 10 , The log is below:

> zigbee2mqtt@1.12.0 start D:\repo\Z2M\zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-29 21:48:19: Logging to console and directory: 'D:\repo\Z2M\zigbee2mqtt\data\log\2020-03-29.21-48-19' filename: log.txt
zigbee2mqtt:info  2020-03-29 21:48:20: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-03-29 21:48:20: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-03-29 21:48:27: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-29 21:48:27: Failed to start zigbee
zigbee2mqtt:error 2020-03-29 21:48:27: Exiting...
zigbee2mqtt:error 2020-03-29 21:48:27: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (D:\repo\Z2M\zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (D:\repo\Z2M\zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\zrainx\AppData\Roaming\npm-cache\_logs\2020-03-29T13_48_27_683Z-debug.log

pixeldoc2000, 2020-03-30:
Maybe you (virtual) com port hast changed? Verify you are using the correct Com Port at the Device Manager in Windows.

Did you connect the CH340 to a different USB-Port?

BPA12, 2020-03-31:
Same problem for me. I installed zigbee2mqtt on Openhabian. 
If I start it with "sudo npm start", it works fine.
But starting as a deamon doesn't work.

Log:
Mar 31 21:20:34 openhab systemd[1]: Started zigbee2mqtt.
Mar 31 21:20:37 openhab npm[5395]: > zigbee2mqtt@1.12.0 start /opt/zigbee2mqtt
Mar 31 21:20:37 openhab npm[5395]: > node index.js
Mar 31 21:20:39 openhab npm[5395]: zigbee2mqtt:info  2020-03-31 21:20:39: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-31.21-20-39' filename: log.txt
Mar 31 21:20:39 openhab npm[5395]: /opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:310
Mar 31 21:20:39 openhab npm[5395]:         throw er
Mar 31 21:20:39 openhab npm[5395]:         ^
Mar 31 21:20:39 openhab npm[5395]: Error: EACCES: permission denied, unlink '/opt/zigbee2mqtt/data/log/2020-03-31.21-15-24/log.txt'
Mar 31 21:20:39 openhab npm[5395]:     at Object.unlinkSync (fs.js:956:3)
Mar 31 21:20:39 openhab npm[5395]:     at rimrafSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:303:17)
Mar 31 21:20:39 openhab npm[5395]:     at options.readdirSync.forEach.f (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:336:39)
Mar 31 21:20:39 openhab npm[5395]:     at Array.forEach (<anonymous>)
Mar 31 21:20:39 openhab npm[5395]:     at rmkidsSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:336:26)
Mar 31 21:20:39 openhab npm[5395]:     at rmdirSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:329:7)
Mar 31 21:20:39 openhab npm[5395]:     at Function.rimrafSync [as sync] (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:301:9)
Mar 31 21:20:39 openhab npm[5395]:     at directories.forEach (/opt/zigbee2mqtt/lib/util/logger.js:92:20)
Mar 31 21:20:39 openhab npm[5395]:     at Array.forEach (<anonymous>)
Mar 31 21:20:39 openhab npm[5395]:     at DerivedLogger.cleanup (/opt/zigbee2mqtt/lib/util/logger.js:90:21)
Mar 31 21:20:39 openhab npm[5395]: npm ERR! code ELIFECYCLE
Mar 31 21:20:39 openhab npm[5395]: npm ERR! errno 1
Mar 31 21:20:39 openhab npm[5395]: npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
Mar 31 21:20:39 openhab npm[5395]: npm ERR! Exit status 1
Mar 31 21:20:39 openhab npm[5395]: npm ERR!
Mar 31 21:20:39 openhab npm[5395]: npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
Mar 31 21:20:39 openhab npm[5395]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Mar 31 21:20:39 openhab npm[5395]: npm ERR! A complete log of this run can be found in:
Mar 31 21:20:39 openhab npm[5395]: npm ERR!     /home/openhabian/.npm/_logs/2020-03-31T19_20_39_749Z-debug.log
Mar 31 21:20:39 openhab systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Mar 31 21:20:39 openhab systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Mar 31 21:20:40 openhab systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Mar 31 21:20:40 openhab systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 131.
Mar 31 21:20:40 openhab systemd[1]: Stopped zigbee2mqtt.

Does anyone knows a solution?

pixeldoc2000, 2020-04-03:
> Mar 31 21:20:39 openhab npm[5395]: Error: EACCES: permission denied, unlink '/opt/zigbee2mqtt/data/log/2020-03-31.21-15-24/log.txt'

I think your `/opt/zigbee2mqtt/data` permissions are not correct. I think the service has no permission the write to the directory according to the error message.
If you run it with sudo (as root) you can write "everywhere". 

Please use "Insert Code" at the Toolbar to post your logs.

zrainx, 2020-04-04:
The problem has been solved yesterday. It's a hardware problem. My hardware board uses P1.4 and P1.5 as uart pins，but P0.2 and P0.3 is used as uart pins in the the CC253X firmware (version: 20190608). 
I's ok after I changed to P0.2 and P0.3 .

closed 2020-04-04 by zrainx