Koenkk/zigbee2mqtt #3235: Can't pair Tradfri because of missing response to Node Descriptor Requests

TilmanK, 2020-03-29:
#Bug Report
## What happened
I'm back at the point where I need to repair my Tradfri LED drivers (https://www.zigbee2mqtt.io/devices/ICPSHC24-30EU-IL-1.html), but I'm not able to get them paired properly.

I tried to compare the traffic between Z2M and a HUE bridge. After the Transport key has been sent, Z2M doesn't receive any messages anymore from the device (it then fails after some Node Descriptor Requests). This does not happen with the HUE bridge.

## Debug Info
zigbee2mqtt version: 1.12.0
CC253X firmware version: 20191106
[logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/4399124/logs.zip)

TilmanK, 2020-03-29:
I attached pcap logs for HUE and Z2M, as well as Z2M debug logs.

TilmanK, 2020-03-29:
And some additional Information: My Hue Bridge is on Channel 25, Z2M on 24. I use a custom key in Z2M, the Hue Bridge uses the default setting.

Koenkk, 2020-03-29:
When pairing trying to pair it with zigbee2mqtt, could you sniff the traffic on channel 25 (of the hue bridge)?

TilmanK, 2020-03-29:
I switch off Z2M when using the Hue bridge and the Hue bridge when using Z2M.

You're assuming that it talks to the Hue bridge instead of Z2M, right? Shall I still sniff, even though the Hue bridge is offline while pairing with Z2M?

sjorge, 2020-03-29:
Also try unplugging and replying your coordinator stick before trying to pair them.

TilmanK, 2020-03-31:
@sjorge Yeah, I tried that. Doesn't help sadly. What could be the reason that the device doesn't send anything to the network after receiving the transport key? This happens to all my three Tradfri devices.

Koenkk, 2020-03-31:
@TilmanK I my case this happens when the bulb is not reset correctly, for some devices this seems to be tricky even though the bulbs blinks indicating that is has been reset. I would recommend using touchlink in this case to make sure that the reset isn't the problem https://www.zigbee2mqtt.io/information/touchlink.html

TilmanK, 2020-03-31:
@Koenkk I just tried resetting via Touchlink. With Z2M and a Hue Dimmer. I can see the scan request in the sniffed traffic without any positive result. The device just doesn't send anything after the transport key was sent. I have the issue on all 3 led drivers, hence I assume that this must be a problem with Z2M or my coordinator.

Isn't there someone else experiencing the same? I just can't imagine what the problem could be.

Koenkk, 2020-03-31:
Could you maybe share the sniff when pairing it to hue vs z2m? (together with the z2m debug log)

TilmanK, 2020-03-31:
The logs are attached to the initial report already. 

I'll flash the firmware tomorrow and will try again.

TilmanK, 2020-04-01:
@Koenkk It's exactly the same issue with the new firmware for the CC26X2R1. No response at all from the device. Can you see something in the logs from above? I cannot find any difference that would explain the issue.

Koenkk, 2020-04-01:
I see the following differences.

1. In the z2m sniff multiple devices respond to the beacon request, while in the Hue only one responds (the bridge) but this make sense as there is probably only 1 device.
[image]
2. Right after the join in the Hue network the tradfri does a device announcement while it doesn't in the z2m sniff
[image]
3. In both hue and z2m, the first node descriptor request fails, after that both z2m and hue do a route request, in hue it's answered, in z2m not

Let's first make sure difference 1 isn't the issue. Could you power off all routes in your network and try to pair the device again (and provide a sniff of this)?

TilmanK, 2020-05-10:
Sorry for the late response, I just did not find the time for doing that test.

I powered of every router connected to my network - unfortunately, the issue stays. I upgraded to 1.13.0 before doing that test.

[tradfri_wo_other_router.zip](https://github.com/Koenkk/zigbee2mqtt/files/4605425/tradfri_wo_other_router.zip)

closed 2020-07-18 by stale[bot]

TilmanK, 2020-10-04:
Don't want to warm this up, but just for others that have the same problem. My zigbee network was configured to use channel 24. I set up a test network on channel 11 and pairing worked flawless. Seems the Tradfris have a problem with those custom channels.

Happy repairing.