Koenkk/zigbee2mqtt #3241: struggling with re-joining Eurotronic Spirit Zigbee

bartplessers, 2020-03-29:
# Bug Report
## What happened
Eurotronic Spirit Zigbee
- joined to network
- un-joined
- trying to re-join

but no succes. It seems I can not re-join the device to my zigbee network anymore.

Last message from un-joining:
```

Client mosq-2rWcjwaJAFYzCqllFa received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/config', ... (208 bytes))
{"version":"1.12.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}

Client mosq-2rWcjwaJAFYzCqllFa received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x00158d00053d404d', ... (373 bytes))
{"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"linkquality":13,"local_temperature":22.5,"occupied_heating_setpoint":18,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"current_heating_setpoint":18,"eurotronic_error_status":0,"battery":100,"battery_low":0}

Client mosq-2rWcjwaJAFYzCqllFa received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/log', ... (96 bytes))
{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x00158d00053d404d"}}
```

## What did you expect to happen
device can be re-joined to network

Remark: I joined the device in an earlier version of zigbee2mqtt (can't remember exact version)
So maybe it's not only a "re-joining" problem, but a general "joining" problem. I can't check that because I don't have a virgin device anymore.

## Debug Info
running 
- C2531 on RPI3
- docker
- hassio in docker
- zigbee2mqtt hassio addon
- {"version":"1.12.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}

bartplessers, 2020-03-29:
Important:
I just succeeded to re-join the device.
I had to shutdown my RPI completely, and boot it up again.
- restarting zigbee2mqtt addon was NOT enough
- restarting hassio docker was NOT engough

so, maybe this issue can be closed...
But I leave it open for comments: I think re-joining should be possible without shutting down my server.

kind regards,
B

sjorge, 2020-03-29:
Yeah I have the bug a lot too, see #3177
You can also reset the stick via mqtt and then restart z2m

antonakv, 2020-03-30:
I have the same issue

closed 2020-06-06 by stale[bot]